SpikeInterface/spikeinterface #1582: How to read spikes after they are saved by run_sorter?

programerkotik, 2023-04-26:
Hi,

How one should read detected spikes after the run sorter saves output? 
I have the folder saved by spikeinterface with subfolders clustering, sorting, waveforms, and some json files. Now, I would like to continue working on it so I try to initialize it using `spikeinterface.core.load_extractor`. I send to the function the path to the folder which was saved by run_sorter. It returns the error: 
`ValueError: This folder is not a cached folder processing/spike-sorting/spykingcircus2_output/sorter_output`

Would you know what is wrong here?

Thanks and best regards,
Anna

samuelgarcia, 2023-04-26:
Hi Anna,
there is way of doing this.

Save explicitly the sorting object
```python
sorter_output_folder = '/path/to/spkykingcircus2'
sorting = run_soter('spkykingcircus2', recording, output_folder=sorter_output_folder)
save_folder = '/path/to/sorting_saved'
sorting.save(folder=save_folder)

# and then later
sorting_reload = load_extractor(save_folder)

```

Or reload the sorter output folder but you need to keep a bigger folder
```python
sorter_output_folder = '/path/to/spkykingcircus2'
sorting = run_soter('spkykingcircus2', recording, output_folder=sorter_output_folder)

# and then later
sorting_reload = read_sorter_folder(sorter_output_folder)
```

programerkotik, 2023-04-28:
Thanks, it works!

closed 2023-04-28 by programerkotik