SpikeInterface/spikeinterface #1585: run_sorter fails to find files on computing cluster

skyetomez, 2023-04-26:
Hello,

When I use the `run_sort` command in my python script I always get the following error: 
```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
File ~/.conda/envs/spike_sort/lib/python3.10/site-packages/urllib3/connectionpool.py:703, in HTTPConnectionPool.urlopen(self, method, url, body, headers, retries, redirect, assert_same_host, timeout, pool_timeout, release_conn, chunked, body_pos, **response_kw)
    702 # Make the request on the httplib connection object.
--> 703 httplib_response = self._make_request(
    704     conn,
    705     method,
    706     url,
    707     timeout=timeout_obj,
    708     body=body,
    709     headers=headers,
    710     chunked=chunked,
    711 )
    713 # If we're going to release the connection in ``finally:``, then
    714 # the response doesn't need to know about the connection. Otherwise
    715 # it will also try to release it and we'll have a double-release
    716 # mess.

File ~/.conda/envs/spike_sort/lib/python3.10/site-packages/urllib3/connectionpool.py:398, in HTTPConnectionPool._make_request(self, conn, method, url, timeout, chunked, **httplib_request_kw)
    397     else:
--> 398         conn.request(method, url, **httplib_request_kw)
    400 # We are swallowing BrokenPipeError (errno.EPIPE) since the server is
    401 # legitimately able to close the connection after sending a valid response.
    402 # With this behaviour, the received response is still readable.

File ~/.conda/envs/spike_sort/lib/python3.10/http/client.py:1282, in HTTPConnection.request(self, method, url, body, headers, encode_chunked)
   1281 """Send a complete request to the server."""
-> 1282 self._send_request(method, url, body, headers, encode_chunked)

File ~/.conda/envs/spike_sort/lib/python3.10/http/client.py:1328, in HTTPConnection._send_request(self, method, url, body, headers, encode_chunked)
   1327     body = _encode(body, 'body')
-> 1328 self.endheaders(body, encode_chunked=encode_chunked)

File ~/.conda/envs/spike_sort/lib/python3.10/http/client.py:1277, in HTTPConnection.endheaders(self, message_body, encode_chunked)
   1276     raise CannotSendHeader()
-> 1277 self._send_output(message_body, encode_chunked=encode_chunked)

File ~/.conda/envs/spike_sort/lib/python3.10/http/client.py:1037, in HTTPConnection._send_output(self, message_body, encode_chunked)
   1036 del self._buffer[:]
-> 1037 self.send(msg)
   1039 if message_body is not None:
   1040 
   1041     # create a consistent interface to message_body

File ~/.conda/envs/spike_sort/lib/python3.10/http/client.py:975, in HTTPConnection.send(self, data)
    974 if self.auto_open:
--> 975     self.connect()
    976 else:

File ~/.conda/envs/spike_sort/lib/python3.10/site-packages/docker/transport/unixconn.py:30, in UnixHTTPConnection.connect(self)
     29 sock.settimeout(self.timeout)
---> 30 sock.connect(self.unix_socket)
     31 self.sock = sock

FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

ProtocolError                             Traceback (most recent call last)
File ~/.conda/envs/spike_sort/lib/python3.10/site-packages/requests/adapters.py:489, in HTTPAdapter.send(self, request, stream, timeout, verify, cert, proxies)
    488 if not chunked:
--> 489     resp = conn.urlopen(
    490         method=request.method,
    491         url=url,
    492         body=request.body,
    493         headers=request.headers,
[... 192 lines cut ...]
    198 else:
    199     self._version = version

File ~/.conda/envs/spike_sort/lib/python3.10/site-packages/docker/api/client.py:221, in APIClient._retrieve_server_version(self)
    216     raise DockerException(
    217         'Invalid response from docker daemon: key "ApiVersion"'
    218         ' is missing.'
    219     )
    220 except Exception as e:
--> 221     raise DockerException(
    222         f'Error while fetching server API version: {e}'
    223     )

DockerException: Error while fetching server API version: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))
```
I tried this using the following images: 
- docker/singularity_image = ninjaben/kilosort3:v0.0.28
- docker/singularity_image = spikeinterface/kilosort3-compiled-base:latest
- docker/singularity_image = True/False
- docker/singularity_image = "</absolute/path/to/built/image.sif>"
- docker/singularity_image = "</relative/path/to/built/image.sif>"

I'm running the script on our HPC so I don't think docker would work anyways. We use the apptainer fork instead of singularity also. 

Thank you for the advice.

samuelgarcia, 2023-04-27:
Hi Skyler.
given your trace back it is hard where the error come from.

Are you able to run the standard test we have like`test_run_sorter_singularity()` in the sorters module ?
Could it be a permission problem ?

h-mayorquin, 2023-04-27:
It seems that we need better loggin methods for the sorters and expose them through the docker interface.

skyetomez, 2023-04-27:
Please use my screenname @samuelgarcia. 

When I run `spikeinterface.sorters.test_run_sorter_singularity()`  it returns an attribute not found.

When I run `test_run_sorter_singularity` from the following code:
```
import spikeinterface

def test_run_sorter_singularity():
    mearec_filename = spikeinterface.download_dataset(
        remote_path='mearec/mearec_test_10s.h5', unlock=True)
    output_folder = cache_folder / 'sorting_tdc_singularity'

    recording, sorting_true = read_mearec(mearec_filename)

    sorter_params = {'detect_threshold': 4.9}

    singularity_image = 'spikeinterface/tridesclous-base:1.6.4-1'

    sorting = run_sorter('tridesclous', recording, output_folder=output_folder,
                         remove_existing_folder=True, delete_output_folder=False,
                         verbose=True, raise_error=True, singularity_image=singularity_image,
                         **sorter_params)
    print(sorting)

    # basic check to confirm sorting was successful
    assert 'Tridesclous' in sorting.to_dict()['class']
    assert len(sorting.get_unit_ids()) > 0
```
I receive the error `ModuleNotFoundError: No module named 'datalad'`.

I also tried building the containers using singularity from docker from `spikeinterface/kilosort3-compiled-base:latest' but it returns the error that python is not on path. 

Thank you for the suggestions.  Is there something else I can provide?

h-mayorquin, 2023-04-28:
You should install datalad:

https://spikeinterface.readthedocs.io/en/latest/development/development.html#installing-datalad

skyetomez, 2023-04-28:
I cannot install datalad as a user on a cluster since I'm a user and haven't got sudo privileges. 
I can build a test container tomorrow and try. 

Is there any advice for the original problem for running a singularity image from the error? Anything I can provide?
Thank you

h-mayorquin, 2023-05-02:
How large is your date? Can you reproduce the error with data that you can share? It would be way easier to debugg if we can reproduce the error in our side.

skyetomez, 2023-05-02:
@h-mayorquin That is a good idea. I will try to generate data and see provide the steps to reproduce the error.

alejoe91, 2023-11-06:
@skyetomez were you able to reproduce the error without datalad?

skyetomez, 2023-11-06:
I ended up just downloading the sorter and manually pointing to it. IIRC, I was running this without datalad. This can be closed. Thank you!

closed 2023-11-06 by skyetomez