SpikeInterface/spikeinterface #1588: run sorter gives error (but still gives output)

ZoeChen96, 2023-04-27:
Hi there,

I am using spikeinterface for several sorters. I ran it correctly before, but today when redo it I face many KeyError. Weirdly, although it raises errors, the sorter still finishes running and gives the accuracy report. I paste the errors below:

```
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
    registry[rtype][name] -= 1
[... 111 lines cut ...]
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
[... cut ...]
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/joblib/externals/loky/backend/resource_tracker.py", line 287, in main
    registry[rtype][name] -= 1
KeyError: '/dev/shm/joblib_memmapping_folder_4103353_837e1c5314a546af860c0d868a0bfba4_a86f77cc47324c679e8e3495eac995b4/4103353-139972258045376-5acdeea71d4a4d188b5bc11b8a0ae7d5.pkl'
```

ZoeChen96, 2023-04-27:
BTW, the script I used is attached too: (sorry for the beginner level's code)
```
import spikeinterface.extractors as se 
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
from pathlib import Path
import warnings
warnings.simplefilter("ignore")
import sys

set_list = ['30k_Q0_Fsw3_A0']
fs = 30000
#params = ss.get_default_sorter_params("herdingspikes")

base_folder = Path(".")
cmp_method = ("raw_count", "by_unit", "pooled_with_average")
original_stdout = sys.stdout # Save a reference to the original standard output
sorter_params = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
ss.IronClustSorter.set_ironclust_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/ironclust')
ss.Kilosort3Sorter.set_kilosort3_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/Kilosort')

def sorter_loop(srtr,sorter,set_list): #set list: ['1','2','3','3_14','4_13']
    filename = 'new_templates_sorter_loop2.txt'
    acc_list = []
    for i in range(len(set_list)):
        set_name = set_list[i]
        print('processing ' + set_name +'...')
        recording_name = '/imec/other/macaw/projectdata/mearec_datasets/new_templates_datasets/set'+set_name+'/set'+set_name+'_recording.h5'
        #pdb.set_trace()
        recording = se.MEArecRecordingExtractor(recording_name)
        #recording = recording.frame_slice(start_frame=0*fs, end_frame=60*fs)
        sorting_GT = se.MEArecSortingExtractor(recording_name)
        if srtr == ('IC' or 'KS'): 
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results"+srtr+set_name), verbose=True,**sorter_params)
        else:
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results"+srtr+set_name), verbose=False)
        comp = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=True,delta_time = 1)
        #comp = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=True)
        tmp_accuracy = comp.get_performance(method = cmp_method[2])['accuracy']
        acc_list.append(tmp_accuracy)
        print('with filter, set = ',set_name,',accuracy = ', tmp_accuracy)
    print('------------------'+srtr+' finish-----------------------')

    with open(filename, "a") as f:    
        sys.stdout = f # Change the standard output to the file we created.   
        print('------------------Results for '+ sorter+'---------------')  
        for i in range(len(set_list)):
            print('with filter, set = ',set_list[i],',accuracy = ', acc_list[i])
    sys.stdout = original_stdout
    return

def sorter_loop_nofilt(srtr,sorter,set_list): #set list: ['1','2','3','3_14','4_13']
    filename = 'new_templates_sorter_60dur.txt'
    acc_list = []
    for i in range(len(set_list)):
        set_name = set_list[i]
        print('processing ' + set_name +'...')
        recording_name = '/imec/other/macaw/projectdata/mearec_datasets/new_templates_datasets/set'+set_name+'/set'+set_name+'_recording.h5'
        #pdb.set_trace()
        recording = se.MEArecRecordingExtractor(recording_name)
        sorting_GT = se.MEArecSortingExtractor(recording_name)
        recording = recording.frame_slice(start_frame=0*fs, end_frame=60*fs)
        if srtr == 'IC': 
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), filter = False, verbose=False,**sorter_params)
        elif srtr == 'KS':  
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), freq_min = 1,verbose=False,**sorter_params)
        elif srtr == 'pyKS' :
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), fshigh = 1,verbose=False,**sorter_params)
        elif srtr == 'HS':
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), filter = False, verbose=False)
        elif srtr == 'TDC': 
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name),  apply_preprocessing = False, verbose=False)
        elif srtr == 'SC': 
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name),  apply_preprocessing = False, verbose=False)
        #comp = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=True,delta_time = 1)
        comp = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=True)
        tmp_accuracy = comp.get_performance(method = cmp_method[2])['accuracy']
        acc_list.append(tmp_accuracy)
        print('no filter,set = ',set_name,',accuracy = ', tmp_accuracy)
    print('------------------'+srtr+' finish-----------------------')

    with open(filename, "a") as f:    
        sys.stdout = f # Change the standard output to the file we created.   
        print('------------------Results for '+ sorter+'---------------')  
        for i in range(len(set_list)):
            print('no filter, set = ',set_list[i],',accuracy = ', acc_list[i])
    sys.stdout = original_stdout
    return
#sorter_loop_nofilt('HS','herdingspikes',set_list)
sorter_loop('HS','herdingspikes',set_list)
#sorter_loop_nofilt('SC','spykingcircus2',set_list)
sorter_loop('SC','spykingcircus2',set_list)
#sorter_loop_nofilt('TDC','tridesclous2',set_list)
sorter_loop('TDC','tridesclous2',set_list)
#sorter_loop_nofilt('IC','ironclust',set_list)
sorter_loop('IC','ironclust',set_list)
#sorter_loop('KS','kilosort3',set_list)
```

alejoe91, 2023-04-27:
Hi @ZoeChen96 

This is an (annoying) internal error from joblib. As you noticed it does not affect the results so you shouldn't worry about it ;)

Cheers
Alessio

ZoeChen96, 2023-04-27:
OK! Then at least I know it will not cause trouble.  Thanks for answering!

closed 2023-04-27 by ZoeChen96