SpikeInterface/spikeinterface #1594: SortingView backend broken

maxjuv, 2023-04-28:
Hello,
I recently update Spikeinterface to 0.98.0.dev0. Since then I cannot use the SortingView plot with the following lines:

sorting = get_my_sorting()
waveform_folder =dataio.get_waveform_folder_path(run_key=session_ID, probe_index=probe_index, sorter=sorter, clean = clean)
we = si.WaveformExtractor.load_from_folder(waveform_folder)
job_kwargs = dict(
    n_jobs=10,
    chunk_duration='1s',
    progress_bar=True)
_ = compute_spike_amplitudes(we,**job_kwargs)
_ = compute_unit_locations(we)
_ = compute_template_similarity(we)
_ = compute_correlograms(we)
plot_sorting_summary(we, curation=True, backend='sortingview')

The associated issue is the following :
Traceback (most recent call last):
  File "/srv/beegfs/scratch/users/j/juventin/Neuropixel_syt17/script_analysis/SortingView_curation.py", line 48, in <module>
    Viewer(session_ID, probe_index, sorter, clean)
  File "/srv/beegfs/scratch/users/j/juventin/Neuropixel_syt17/script_analysis/SortingView_curation.py", line 30, in Viewer
    plot_sorting_summary(we, curation=True, backend='sortingview')
  File "/home/users/j/juventin/spike_sorters/spikeinterface/src/spikeinterface/widgets/base.py", line 116, in widget_func
    W.do_plot(W.backend, **W.backend_kwargs)
  File "/home/users/j/juventin/spike_sorters/spikeinterface/src/spikeinterface/widgets/base.py", line 53, in do_plot
    plotter.do_plot(self.plot_data, **backend_kwargs)
  File "/home/users/j/juventin/spike_sorters/spikeinterface/src/spikeinterface/widgets/sortingview/sorting_summary.py", line 28, in do_plot
    v_average_waveforms = template_plotter.do_plot(dp.templates, generate_url=False,
  File "/home/users/j/juventin/spike_sorters/spikeinterface/src/spikeinterface/widgets/sortingview/unit_templates.py", line 28, in do_plot
    aw_items = [
  File "/home/users/j/juventin/spike_sorters/spikeinterface/src/spikeinterface/widgets/sortingview/unit_templates.py", line 31, in <listcomp>
    channel_ids=list(unit_id_to_channel_ids[u]),
KeyError: 4

Best,
Maxime

alejoe91, 2023-04-28:
Hi Maxime,

This is strange..is your waveform extractor sparse? My intuition is that when loading the waveforms, the unit IDs are cast to string while the sparsity has unit in ints, or something along these lines. Can you try to `print(we.unit_ids)` and check if they correspond to the `we.sparsity.unit_ids_to_channel_ids.keys()`?

maxjuv, 2023-04-28:
we.unit_id and we.sparsity.unit_ids_to_channel_ids, corresponds and are bot int
    print(we.unit_ids)
    print(type(we.unit_ids[0]))
    print(we.sparsity.unit_id_to_channel_ids.keys())
    print(type(list(we.sparsity.unit_id_to_channel_ids.keys())[0]))
-->
[  0   1   3   4   5  17  23  24  25  26  27  28  29  30  31  33  34  35
  36  37  38  39  40  41  42  43  44  45  46  48  50  51  52  53  54  55
  58  59  60  61  62  63  67  68  70  71  73  74  75  76  78  79  80  81
  84  85  86  87  88  92  94  96  97 100 101 103 104 105 106 110 111 112
 113 114 116 123 125 126 127 128 131 132 133 134 135 136 137 138 139 140
 141 142 144 145 146 147 148 150 152 153 154 155 156 157 158 159 160 161
 162 163 164 166 170 172 173 174 175 176 177 178 179 180 182 183 184 185
 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203
 204 206 207 208 210 211 212 213 214 215 216 217 218 219 220 221 223 224
 227 228 229 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245
 246 247 248 249 250 251 252 253 258 259 260 261 262 263 264 265 266 267
 268 269 270 271 272 273 274 275 276 277 278 279 282 283 284 285 286 287
 288 289 290 292 293 294 295 296 297 298 299 300 301 302 303 306 307 308
 309 310 311 312 313 314 315 316 317 318]
<class 'numpy.int64'>
dict_keys([0, 1, 3, 4, 5, 17, 23, 24, 25, 26, 27, 28, 29, 30, 31, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 48, 50, 51, 52, 53, 54, 55, 58, 59, 60, 61, 62, 63, 67, 68, 70, 71, 73, 74, 75, 76, 78, 79, 80, 81, 84, 85, 86, 87, 88, 92, 94, 96, 97, 100, 101, 103, 104, 105, 106, 110, 111, 112, 113, 114, 116, 123, 125, 126, 127, 128, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 144, 145, 146, 147, 148, 150, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 166, 170, 172, 173, 174, 175, 176, 177, 178, 179, 180, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 206, 207, 208, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 223, 224, 227, 228, 229, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 282, 283, 284, 285, 286, 287, 288, 289, 290, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318])
<class 'numpy.int64'>

The same code was running fine last week. It did not work since I updated Spikeinterface on Monday (git pull & pip install -e). The git checkout identifier from my Spikeinterface is 20fdfff3 (last commit from Sam, April 5th)

alejoe91, 2023-04-28:
Thank you Maxime, this is helpful. I'll try to look at it next week. Let us know if you find a solution before!

maxjuv, 2023-04-28:
The issue might come from the "make_serialiazable" fonction, in widgets.sortingview.unit_templates.py

unit_id_to_channel_ids = dp.sparsity.unit_id_to_channel_ids
unit_id_to_channel_indices = dp.sparsity.unit_id_to_channel_indices

unit_ids, channel_ids, unit_id_to_channel_ids = \
    self.make_serializable(dp.unit_ids, dp.channel_ids, unit_id_to_channel_ids)

unit_id_to_channel_ids is a dict which keys are 'int', but after make_seraliazble, the keys are 'str'

closed 2023-05-25 by alejoe91