SpikeInterface/spikeinterface #1595: get_waveforms() should work without sparsity

yger, 2023-04-28:
Currently, when using a sparse waveform_extractor, we have a minor discrepancies in the fact that get_all_templates() are returning the templates in the full space (setting 0 to sparse channels), while get_waveforms() is getting the data in the sparse space, i.e. returning only the channels where you have information. I think it should be good if we could have an option to make get_waveforms() return dense waveforms, even for a sparse extractor. What do you think? This would be much easier for downstreams analysis.

samuelgarcia, 2023-04-28:
yes we could add an option in get_waveforms() to force it dense. But I would be keep sparse by default to avoid memory copy.

alejoe91, 2023-04-28:
I agree! Note that the same behavior is for the `get_template()`, but for returning all templates we need a dense structure because sparsity length might not be constant across units

yger, 2023-04-28:
Ok, I'll do a PR to achieve this