SpikeInterface/spikeinterface #1596: `import spikeinterface.extractors` fails if NWB not installed

DradeAW, 2023-04-28:
Hey,

The typing here creates a crash if NWB is not installed (https://github.com/SpikeInterface/spikeinterface/blob/72b0835cd855cb3fac165a362646a7312c0c73f3/src/spikeinterface/extractors/nwbextractors.py#L40)

```
$ pytest
ImportError while loading conftest '/home/aurelien/Documents/lussac2/tests/conftest.py'.
tests/conftest.py:6: in <module>
    from lussac.core import LussacData, LussacPipeline, MonoSortingData, MultiSortingsData
src/lussac/core/__init__.py:1: in <module>
    from .lussac_data import LussacData, MonoSortingData, MultiSortingsData
src/lussac/core/lussac_data.py:12: in <module>
    import spikeinterface.extractors as se
../../dev/spikeinterface/spikeinterface/src/spikeinterface/extractors/__init__.py:1: in <module>
    from .extractorlist import *
../../dev/spikeinterface/spikeinterface/src/spikeinterface/extractors/extractorlist.py:16: in <module>
    from .nwbextractors import (NwbRecordingExtractor, NwbSortingExtractor,
../../dev/spikeinterface/spikeinterface/src/spikeinterface/extractors/nwbextractors.py:40: in <module>
    def retrieve_electrical_series(nwbfile: NWBFile, electrical_series_name: Optional[str] = None) -> ElectricalSeries:
E   NameError: name 'NWBFile' is not defined
```

h-mayorquin, 2023-04-28:
@DradeAW 
Could you check ?
https://github.com/SpikeInterface/spikeinterface/pull/1597

DradeAW, 2023-04-28:
Yup, with the PR it now works!

Thanks :)

PR #1600 (2023-04-28): Use annotations from future (PEP563) to solve type hint references
OK, so #1597 turn out too big as there were a lot of imports that were exposed naively. I am passing the small patch to fix the import with pynwb here as that one will probably take longer to reivew. 

Copying PR text verbatim from here:

This PR resolves #1596.

As described in the issue, importing extractors fails when pynwb is not installed. The problem arises from the newly added type-hints, which are currently evaluated at import time.

PR #1593 suggests a solution that employs forward references, where type hints are defi [... cut]
closed 2023-04-28 by h-mayorquin