SpikeInterface/spikeinterface #1603: Problems with Kilosort/pyKilosort filtering

ZoeChen96, 2023-04-29:
Hi there, I am trying to use MEArec + SpikeInterface for some sorters accuracy test.

As I want to test around different filtering techniques, I need to disable the filters in sorters. I can disable filters perfectly with herdingspikes/spykingcircus2/tridesclous2/ironclust, but with pykilosort and kilosort it gives the error. I pasted the error for kilosort3 here:
```
Time   0s. Computing whitening matrix.. 

matlab

signal_toolbox

Getting channel whitening matrix... 

----------------------------------------Error using svd

Input to SVD must not contain NaN or Inf.
Error running kilosort3
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/external/kilosortbase.py", line 179, in _run_from_folder
    raise Exception(f'{cls.sorter_name} returned a non-zero exit code')
Exception: kilosort3 returned a non-zero exit code

Traceback (most recent call last):
  File "nofilter_func_loop_sorters.py", line 100, in <module>
    sorter_loop_nofilt('KS','kilosort3',set_list)
  File "nofilter_func_loop_sorters.py", line 65, in sorter_loop_nofilt
    sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_10filt"+srtr+set_name), freq_min = 10,verbose=True,**sorter_params)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 163, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 262, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in mearec_sorters_test/results_10filtKS30k_Q0_F0_A0/spikeinterface_log.json
```

And for pykilosort:
```
Error running pykilosort
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/external/pykilosort.py", line 190, in _run_from_folder
    run(
  File "/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/testpyks/pykilosort/pykilosort/main.py", line 131, in run
    ir.Wrot = get_whitening_matrix(
  File "/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/testpyks/pykilosort/pykilosort/preprocess.py", line 230, in get_whitening_matrix
    Wrot = whiteningLocal(CC, yc, xc, whiteningRange)
  File "/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/testpyks/pykilosort/pykilosort/preprocess.py", line 166, in whiteningLocal
    wrot0 = cp.asnumpy(whiteningFromCovariance(CC[np.ix_(ilocal, ilocal)]))
  File "/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/testpyks/pykilosort/pykilosort/preprocess.py", line 145, in whiteningFromCovariance
    E, D, _ = cp.linalg.svd(CC)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/cupy/linalg/_decomposition.py", line 566, in svd
    gesvd(
  File "cupy_backends/cuda/libs/cusolver.pyx", line 2731, in cupy_backends.cuda.libs.cusolver.dgesvd
  File "cupy_backends/cuda/libs/cusolver.pyx", line 2740, in cupy_backends.cuda.libs.cusolver.dgesvd
  File "cupy_backends/cuda/libs/cusolver.pyx", line 1079, in cupy_backends.cuda.libs.cusolver.check_status
cupy_backends.cuda.libs.cusolver.CUSOLVERError: CUSOLVER_STATUS_EXECUTION_FAILED

Traceback (most recent call last):
  File "nofilter_func_loop_sorters.py", line 101, in <module>
    sorter_loop_nofilt('pyKS','pykilosort',set_list)
  File "nofilter_func_loop_sorters.py", line 67, in sorter_loop_nofilt
    sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_10filt"+srtr+set_name), fshigh = 10,verbose=True,**sorter_params)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 163, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks133/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 262, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in mearec_sorters_test/results_10filtpyKS30k_Q0_F0_A0/spikeinterface_log.json
```

ZoeChen96, 2023-04-29:
The code I used for running the sorters are as follows:
```
import spikeinterface.extractors as se 
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
from pathlib import Path
import warnings
warnings.simplefilter("ignore")
import sys

set_list = ['30k_Q0_F0_A0','20k_Q0_Fsw3_A0','30k_Q0_Fsw3_A0','40k_Q0_Fsw3_A0']
#fs = 30000
#params = ss.get_default_sorter_params("herdingspikes")

base_folder = Path(".")
cmp_method = ("raw_count", "by_unit", "pooled_with_average")
original_stdout = sys.stdout # Save a reference to the original standard output
sorter_params = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
ss.IronClustSorter.set_ironclust_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/ironclust')
ss.Kilosort3Sorter.set_kilosort3_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/Kilosort')

def sorter_loop(srtr,sorter,set_list): #set list: ['1','2','3','3_14','4_13']
    filename = 'new_templates_sorter_loop2.txt'
    acc_list = []
    for i in range(len(set_list)):
        set_name = set_list[i]
        print('processing ' + set_name +'...')
        recording_name = '/imec/other/macaw/projectdata/mearec_datasets/new_templates_datasets/set'+set_name+'/set'+set_name+'_recording.h5'
        #pdb.set_trace()
        recording = se.MEArecRecordingExtractor(recording_name)
        #recording = recording.frame_slice(start_frame=0*fs, end_frame=60*fs)
        sorting_GT = se.MEArecSortingExtractor(recording_name)
        if srtr == ('IC' or 'KS' or 'pyKS'): 
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results"+srtr+set_name), verbose=True,**sorter_params)
        else:
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results"+srtr+set_name), verbose=False)
        comp = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=True,delta_time = 1)
        #comp = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=True)
        tmp_accuracy = comp.get_performance(method = cmp_method[2])['accuracy']
        acc_list.append(tmp_accuracy)
        print('with filter, set = ',set_name,',accuracy = ', tmp_accuracy)
    print('------------------'+srtr+' finish-----------------------')

    with open(filename, "a") as f:    
        sys.stdout = f # Change the standard output to the file we created.   
        print('------------------Results for '+ sorter+'---------------')  
        for i in range(len(set_list)):
            print('with filter, set = ',set_list[i],',accuracy = ', acc_list[i])
    sys.stdout = original_stdout
    return

def sorter_loop_nofilt(srtr,sorter,set_list): #set list: ['1','2','3','3_14','4_13']
    filename = 'new_templates_sorter_60dur.txt'
    acc_list = []
    for i in range(len(set_list)):
        set_name = set_list[i]
        print('processing ' + set_name +'...')
        recording_name = '/imec/other/macaw/projectdata/mearec_datasets/new_templates_datasets/set'+set_name+'/set'+set_name+'_recording.h5'
        #pdb.set_trace()
        recording = se.MEArecRecordingExtractor(recording_name)
        sorting_GT = se.MEArecSortingExtractor(recording_name)
        #recording = recording.frame_slice(start_frame=0*fs, end_frame=60*fs)
        if srtr == 'IC': 
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), filter = False, verbose=False,**sorter_params)
        elif srtr == 'KS':  
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_10filt"+srtr+set_name), freq_min = 10,verbose=True,**sorter_params)
        elif srtr == 'pyKS' :
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_10filt"+srtr+set_name), fshigh = 10,verbose=True,**sorter_params)
        elif srtr == 'HS':
            sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), filter = False, verbose=False)
[... 7 lines cut ...]
        acc_list.append(tmp_accuracy)
        print('10Hz filter,set = ',set_name,',accuracy = ', tmp_accuracy)
    print('------------------'+srtr+' finish-----------------------')

    with open(filename, "a") as f:    
        sys.stdout = f # Change the standard output to the file we created.   
        print('------------------Results for '+ sorter+'---------------')  
        for i in range(len(set_list)):
            print('no filter, set = ',set_list[i],',accuracy = ', acc_list[i])
    sys.stdout = original_stdout
    return
#sorter_loop_nofilt('HS','herdingspikes',set_list)
#sorter_loop('HS','herdingspikes',set_list)
#sorter_loop_nofilt('SC','spykingcircus2',set_list)
#sorter_loop('SC','spykingcircus2',set_list)
#sorter_loop_nofilt('TDC','tridesclous2',set_list)
#sorter_loop('TDC','tridesclous2',set_list)
#sorter_loop_nofilt('IC','ironclust',set_list)
#sorter_loop('IC','ironclust',set_list)

#sorter_loop('KS','kilosort3',set_list)
#sorter_loop('pyKS','pykilosort',set_list)
sorter_loop_nofilt('KS','kilosort3',set_list)
sorter_loop_nofilt('pyKS','pykilosort',set_list)
```

DradeAW, 2023-05-01:
I also wanted to apply myself a filter without Kilosort filtering itself.
Unfortunately it's not possible without modifying the source code underneath.

I set the high-pass filter to 40 Hz, this worked for me (lower values tend to crash for unknown reasons). This value is low enough that I think it's going to be very similar to no filtering if you apply your filter beforehand :)

samuelgarcia, 2023-05-02:
Here an open PR #1418 to skip the kilosort2.5 preprocessing.
This should works.

ZoeChen96, 2023-05-02:
Hi, thanks @DradeAW ！ I will try that.
@samuelgarcia , does it mean that the skipping now only works for Kilosort2.5? Thank you in advance!

samuelgarcia, 2023-05-02:
At the moment yes.
Alessio should implement the same for KS3.

alejoe91, 2023-05-03:
@ZoeChen96 this is not entirely correct. #1418 does not only skips the highpass filter which is hardcoded in KS, but also whitening (which is very important for the algorithm). @samuelgarcia has made some successfull tests to skip KS pre-processing and applying the same preprocessing from SI, but this requires some additional pre-processing on the SI side.

ZoeChen96, 2023-05-03:
Hi @alejoe91 , thanks for the answer! Let me understand better. So now the highpass filter + other preprocessing (e.g. whitening)  of KS can be skipped in SI, but if I only disable the preprocessing, I also disable the whitening. 
That means, if I need to only disable filter, I need to :
```
recording_whiten = whiten(recording)
ss.run_sorter( 'kilosort', recording_whiten, skip_kilosort_preprocessing = True,.....)
```
Is that what you mean?

alejoe91, 2023-05-03:
Exactly! Note that the PR is not merged yet, since we still need to propagate the behavior to the different KS versions and re-build the Docker images..so it might take a couple of weeks. If you want to experiment, you can try to install from SI from that PR!

ZoeChen96, 2023-05-03:
OK, thanks a lot!

closed 2023-05-03 by ZoeChen96