SpikeInterface/spikeinterface #1605: 'WaveformExtractor' object has no attribute '_memory_objects'

vncntprvst, 2023-05-02:
I'm writing a script for spike curation and I'm getting this error when selecting the waveforms: 

```
we_curated = we.select_units(keep_unit_ids, new_folder=None, use_relative_path="waveforms_curated")
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[39], line 2
      1 # we_curated = we.select_units(keep_unit_ids, new_folder="waveforms_cur")
----> 2 we_curated = we.select_units(keep_unit_ids, new_folder=None, use_relative_path="waveforms_curated")

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py:732, in WaveformExtractor.select_units(self, unit_ids, new_folder, use_relative_path)
    730         we._memory_objects = {"wfs_arrays": {}, "sampled_indices": {}}
    731         for unit_id in unit_ids:
--> 732             we._memory_objects["wfs_arrays"][unit_id] = self._memory_objects["wfs_arrays"][unit_id]
    733             we._memory_objects["sampled_indices"][unit_id] = self._memory_objects["sampled_indices"][unit_id]
    735 # finally select extensions data

AttributeError: 'WaveformExtractor' object has no attribute '_memory_objects'
```
I'm using spikeinterface version 0.97.1.

The code before that line runs with no issue:
```
    sparsity = comp_sparsity(we)
    qm = comp_quality_metrics(we, sparsity[0], job_kwargs)
                
    curation_query = f"(amplitude_cutoff < {amplitude_cutoff_thresh}) & (isi_violations_ratio < {isi_violations_ratio_thresh}) & (presence_ratio > {presence_ratio_thresh})"
    print(f"Starting curation using {curation_query}")

    keep_units = qm.query(curation_query)
    keep_unit_ids = keep_units.index.values

    sorting_auto = sorting.select_units(keep_unit_ids)
    print(f"Number of units before curation: {len(sorting.get_unit_ids())}")
    print(f"Number of units after curation: {len(sorting_auto.get_unit_ids())}")
```
THe waveforms are sparse. I've tried loading them in different ways 
`we = si.load_waveforms(rec_folder / 'sorting_pyKS' / 'waveforms-sparse')` or `we = si.load_waveforms(rec_folder / 'results_pyKS' / 'waveforms-sparse',  with_recording=True, sorting=sorting)`, but I get the same result downstream. 

Any idea? 
Thanks

vncntprvst, 2023-05-02:
The same error pop us with dense waveforms.

vncntprvst, 2023-05-02:
I assume this is related to [this code](https://github.com/SpikeInterface/spikeinterface/blob/a8ccf2ca2b7a424efe248270d44bbb8dcb07239f/src/spikeinterface/core/waveform_extractor.py#L102), but I don't know what to change to fix this.

samuelgarcia, 2023-05-02:
Hi Vincent.
Thank you for the report.
If I am correctly:
 * you have a waveform_extractor in folder mode.
 * you select some unit to get another waveform_etxractor bu in memory mode
 * you have third waveform extarctor that select some units on the sconde one
Is it somethign like this ?
There some weak stuff in the code when a WE is going from folder to memory.

vncntprvst, 2023-05-02:
Hi Samuel,
It's only two extractors (unless you count the original extraction?). Here are the steps in order

1. Loading previously extracted waveforms with `we = si.load_waveforms(rec_folder / 'sorting_pyKS' / 'waveforms-sparse')`
2. Computing the `keep_units` index with
```
    sparsity = comp_sparsity(we)
    qm = comp_quality_metrics(we, sparsity[0], job_kwargs)
    curation_query = f"(amplitude_cutoff < {amplitude_cutoff_thresh}) & (isi_violations_ratio < {isi_violations_ratio_thresh}) & (presence_ratio > {presence_ratio_thresh})"
    keep_units = qm.query(curation_query)
    keep_unit_ids = keep_units.index.values
```
3. Apply the index to get waveforms to keep, with `we_curated = we.select_units(keep_unit_ids, new_folder=None, use_relative_path="waveforms_curated")`

alejoe91, 2023-05-02:
Thanks for reporting this Vincent! I'll take a look and add some tests on this behavior! Can you confirm that it works if you set the new_folder?

vncntprvst, 2023-05-02:
Hi Alessio,
Yes, to confirm, it works. 
`we_curated = we.select_units(keep_unit_ids, new_folder= rec_folder / 'sorting_pyKS' / 'waveforms_cur')`
And re-loading waveforms with the curated sorting also works:
`we_curated = si.load_waveforms(rec_folder / 'results_pyKS' / 'waveforms-sparse',  with_recording=True, sorting=sorting_curated)`
I'll use either one for now :) 
Thanks

PaolaGith, 2023-05-03:
Hello,
I am also facing some issues in loading waveform folder.
I performed the spike sorting, extracted the waveforms and applied the curation. Now, on a different computer, I would like to load the waveform folder again and redo the curation, but I am not able to load the folder.
My spikeinterface version is 0.98.0.dev0.

The folder that contains the result of waveform extraction on the current computer is:

> base_folder= '/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M4/Spike_sorting_results_M4/wf_MS_all'

and contains:
params.json	      recording.json  spike_amplitudes	     templates_std.npy
principal_components  recording_info  template_metrics	     unit_locations
quality_metrics       sorting.json    templates_average.npy  waveforms

I tried to load it in several ways and I always get error:

> a=si.load_waveforms(base_folder)

and I get an exception

> Exception: The recording could not be loaded. You can use the `with_recording=False` argument

So I try 

> a=si.load_waveforms(base_folder,with_recording=False)

and I get 

> FileNotFoundError: [Errno 2] No such file or directory: '/mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M4/results_MS/sorter_output/firings.npz'

Which is the path of the computer on which I ran the script to perform spike sorting, wave extraction etc. 
(In the current computer the path of the folder containing the sorter results is: '/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M4/Spike_sorting_results_M4/results_MS')

Lastly I tried

> a=si.load_waveforms(base_folder/'waveforms')

and got

> TypeError: unsupported operand type(s) for /: 'str' and 'str'

I am not sure what I am doing wrong, but I have issues in loading folders or files. Do you have any idea on how to fix the problem?

Thank you!

PaolaGith, 2023-05-05:
Thanks alejoe91 and vncntprvst.

Even if I manually move the recording file in the folder, it is still unable to open it. And, as I said, I get error also if I set with_recording=False.

It still recalls the path of the computer I used to process the data, which is the lab workstation, now I am trying to redo the curation on my personal pc. 
Do you want me to open a new issue?

closed 2023-11-06 by alejoe91