SpikeInterface/spikeinterface #1609: Question: How to enable GPU support for sorters? (e.g. Ironclust)

ZoeChen96, 2023-05-04:
Hi there,

I have the GPU resources + CUDA, so I can run `Kilosort3` well. However, I am also comparing with other sorters, like `Ironclust/Tridesclous2/SpykingCircus2/Herdingspikes`. As the duration is 600 seconds, CPU running is super slow. Can I change some parameters to enable GPU support for these sorters too? 

I did ss.get_default_sorter_params(sorter), and I found:
ironclust has parameter: 'fGpu': True
the other three (Tridesclous2/SpykingCircus2/Herdingspikes) does not have GPU related parameters.

Does it mean only Ironclust can support GPU? And is the way of keep fGpu = True to enable GPU? 
Or if my server has CUDA+ GPU, the GPU is internally used already?
Thank you!

alejoe91, 2023-05-13:
Hi @ZoeChen96 

Indeed, the other sorters ton't have GPU support. Ironclust does support GPU an you have to set `fGpu=True` (not sure if this is default).

ZoeChen96, 2023-05-13:
OK, Thank you very much for answering!

closed 2023-05-13 by ZoeChen96