SpikeInterface/spikeinterface #1614: Spike Interface Custom Format

nadirbt, 2023-05-05:
I am running into an issue with saving my data into a recording extractor. My data comes from a custom chip and is stored in a custom NI TDMS format at first and then converted into a custom h5 format. All the formats that are compatible with SpikeInterface are linked to a particular acquisition system so I was wondering if there is an easy way of transferring my data into a format read by spike interface.

I tried to save my data into a recordingBinaryRecordingExtractor but I could not find a way without first loading it into a NumpyExtractor. I was of course unable to put all that data (300 GB) in memory first. Is there a way of gradually adding channels to the Extractor class without putting everything in memory? I have 4096 channels to analyze.

alejoe91, 2023-05-05:
Hi @nadirbt 

Thanks for the question! My suggestion is to make a custom extractor as an external library, that inherits from SpikeInterface objects.

This is relatively simple, and you can find here a couple of examples for similar devices (that you're probably familiar with :)):
https://git.bsse.ethz.ch/abuccino/belextractors/-/tree/master/belextractors

alejoe91, 2023-06-12:
@nadirbt can I close this?