SpikeInterface/spikeinterface #1618: Loading waveform folder

PaolaGith, 2023-05-08:
Hello,
I am facing some issues in loading waveform folder.
I performed the spike sorting, extracted the waveforms and applied the curation. Now, on a different computer, I would like to load the waveform folder again and redo the curation, but I am not able to load the folder.
My spikeinterface version is 0.98.0.dev0.

The folder that contains the result of waveform extraction on the current computer is:

> base_folder= '/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M4/Spike_sorting_results_M4/wf_MS_all'

and contains:
params.json	      recording.json  spike_amplitudes	     templates_std.npy
principal_components  recording_info  template_metrics	     unit_locations
quality_metrics       sorting.json    templates_average.npy  waveforms

I tried to load it in several ways and I always get error:

> a=si.load_waveforms(base_folder)

and I get an exception

> Exception: The recording could not be loaded. You can use the `with_recording=False` argument

So I try 

> a=si.load_waveforms(base_folder,with_recording=False)

and I get 

> FileNotFoundError: [Errno 2] No such file or directory: '/mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M4/results_MS/sorter_output/firings.npz'

Which is the path of the computer on which I ran the script to perform spike sorting, wave extraction etc. 
(In the current computer the path of the folder containing the sorter results is: '/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M4/Spike_sorting_results_M4/results_MS')

Lastly I tried

> a=si.load_waveforms(base_folder/'waveforms')

and got

> TypeError: unsupported operand type(s) for /: 'str' and 'str'

I am not sure what I am doing wrong, but I have issues in loading folders or files. Do you have any idea on how to fix the problem?

Thank you!

_Originally posted by @PaolaGith in https://github.com/SpikeInterface/spikeinterface/issues/1605#issuecomment-1532646651_

alejoe91, 2023-05-12:
Hi @PaolaGith 

In order to make the waveform extractor "portable" it needs to have access to the `sorting` object, which is required.

This is what you can do (assuming the folder you plan to copy is called `processed`):
```
# save the sorting object to the `processed` folder
sorting = sorting.save(folder="path-to-processed/sorting")

# extract waveforms inside processed, using relative paths
we = si.extract_waveforms(recording, sorting, folder="path-to-processed/sorting", use_relative_path=True, **job_kwargs)
```

Now you can copy the `processed` folder to the other computer and load the waveforms without the recording:
```
we_loaded = si.load_waveforms(`processed`, with_recording=False)
```

I'll update the docs to describe this portability feature! Thanks for opening the issue and let me know if the proposed solution works!

alejoe91, 2023-05-12:
See #1634

PaolaGith, 2023-05-13:
Dear alejoe91,
thanks a lot for your answer and for addressing this issue!
I will definitely save the recordings in this way for my next data.

For my current data, which have been already sorted, the sorting object was saved automatically in this way

`
sorting_MS = ss.run_sorter('mountainsort4', recording_saved, output_folder=base_folder/'results_MS', verbose=False, filter=False, whiten=True)`

and the firings.npz file is in the saved folder:

`'/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M1/Spike_sorting_results_M1/results_MS/sorter_output'`

The waveforms instead are calculated and saved in this way:

`we_all = si.extract_waveforms(recording_saved, sorting_MS, folder=base_folder / "wf_MS_all",                   max_spikes_per_unit=None, progress_bar=True, load_if_exists=True)`

And the path is:

`'/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M4/Spike_sorting_results_M4/wf_MS_all'`

So in order to be able to load my current data should I merge the two folders?

I am a bit confused, sorry...

Anyway, I don't have any issue if I use the same computer I used to perform the spike sorting and to extract the waveforms.

alejoe91, 2023-05-13:
The important things are: 

1. The relative paths between the sorter folder and the waveforms folder are maintained (in your case it's the case if you move the  `Spike_sorting_results_M4` folder
2. when you extract waveforms you use the `use_relative_path=True`

So you need to re-extract waveforms adding the `use_relative_path=True` argument and then you should be anle to reload the waveforms in the other computer, but you have to move the `Spike_sorting_results_M1`, not only the `wf_MS_all` folder.

Is this clearer? :)

PaolaGith, 2023-05-15:
Thanks!
So I am sure that I am missing something , but when I try to re-extract the waveforms I still get error:

I need to run this command:

`we_all_2 = si.extract_waveforms(recording_saved, sorting_MS, folder=base_folder / "wf_MS_all2", max_spikes_per_unit=None, progress_bar=True, load_if_exists=True, use_relative_path=True)`

So I load the folders that contain the preprocessed data and the sorting result:

```
sorting_folder = '/mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M4/results_MS'
base_folder = '/mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M4'
sorting_MS=ss.read_sorter_folder(sorting_folder)
recording_saved = si.load_extractor(base_folder/'preprocessed')
```
But I still get an error when trying to load the folder that contain the preprocessed data:

```
TypeError                                 Traceback (most recent call last)
Cell In[20], line 6
      2 base_folder = '/mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M4'
      5 sorting_MS=ss.read_sorter_folder(sorting_folder)
----> 6 recording_saved = si.load_extractor(base_folder/'preprocessed')

TypeError: unsupported operand type(s) for /: 'str' and 'str'

```

alejoe91, 2023-05-15:
The `/` syntax needs pathlib Paths:
```
from pathlib import Path

base_folder = Path('/mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M4')
recording_saved = si.load_extractor(base_folder / 'preprocessed')
```

alejoe91, 2023-05-16:
HI Paola,

The `ss.read_sorter_folder()` is in the `sorters` module and it expects to find the recording associated with it.

You can load the mountainsort object with:
```
import spikeinterface as si

sorting_ms4 = si.read_npz_sorting(base_folder/ 'results_MS' / 'sorter_output / 'firings.npz')
```

Since reading from a spike sorted folder is sorter-specific, I strongly recommend saving the sorting object to a SpikeInterface folder after spike sorting:

```
sorting_ms4 = ss.run_sorter(...)

sorting_ms4 = sorting_ms4.save(folder=base_folder/ 'sorting')
```

Then you can simply reload the object with:
```
sorting_loaded = si.load_extractor(base_folder/ 'sorting')
```

PaolaGith, 2023-05-18:
Thanks a lot for your precious help!

alejoe91, 2023-06-12:
@PaolaGith this seems solved. Closing