SpikeInterface/spikeinterface #1619: Loading sorting from pickle takes an absurd amount of time

DradeAW, 2023-05-08:
Hi,

I'm saving my sorting's provenance using the pickle format, and loading them from pickle.
I noticed that when the sorting object becomes a bit complicated (with multiple parents), the loading takes an incredibly absurd amount of time (4 minutes and 56 seconds) to load (writing is relatively fast).

Here is a sample of the code:

```python
sorting.dump_to_pickle(file_path="ks2_best.pkl", include_properties=True)
# Do stuff
sorting = si.load_extractor("ks2_best.pkl")
```

Here is the result of `sorting.to_dict()`:

```
sorting.to_dict()
{'class': 'spikeinterface.core.unitsaggregationsorting.UnitsAggregationSorting', 'module': 'spikeinterface', 'kwargs': {'sorting_list': [AlignSortingExtractor: 20 units - 1 segments - 30.0kHz, UnitsSelectionSorting: 46 units - 1 segments - 30.0kHz], 'renamed_unit_ids': [4, 6, 7, 12, 14, 16, 17, 19, 22, 28, 30, 31, 34, 36, 37, 38, 40, 53, 58, 60, 8, 13, 21, 23, 24, 49, 54, 55, 56, 61, 63, 64, 0, 1, 2, 3, 5, 9, 10, 11, 15, 20, 25, 26, 27, 29, 32, 33, 35, 39, 41, 42, 43, 44, 45, 46, 47, 48, 50, 51, 52, 57, 59, 62, 65, 66]}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {}, 'properties': {}}
>>> sorting.dump_to_json("ks2_best.json")
```

I also did a `sorting.dump_to_json`, but the file is 3,286 lines long. I can send it via Slack if helpful.

I don't understand why it's taking that long?
Loading the same `sorting` from json takes between 0.5-1 s.

alejoe91, 2023-05-08:
Thanks for the report.Is there a reason why you're using pkl rather than JSON?

DradeAW, 2023-05-08:
Yep, 2 reasons:

First - I believe with pickle, numpy arrays stays as arrays and not lists. There's no conversion which is convenient.
Second - json creates big files. In this example (which is still a relatively small sorting for my use case), the pickle file is 14.3 kB, whereas the json file is 241.2 kB. For bigger sorting objects, the json file is something like 14 MB ...

DradeAW, 2023-05-08:
Look at all these spaces in the json file. It makes it human readable which is nice, but for complex sorting objects it's a lot of space on the disk (and this is a case with not a lot of parents).

[image]

h-mayorquin, 2023-05-08:
Can you speak more of which format you use and what is your pre-processing pipeline?

You can do a simple profile to see what is the thing that takes that long.

DradeAW, 2023-05-08:
I'm sorry, I don't understand the question?

I have a sorting object which undergoes many steps (I can send you the json dumped file). And at each step, I want to export the provenance (in pickle format).
I notice that every time I add a step, the time taken to export the sorting to pickle format increases (more than linearly). However, this issue doesn't happen with an export to json format.

What's taking long is simply the line `sorting.dump_to_pickle(file_path="ks2_best.pkl", include_properties=True)`

DradeAW, 2023-05-08:
It is not sorter dependent (happens with different spike-sorting algorithms). However, I only tested the `PhySortingExtractor` as the original sorting object.

I sent you the json file (in Slack) of the json file that took 5 minutes to load from pickle. You can see there that there are many steps.

I've done some profiling in Java, but never in Python.
Using `pickle.load` is also absurdly slow, so I'm guessing the problem is that storing the sorting as it is is too slow. That's why I was thinking saving the `to_dict` is probably better (but not as json, I want to keep my numpy arrays as arrays).
Except that `to_dict()` doesn't call `to_dict` on parent sortings. I think that's probably the problem.

samuelgarcia, 2023-05-09:
I think we should put back something like
`rec_or_sorting.to_dict(recusive=False/True)` maybe the picklisation would be faster.

DradeAW, 2023-05-09:
I agree, but then `load_from_pickle` would also need to recursively load all dicts right?

h-mayorquin, 2023-05-14:
I have another question, @DradeAW. 

You say:

> First - I believe with pickle, numpy arrays stays as arrays and not lists. There's no conversion which is convenient.

How is this convenient? Which cases do you care about? Can you elaborate?

DradeAW, 2023-05-14:
It's just to make sure that everything stays exactly the same.
If you convert to a list then reload and cast it to a numpy array, then the dtype might be different.

With pickle, you get smaller files and you make sure that all the stuff is unchanged. It might have no impact, but I do care about the smaller file size for my use case. I don't mind losing the non human-readable aspect, I can always load from pickle and export to json if needed.

h-mayorquin, 2023-05-14:
I agree about the size problem that you pointed out above. I was curios if you had more concrete problems with the representation of arrays as lists. 

Btw, have you tried Zarr for this?

DradeAW, 2023-05-14:
I haven't encountered any problems with the representation of arrays as lists for now. I'm just worried because if at one point on 1 platform there is a problem, this is the kind of thing that can be invisible ...
But nothing concrete :)

I haven't tried it no!
What are the differences between zarr and pickle?

h-mayorquin, 2023-06-05:
Just to confirm before closing this issue, your problems were sorted with https://github.com/SpikeInterface/spikeinterface/pull/1674, @DradeAW ?

Is there anything that we should add as a follow-up.

I also realize that I did not answer you about Zarr. Zarr should be a good format for working on data that is meant to be shared as it allows for working with larger than memory data, good compresion and streaming. One disavantage is that you will need to change your extractor as you will need to work with a zarr extractor instead:

```
mearec_extractor = read_mearec()
zarr_extractor  = mearec_extractor.save_to_zarr()  # Don't remember the name of the method 
...  # Your pipeline
```

Pickle is not usually recommended as an exchange format as its protocol its version dependent but it is usually used to save state and internal caching (it is used by multiprocessing for example). 

I dont know what would work better for your use case. You will get more performance with zarr probably but you will lose your extractor identity if that something that matters.

DradeAW, 2023-06-05:
> Just to confirm before closing this issue, your problems were sorted with #1674, @DradeAW ?

Yep!
I just tested it now, and it's super fast even with very complex sorting objects! (and doesn't take a lot of RAM)
Thank you very much :)
I'm guessing the problem was with the iterative `deepcopy` ^^

Thanks for the Zarr information!
There are 2 purpose of my save: my pipeline is heavy, and I want to save the state of objects at each state. That way if there is a crash somewhere, it can pick up where it left of. Also, I want users to be able to check the provenance and see exactly what happened.

From what you're telling me, pickle seems to be the correct approach. Now with this fix, it's really fast, light-weight, and contains all the necessary information!

closed 2023-06-05 by DradeAW

h-mayorquin, 2023-06-05:
That's amazing! Good luck : )