SpikeInterface/spikeinterface #1627: Error using Phy

guilhemibos, 2023-05-11:
Hi

I'm testing SpikeInterface with some of my data recorded with open ephys (binary data) but I don't manage to load results of any sorters in Phy:
__init__:62          An error has occurred (AssertionError):
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\phylib\io\model.py", line 1433, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\phylib\io\model.py", line 339, in __init__
    self._load_data()
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\phylib\io\model.py", line 404, in _load_data
    self.sparse_templates = self._load_templates()
  File "C:\Users\invibeadmin\.conda\envs\SpikeInterface\Lib\site-packages\phylib\io\model.py", line 701, in _load_templates
    assert data.dtype in (np.float32, np.float64)
AssertionError
QWidget: Must construct a QApplication before a QWidget

I recorded from 163 channels (30 kHz), 32 channels from a linear probe, 64 channels from one utah array, 64 channels from another Utah array, 3 analog channels (eyes):
Preprocessing follows these steps:
1. load data; 
3. I remove the 3 analog channels (163 to 160 channels); 
4. I add the probes which describe the set-up (channel1-32 shank 0,  x=0, y spaced with 50 um; channel 32-96 shanks 1 x=1000 y spaced every 1000 um; channel 97-160 shank 2 x=2000 y spaced every 1000 um)
recording.set_probegroup(probe)
groups = [0] * 32 + [1] * 64 + [2] * 64
recording_wprobe.set_channel_groups(groups=groups)
recording_areas=recording_wprobe.split_by("group")
5. bandpass filter and common_reference only the shank 0 (the 32 channels of the  linear probe)
recording_f = spre.bandpass_filter(recording_areas[0], freq_min=300, freq_max=6000)
recording_cmr = spre.common_reference(recording_f, reference='global', operator='median')
6. I run Kilosort (same issue with other sorters) from docker image qnd extract waveform
sorting_KS = ss.run_sorter("kilosort2",
        recording=recording_cmr,
        output_folder="D:/test/KS",
        docker_image="spikeinterface/kilosort2-compiled-base:latest", verbose=True)
we_KS = si.extract_waveforms(recording_cmr, sorting_KS, folder=base_folder / 'SpikeInterface' / 'KS'/ 'waveforms_folder', overwrite=True)
7. I export to phy:
export_to_phy(we_KS, 'D:/test/KS/phy_folder_for_KS', compute_pc_features=True, compute_amplitudes=True)
output:
Run:
phy template-gui D:\test\KS\phy_folder_for_KS\params.py
which ends up in the previously described error.

Any idea?

Guilhem

alejoe91, 2023-05-12:
@guilhemibos can you launch Phy on other data? 

The problem seems to be due to the Phy installation itself, rather than SpikeInterface. Can you try to post on the Phy repo?

PaolaGith, 2023-05-13:
Hi,
I have the same issue. I can export correctly in the folder of PHY and 
`Run: phy template-gui /mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M1/Spike_sorting_results_M1/params.py`
But I don't get to visualize the data.

However I don't get any error either...

zm711, 2023-06-09:
I wonder if the issue is that open ephys is int16
https://github.com/NeuralEnsemble/python-neo/blob/4d4eb85e459c7e35cc1d43ce423fd58907fb3002/neo/rawio/openephysbinaryrawio.py#LL539C21-L539C53
So when waveforms are extracted without passing a `dtype` it stays `int16`
https://github.com/SpikeInterface/spikeinterface/blob/6c5c0b89791dd943b1c77393e4074f5f51c5e6ee/src/spikeinterface/core/waveform_extractor.py#L1470-L1471
Then when exporting to Phy in `export_to_phy` it would retain the `int16`
https://github.com/SpikeInterface/spikeinterface/blob/6c5c0b89791dd943b1c77393e4074f5f51c5e6ee/src/spikeinterface/exporters/to_phy.py#L158
And the error is being caused because `Phy` requires either np.float32 or np.float64 as you can seen from your error.
@guilhemibos if that hasn't been fixed could you open the `templates.npy` file with an `np.load` and see what the datatype is?

mandarmp, 2023-06-16:
> Hi, I have the same issue. I can export correctly in the folder of PHY and `Run: phy template-gui /mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M1/Spike_sorting_results_M1/params.py` But I don't get to visualize the data.
> 
> However I don't get any error either...

Same problem here. I get to export to phy ..but no visualizations and no error.

closed 2023-07-11 by alejoe91