SpikeInterface/spikeinterface #1636: error with Sliding refractory period violations (sliding_rp_violations)

sachuriga283, 2023-05-14:
I have running the mose recently released spikeinterface. But I encountered an error. Would you please have a look at the problem.
It appears like:

sliding_rp_violations=sqm.compute_sliding_rp_violations(we,min_spikes=0, bin_size_ms=0.25, window_size_s=1, exclude_ref_period_below_ms=0.5, max_ref_period_ms=10, contamination_values=None)
TypeError: compute_sliding_rp_violations() got an unexpected keyword argument 'min_spikes'

alejoe91, 2023-05-14:
Hi, it's weird because the `min_spikes` argument should be there: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/qualitymetrics/misc_metrics.py#L386

Can you post the full error?

sachuriga283, 2023-05-14:
[ks3_20230508.txt](https://github.com/SpikeInterface/spikeinterface/files/11472399/ks3_20230508.txt)
[ks3_f_probe_20230508_expand_edition.txt](https://github.com/SpikeInterface/spikeinterface/files/11472400/ks3_f_probe_20230508_expand_edition.txt)

hi here is the .py and .log file

sachuriga283, 2023-05-15:
hi,
[misc_metrics.txt](https://github.com/SpikeInterface/spikeinterface/files/11477273/misc_metrics.txt)

Actually I found that codes are quite different from mine installed version vs yours.
pip show spikeinterface is showing the version of 0.97, would you have any idea that how this happend.

alejoe91, 2023-05-16:
Hi,

Can you make sure you get the latest version: `pip install -U spikeinterface`

Then you can inspect the correct signature for that version in `ipython` or a `jupyter notebook` with:

```
sqm.compute_sliding_rp_violations?
```

alejoe91, 2023-06-12:
@sachuriga283 did this work?

sachuriga283, 2023-06-12:
hi，I just removed the min_spike arguiment and it worked. I didn`t test
further.

alejoe91, 2023-06-12:
ok, let me close for now then. Feel fre to open another issue for other problems!