SpikeInterface/spikeinterface #1637: Error with Spykingcircus2

ZoeChen96, 2023-05-14:
Hello, I am trying to run different spike sorters with MEArec dataset. Pykilosort/kilosort/Herdingspikes/Ironclust works well, but with Spykingcircus2, I encouter the error:
```
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/si_based_sorters/spyking_circus2.py", line 111, in _run_from_folder
    labels, peak_labels = find_cluster_from_peaks(recording_f, selected_peaks, method='random_projections',
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sortingcomponents/clustering/main.py", line 39, in find_cluster_from_peaks
    labels, peak_labels = method_class.main_function(recording, peaks, params)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sortingcomponents/clustering/random_projections.py", line 114, in main_function
    clustering = hdbscan.hdbscan(hdbscan_data, **d['hdbscan_kwargs'])
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/hdbscan/hdbscan_.py", line 721, in hdbscan
    X = check_array(X, accept_sparse="csr", force_all_finite=False)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/sklearn/utils/validation.py", line 909, in check_array
    raise ValueError(
ValueError: Found array with 0 sample(s) (shape=(0, 10)) while a minimum of 1 is required.

Traceback (most recent call last):
  File "/imec/other/macaw/chen14/mearec/1.8.0/slurm_jobs/5-quantize-test.py", line 122, in <module>
    
  File "/imec/other/macaw/chen14/mearec/1.8.0/slurm_jobs/5-quantize-test.py", line 100, in sorter_loop_nofilt
    sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("mearec_sorters_test/results_nofilt"+srtr+set_name), delete_output_folder = True,  apply_preprocessing = False, verbose=False)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 163, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 262, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in mearec_sorters_test/results_nofiltSC30k_Q6_F1_15k_A0_dur600_seed1/spikeinterface_log.json
```

How shall I deal with it please? Thank you!

yger, 2023-05-17:
I'll have a look and handle the case, but it seems to me that there are no peaks found in the recording. Can it be the case, or are the data properly sorted by other sorters?

ZoeChen96, 2023-05-17:
@yger Data are properly sorted by kilosort/ironclust/...
But maybe it's no peaks, as I quantized the signals externally to 6 bits. Maybe that's the problem that spykingcircus2 can't deal with highly quantized signals?
I tried 8 bit and above to float point, spykingcircus2 is working correctly.

alejoe91, 2023-11-06:
@ZoeChen96 can you try with the latest release? SC2 was updated quite a but by @yger 

Closing this for now, feel free to reopen if you still have the same issue!

closed 2023-11-06 by alejoe91