SpikeInterface/spikeinterface #1640: container method not working.

sachuriga283, 2023-05-15:
Hi,

The code are running well till today, but after I updaeted to 0.97.1version suddenly apears the error below, does anyone know what cause the problem?

Sampling frequency (Hz): 30000.0
Singularity: pulling image spikeinterface/kilosort3-compiled-base
singularity pull --name kilosort3-compiled-base.sif docker://spikeinterface/kilosort3-compiled-base
[33mWARNING:[0m DEPRECATED USAGE: Environment variable SINGULARITY_PULLFOLDER will not be supported in the future, use APPTAINER_PULLFOLDER instead
INFO:    Using cached SIF image
[33mWARNING:[0m DEPRECATED USAGE: Environment variable SINGULARITY_PULLFOLDER will not be supported in the future, use APPTAINER_PULLFOLDER instead
INFO:    Using cached SIF image
kilosort3-compiled-base.sif
Starting container
Installing spikeinterface==0.97.0 in spikeinterface/kilosort3-compiled-base
Installing extra requirements: ['neo']
Running kilosort3 sorter inside spikeinterface/kilosort3-compiled-base
Stopping container
Traceback (most recent call last):
  File "/cluster/home/sachur/Overlap_project/code/Python/OpenEphysForm/ks3_f_probe_20230508_Backup.py", line 148, in <module>
    sorting = ss.run_kilosort3(
  File "/cluster/home/sachur/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 691, in run_kilosort3
    return run_sorter('kilosort3', *args, **kwargs)
  File "/cluster/home/sachur/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 137, in run_sorter
    return run_sorter_container(
  File "/cluster/home/sachur/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 578, in run_sorter_container
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/cluster/home/sachur/Overlap_project/SpkSorting/RawDate/in_container_sorter_script.py", line 3, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'

alejoe91, 2023-05-16:
Hi, did it work before? 

The problem seems to be that spikeinterface fails to install inside the container.. If you installed version `0.97.1`, the same version should be installed in the container, but you can see from the log that version `0.97.0` is attempted to be installed. Not sure why!

Did you also updgrade singularity? Can you share the entire code to reproduice the issue?

DradeAW, 2023-05-19:
If I remember correctly, I had the same issue with an older version of Singularity.

It disappeared when I installed the latest version3

sachuriga283, 2023-05-22:
hi

Thanks all, I upgraded the singularity, but the problem is still there.
Below is my environment. it based on the computer group running on Linux system.
And I will phaste the code in other comment.

absl-py @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/abslpy/absl-py-1.4.0
alabaster @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/alabaster/alabaster-0.7.12
appdirs @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/appdirs/appdirs-1.4.4
asciitree==0.3.3
asn1crypto @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/asn1crypto/asn1crypto-1.5.1
astor @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/astor/astor-0.8.1
astunparse @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/astunparse/astunparse-1.6.3
atomicwrites @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/atomicwrites/atomicwrites-1.4.0
attrs @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/attrs/attrs-21.4.0
Babel @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/Babel/Babel-2.10.1
backports.entry-points-selectable @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/backportsentry_points_selectable/backports.entry_points_selectable-1.1.1
backports.functools-lru-cache @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/backportsfunctools_lru_cache/backports.functools_lru_cache-1.6.4
bcrypt @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/bcrypt/bcrypt-3.2.2
beniget @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/beniget/beniget-0.3.0
bitstring @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/bitstring/bitstring-3.1.9
blist @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/blist/blist-1.3.6
boto3==1.26.137
botocore==1.29.137
Bottleneck @ file:///tmp/eb_pavlokh/SciPybundle/2022.05/foss-2022a/Bottleneck/Bottleneck-1.3.4
CacheControl @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/CacheControl/CacheControl-0.12.11
cachetools @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/cachetools/cachetools-5.2.1
cachy @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/cachy/cachy-0.3.0
certifi @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/certifi/certifi-2021.10.8
cffi @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/cffi/cffi-1.15.0
chardet @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/chardet/chardet-4.0.0
charset-normalizer @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/charsetnormalizer/charset-normalizer-2.0.12
cleo @ file:///cluster/apps/eb/sources/p/Python/extensions/cleo-0.8.1-py2.py3-none-any.whl
click @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/click/click-8.1.3
clikit @ file:///cluster/apps/eb/sources/p/Python/extensions/clikit-0.6.2-py2.py3-none-any.whl
cloudpickle==2.2.1
colorama @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/colorama/colorama-0.4.4
colorlog==6.7.0
contourpy==1.0.7
crashtest @ file:///cluster/apps/eb/sources/p/Python/extensions/crashtest-0.3.1-py3-none-any.whl
cryptography @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/cryptography/cryptography-37.0.1
cuda-python==12.1.0
cupy==12.0.0
cycler==0.11.0
Cython @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/Cython/Cython-0.29.28
dask==2023.5.0
deap @ file:///tmp/eb_pavlokh/SciPybundle/2022.05/foss-2022a/deap/deap-1.3.1
decorator @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/decorator/decorator-5.1.1
dill @ file:///tmp/eb_pavlokh/dill/0.3.6/GCCcore-11.3.0/dill-0.3.6
distinctipy==1.2.2
distlib @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/distlib/distlib-0.3.4
docopt @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/docopt/docopt-0.6.2
docutils @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/docutils/docutils-0.17.1
ecdsa @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/ecdsa/ecdsa-0.17.0
editables @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/editables/editables-0.3
entrypoints==0.4
fasteners==0.18
fastrlock==0.8.1
filelock @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/filelock/filelock-3.6.0
flake8==6.0.0
flatbuffers @ file:///tmp/eb_pavlokh/flatbuffers/2.0.7/GCCcore-11.3.0/flatbuffers/flatbuffers-2.0.7
flit @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/flit/flit-3.7.1
flit_core @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/flitcore/flit_core-3.7.1
fonttools==4.39.4
fsspec @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/fsspec/fsspec-2022.3.0
future @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/future/future-0.18.2
gast @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/gast/gast-0.4.0
glob2 @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/glob2/glob2-0.7
google-auth @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/googleauth/google-auth-2.16.0
google-auth-oauthlib @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/googleauthoauthlib/google-auth-oauthlib-0.4.6
[... 135 lines cut ...]
tensorboard-plugin-wit @ file:///cluster/apps/eb/sources/t/TensorFlow/extensions/tensorboard_plugin_wit-1.8.1-py3-none-any.whl
tensorflow @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/tensorflow-2.11.0-cp310-cp310-linux_x86_64.whl
tensorflow-estimator @ file:///cluster/apps/eb/sources/t/TensorFlow/extensions/tensorflow_estimator-2.11.0-py2.py3-none-any.whl
[... cut ...]
webencodings @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/webencodings/webencodings-0.5.1
Werkzeug @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/Werkzeug/Werkzeug-2.2.2
wrapt @ file:///tmp/eb_pavlokh/TensorFlow/2.11.0/foss-2022a/wrapt/wrapt-1.14.1
xarray==2022.9.0
xlrd @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/xlrd/xlrd-2.0.1
zarr==2.14.2
zipfile36 @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/zipfile36/zipfile36-0.1.3
zipp @ file:///cluster/work/hrn/build/Python/3.10.4/GCCcore-11.3.0/zipp/zipp-3.8.0

sachuriga283, 2023-05-22:
And here is the code python code.

import spikeinterface as si
import spikeinterface.extractors as se
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp
import spikeinterface.widgets as sw

from spikeinterface.preprocessing import (bandpass_filter, notch_filter, common_reference,
                                          remove_artifacts, preprocesser_dict)

import numpy as np
import os
from pathlib import Path

import warnings
warnings.simplefilter("ignore")

def current_path():
    print("Current working directory before")
    print(os.getcwd())
    print()

os.chdir('/cluster/home/sachur/Overlap_project/SpkSorting/RawDate')
current_path()

num_channels = 64
#sampling_frequency = 20000
gain_to_uV = 0.05
offset_to_uV = 0
#dtype="int16"
#time_axis = 1

base_folder = Path(".")
OpenEphys_file_path = "2023-05-11_18-23-06"
current_file_name_path = os.path.basename(OpenEphys_file_path)
current_file_name = str(current_file_name_path)

stream_name='Record Node 106#Acquisition_Board-100.Rhythm Data'
recording = se.read_openephys(OpenEphys_file_path,stream_name=stream_name)

os.chdir('/cluster/home/sachur/Overlap_project/SpkSorting/RawDate')

recording.annotate(is_filtered=False)

channel_ids = recording.get_channel_ids()
fs = recording.get_sampling_frequency()
num_chan = recording.get_num_channels()
num_segments = recording.get_num_segments()

print(f'Channel ids: {channel_ids}')
print(f'Sampling frequency: {fs}')
print(f'Number of channels: {num_chan}')
print(f"Number of segments: {num_segments}")
trace_snippet = recording.get_traces(start_frame=int(fs*0), end_frame=int(fs*2))
print('Traces shape:', trace_snippet.shape)

import probeinterface as pi
#from probeinterface import plotting

manufacturer = 'cambridgeneurotech'
probe_name = 'ASSY-236-F'

probe = pi.get_probe(manufacturer, probe_name)
print(probe)

#probe.wiring_to_device('cambridgeneurotech_mini-amp-64')
#map channels to device indices
mapping_to_device = [
 # connector J2 TOP
        41, 39, 38, 37, 35, 34, 33, 32, 29, 30, 28, 26, 25, 24, 22, 20,
        46, 45, 44, 43, 42, 40, 36, 31, 27, 23, 21, 18, 19, 17, 16, 14,
        # connector J1 BOTTOM
        55, 53, 54, 52, 51, 50, 49, 48, 47, 15, 13, 12, 11, 9, 10, 8,
        63, 62, 61, 60, 59, 58, 57, 56, 7, 6, 5, 4, 3, 2, 1, 0
]

probe.set_device_channel_indices(mapping_to_device)
probe.to_dataframe(complete=True).loc[:, ["contact_ids", "shank_ids","device_channel_indices"]]
probegroup=pi.ProbeGroup()
probegroup.add_probe(probe)

pi.write_prb(f"{probe_name}.prb",probegroup,group_mode="by_shank")

recording_prb=recording
recording_prb = recording.set_probe(probe, group_mode="by_shank")

print(f'Original channels:{recording.get_channel_ids()}')
print(f'Channels after loading the probe file: {recording_prb.get_channel_ids()}')
print(f'Channel groups after loading the probe file: {recording_prb.get_channel_groups()}')
print("Properties before loading the probe:", list(recording.get_property_keys()))
print("Properties after loading the probe:", list(recording_prb.get_property_keys()))

#brain_area_property_values = ['CA1']*32 + ['CA3']*32
#print(brain_area_property_values)
#recording_prb.set_property(key='brain_area', values=brain_area_property_values)
recording_prb.set_property(key='quality', values=["good"]*(recording_prb.get_num_channels() - 3), ids=recording_prb.get_channel_ids()[:-3])

recording_prb.get_property("quality")
recording_prb.annotate(description="Sachuriga_test1")
print(recording_prb.get_annotation_keys())

recording_cmr=recording_prb
recording_f = bandpass_filter(recording_prb,
                              freq_min=300,
                              freq_max=6000)

recording_cmr = common_reference(recording_f,
                                 reference='global',
                                 operator='median')

fs = recording_cmr.get_sampling_frequency()

recording_sub = recording_cmr
job_kwargs = dict(n_jobs=10,
                  chunk_duration="1s",
                  progress_bar=True)

preprocessed=current_file_name + "_" + "preprocessed"
recording_saved = recording_sub.save(folder=base_folder / preprocessed, **job_kwargs)

print(f'Cached channels ids: {recording_saved.get_channel_ids()}')
print(f'Channel groups after caching: {recording_saved.get_channel_groups()}')

recording_loaded = si.load_extractor(base_folder / preprocessed)

print("Properties after adding custom properties:", list(recording_prb.get_property_keys()))
print(recording_prb._properties.keys())

print(recording_f)
print("Duration (minutes):", recording_f.get_total_duration() / 60)
print("Num. channels:", recording_f.get_num_channels())
print("Sampling frequency (Hz):", recording_f.get_sampling_frequency())

sorter_params = job_kwargs
kilosort3=current_file_name + "_" + "kilosort3"

sorter_params = dict(n_jobs=12,
                     chunk_duration="1s",
                     progress_bar=True,
                     minFR=0.1,
                     minfr_goodchannels=0.1
                     )
#sorting=ss.run_sorter(sorter_name="tridesclous", recording=recording_saved, detect_threshold=4)

sorting = ss.run_kilosort3(recording=recording_saved,
    output_folder=base_folder / kilosort3,
    verbose=True,
    singularity_image=True,
    **sorter_params)

sachuriga283, 2023-05-24:
The problem might be due to singularity is not excluding the home lib, thus the all the thing in the container was shaded by that. yet I don't have any solution to solve this. Would somebody have any idea?

alejoe91, 2023-05-24:
I see, I have mainly used Docker, but I think that @DradeAW had similar issues. Maybe he could help?

alejoe91, 2023-06-14:
Can you paste the full error?

DradeAW, 2023-06-14:
The fix for me was to install the latest version of Go (and not the one recommended in the documentation!)
Then install the latest version of singularity
Then install singularity in the Python environment.
Then installing `cuda-python` (for cuda-based sorters)

sachuriga283, 2023-06-14:
Thanks all, I will try and report back.

rbedfordwork, 2023-06-14:
Here's my full error:

SpikeSortingError                         Traceback (most recent call last)
Cell In[22], line 1
----> 1 sorting = si.run_sorter('kilosort3',rec_clean,output_folder='kilosort3',verbose=True,**sorter_params,singularity_image='/public/singularity/containers/spikesorters/kilosort3.sif')

File ~/local/mamba/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~/local/mamba/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:583, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    581 if run_error:
    582     if raise_error:
--> 583         raise SpikeSortingError(
    584             f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    585 else:
    586     if with_output:

SpikeSortingError: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/lmb/home/rbedford/Documents/testsort/in_container_sorter_script.py", line 3, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'

alejoe91, 2023-06-14:
Thanks, so something is wrong when it attempts to install spikeinterface in the container. What is the output before the error? Also, what version of spikeinterface and singularity are you using? We had some troubles with older versions of singularity

DradeAW, 2023-06-14:
Did you set the environment variable `SPIKEINTERFACE_DEV_PATH`?

I had the same problem until I set that variable to point to the repo :)

DradeAW, 2023-06-14:
I think you have the wrong version of Singularity. Here is my output:

```bash
$ singularity --version
singularity-ce version 3.11.1
```

rbedfordwork, 2023-06-15:
To follow this up: I've spent quite a while troubleshooting this now and have found a couple sources for the error. If I enter the container myself and run:

'pip install --upgrade --no-input spikeinterface[full]==0.97.1'

I get:

'ERROR: Could not install packages due to an EnvironmentError: [Errno 30] Read-only file system'

So it definitely seems like it's failing to install spikeinterface from within the container. I will chat to my sysadmin about our version of singularity and how to allow the container to install software.

Secondly I previously had an issue where the container was finding my local installs of python in my home directory, and I think it was confusing this with its own python installation. I had some conda environments stored under $HOME/local, and changed this to $HOME/.local, and it doesn't seem to find them under the hidden directory. Thought I'd mention this just so people know having conda environments stored like this could cause the container to not use its own python.

An honest question here, just as I'm fairly new to using containers: my understanding is that containers should include all the necessary dependencies for an application pre-installed and ready to run. Why is it then that all the spikeinterface sorting containers on dockerhub don't include spikeinterface preinstalled? Would it be that tricky to have si_env included in the image? Is this so that the latest updates can be installed without having to reupload the docker image? Forgive me for the newbie question, it just seems to me that having a container ready to go as is is kind of the purpose of using containers in the first place.

alejoe91, 2023-06-15:
> An honest question here, just as I'm fairly new to using containers: my understanding is that containers should include all the necessary dependencies for an application pre-installed and ready to [...]

Very honest question! Yes indeed the reason is to decouple the development of individual sorters (each image should "fix" a sorter) to the development of SpikeInterface. SpikeInterface is still slowly changing, so if we installed it directly in the images we would need to update them quite often. Instead we prefer to install the current SI version in the container.

rbedfordwork, 2023-06-15:
That makes sense thanks for explaining!

Just wanted to add for now I'm using the following workaround: I'm building the container using apptainer with the following .def file (substitute any sorter):

```
Bootstrap: docker
From: spikeinterface/klusta-base:latest

%post
pip install spikeinterface[full]==0.95.1
```

This works well and I guess you could just update the definition and rebuild whenever you want a new version of spikeinterface. Hope this helps anyone else who runs into the same issue.

alejoe91, 2023-11-06:
Should be fixed by #2053

(Thanks to @mhhennig)

closed 2023-11-06 by alejoe91