SpikeInterface/spikeinterface #1653: Error '@isSource' in core.py

FlorianJanssen, 2023-05-16:
Hello everyone,

I'm new to this forum and have very little experience with programming in general. Still, I hope I can describe my issue and find someone who might be able to help. Thank you very much in advance.
When I try to run a spike sorting script that makes use of core.py it crashes after a few seconds and gives an error message. It reads:
KeyError: '@isSource' and occurs in line 270 of the core.py code. Unfortunately, I have no idea what causes this error.

If anyone has an idea about why it might occur and how to fix it, I would be very grateful. Thank you.
Kind regards, Florian.

alejoe91, 2023-05-16:
Hi @FlorianJanssen, 

Don't worry, we've all learnt programming along the way :)

There is no `core.py` in SpikeInterface so I suspect that the error is coming from somewhere else...
Can you share your code and the full error? 
Are you using the latest spikeinterface version? (Try `pip list` from terminal)

Cheers
Alessio

FlorianJanssen, 2023-05-17:
Hi @alejoe91 ,

thank you for your reply. It is actually my supervisor's script, in which this bug occurs. He wrote this spikesorting script that can be run in Spyder (Python 3.6). The script makes use of core.py and the error occurs in there.
My supervisor has his code here on github: https://github.com/Aleman-Z/spikesorting
The bugging script is called "run_tetrodes.py".

The output is the following:

Hello World!
/home/samanta/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/spikeextractors/extractors/openephysextractors/openephysextractors.py:16: UserWarning: pyopenephys>=1.1.2 should be installed. Support for older versions will be removed in future releases. Install with:

 pip install --upgrade pyopenephys

  warnings.warn("pyopenephys>=1.1.2 should be installed. Support for older versions will be removed in "
Reading JSON
JSON not found. Running consensus in all tetrodes
Setting IRONCLUST_PATH environment variable for subprocess calls to: /home/samanta/Documents/SpikeSorting/ironclust
Loading Open-Ephys: reading settings.xml...
Traceback (most recent call last):

  File "<ipython-input-1-97d1888f777b>", line 50, in <module>
    sorter.auto(tetrode)

  File "/home/samanta/Documents/SpikeSorting/spikesorting/sorter.py", line 56, in auto
    recording = se.OpenEphysRecordingExtractor(recording_folder)

  File "/home/samanta/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/spikeextractors/extractors/openephysextractors/openephysextractors.py", line 40, in __init__
    self._fileobj = pyopenephys.File(folder_path)

  File "/home/samanta/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/pyopenephys/core.py", line 144, in __init__
    self._experiments.append(Experiment(self._absolute_foldername, id, self))

  File "/home/samanta/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/pyopenephys/core.py", line 184, in __init__
    self._read_settings(id)

  File "/home/samanta/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/pyopenephys/core.py", line 270, in _read_settings
    if 'CHANNEL_INFO' in processor.keys() and processor['@isSource'] == '1':

KeyError: '@isSource'

closed 2023-05-24 by alejoe91

FlorianJanssen, 2023-05-27:
Alright, thank you very much for your time and help!