SpikeInterface/spikeinterface #1662: stream_names not found during sorting

shunnnli, 2023-05-18:
Hi, 

I'm running NP2 recording using spikeinterface. I've been following the neuropixel notebook provided in the documentation and everything runs smoothly until spike sorting, where theres an error saying that it can't find imec0.ap as stream_names is empty. However, when I explicitly called stream_names, it outputs correctly. 

Thanks in advance for the help!
[image]

samuelgarcia, 2023-05-22:
Hi.
Sometimes reading a particular inside the docker have issue.
We need to fix that.
An easy fix is to save the recording to binary outside the docker itself because kilosort3 will need a binary anyway.
Try something like this.

```python
rec_binary = rec.save(folder='/a/tmp/folder', n_jobs=-1, progressbar=True, chunk_duration='1s')
sorting = run_sorter('kilosort3', rec_binary, ...)
```

alejoe91, 2023-05-22:
Thanks @samuelgarcia 

I agree we need to debug this internally and maybe even running tests about reading an extractor in a docker image after mapping the paths!

samuelgarcia, 2023-05-22:
@alejoe91  : what we should do is to call the setup_recording outside the docker and run_from_folder inside the docker.
It was my plan long time ago but never find time to do this quite big refactoring.

LaurenzMuessig, 2023-05-22:
Hi guys,

It seems I am also now getting this error after updating to the latest version of spikeinterface when trying to sort a concatenated recording (in docker). The binary is written to disk beforehand. Is it possible that this got broken again? Thanks for your help. Cheers

zm711, 2024-04-30:
Has anyone seen this issue continue or can this be closed with the current state of `main`?

zm711, 2024-06-19:
I'll close this for lack of response, but feel free to reopen if this comes up again.

closed 2024-06-19 by zm711