SpikeInterface/spikeinterface #1666: timeout error when spike sorting with container

shunnnli, 2023-05-19:
Hi,

I'm recording using neuropixel 2 probe and running kilosort for each shank on docker image. My code is as follows:
`sorting = si.run_sorter_by_property('kilosort3', rec,
                                 grouping_property='group', 
                                 working_folder= base_folder/'kilosort3_output',
                                 mode_if_folder_exists="overwrite",
                                 docker_image=True, verbose=True, **params_kilosort3)`

However, I got a timeout error that says connection aborted. Attached is the detailed error code. Thanks in advance!!

Starting container
---------------------------------------------------------------------------
TimeoutError                              Traceback (most recent call last)
File ~\.conda\envs\spikeinterface\lib\site-packages\urllib3\connectionpool.py:703, in HTTPConnectionPool.urlopen(self, method, url, body, headers, retries, redirect, assert_same_host, timeout, pool_timeout, release_conn, chunked, body_pos, **response_kw)
    702 # Make the request on the httplib connection object.
--> 703 httplib_response = self._make_request(
    704     conn,
    705     method,
    706     url,
    707     timeout=timeout_obj,
    708     body=body,
    709     headers=headers,
    710     chunked=chunked,
    711 )
    713 # If we're going to release the connection in ``finally:``, then
    714 # the response doesn't need to know about the connection. Otherwise
    715 # it will also try to release it and we'll have a double-release
    716 # mess.

File ~\.conda\envs\spikeinterface\lib\site-packages\urllib3\connectionpool.py:449, in HTTPConnectionPool._make_request(self, conn, method, url, timeout, chunked, **httplib_request_kw)
    445         except BaseException as e:
    446             # Remove the TypeError from the exception chain in
    447             # Python 3 (including for exceptions like SystemExit).
    448             # Otherwise it looks like a bug in the code.
--> 449             six.raise_from(e, None)
    450 except (SocketTimeout, BaseSSLError, SocketError) as e:

File <string>:3, in raise_from(value, from_value)

File ~\.conda\envs\spikeinterface\lib\site-packages\urllib3\connectionpool.py:444, in HTTPConnectionPool._make_request(self, conn, method, url, timeout, chunked, **httplib_request_kw)
    443 try:
--> 444     httplib_response = conn.getresponse()
    445 except BaseException as e:
    446     # Remove the TypeError from the exception chain in
    447     # Python 3 (including for exceptions like SystemExit).
    448     # Otherwise it looks like a bug in the code.

File ~\.conda\envs\spikeinterface\lib\http\client.py:1377, in HTTPConnection.getresponse(self)
   1376 try:
-> 1377     response.begin()
   1378 except ConnectionError:

File ~\.conda\envs\spikeinterface\lib\http\client.py:320, in HTTPResponse.begin(self)
    319 while True:
--> 320     version, status, reason = self._read_status()
    321     if status != CONTINUE:

File ~\.conda\envs\spikeinterface\lib\http\client.py:281, in HTTPResponse._read_status(self)
    280 def _read_status(self):
--> 281     line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
    282     if len(line) > _MAXLINE:

File ~\.conda\envs\spikeinterface\lib\site-packages\docker\transport\npipesocket.py:224, in NpipeFileIOBase.readinto(self, buf)
    223 def readinto(self, buf):
--> 224     return self.sock.recv_into(buf)

File ~\.conda\envs\spikeinterface\lib\site-packages\docker\transport\npipesocket.py:25, in check_closed.<locals>.wrapped(self, *args, **kwargs)
     22     raise RuntimeError(
     23         'Can not reuse socket after connection was closed.'
[... 235 lines cut ...]
    586 send_kwargs.update(settings)
--> 587 resp = self.send(prep, **send_kwargs)
    589 return resp

File ~\.conda\envs\spikeinterface\lib\site-packages\requests\sessions.py:701, in Session.send(self, request, **kwargs)
    698 start = preferred_clock()
    700 # Send the request
--> 701 r = adapter.send(request, **kwargs)
    703 # Total elapsed time of the request (approximately)
    704 elapsed = preferred_clock() - start

File ~\.conda\envs\spikeinterface\lib\site-packages\requests\adapters.py:502, in HTTPAdapter.send(self, request, stream, timeout, verify, cert, proxies)
    487     resp = conn.urlopen(
    488         method=request.method,
    489         url=url,
   (...)
    498         chunked=chunked,
    499     )
    501 except (ProtocolError, OSError) as err:
--> 502     raise ConnectionError(err, request=request)
    504 except MaxRetryError as e:
    505     if isinstance(e.reason, ConnectTimeoutError):
    506         # TODO: Remove this in 3.0.0: see #2811

ConnectionError: ('Connection aborted.', TimeoutError())

alejoe91, 2023-05-19:
@shunnnli does this error only appear if you sort by property?

shunnnli, 2023-05-19:
No, I've tried run_sorter as well and it also returns the same result

alejoe91, 2023-05-19:
Ok, can you make sure you followed the instructions here?
https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#running-sorters-in-docker-singularity-containers

Basically you need to have Docker desktop installed, an account on dockerhub, and being logged in (for Windows)

shunnnli, 2023-05-21:
Yes, as you can see from my previous error (issue #1662), I can start enter the container normally.

shunnnli, 2023-05-24:
Hi I restarted my computer and it runs now! But I got the following error: 

Starting container
Installing spikeinterface==0.97.1 in spikeinterface/kilosort2_5-compiled-base
Installing extra requirements: ['neo']
Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base
Stopping container
Starting container
Installing spikeinterface==0.97.1 in spikeinterface/kilosort2_5-compiled-base
Installing extra requirements: ['neo']
Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base
Stopping container
Starting container
Installing spikeinterface==0.97.1 in spikeinterface/kilosort2_5-compiled-base
Installing extra requirements: ['neo']
Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base
Stopping container
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[15], line 2
      1 # Sorting by groups
----> 2 sorting = si.run_sorter_by_property('kilosort2_5', rec,
      3                                  grouping_property='group', 
      4                                  working_folder= base_folder/'kilosort2.5_output',
      5                                  mode_if_folder_exists="overwrite",
      6                                  docker_image=True,verbose=True)

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\launcher.py:119, in run_sorter_by_property(sorter_name, recording, grouping_property, working_folder, mode_if_folder_exists, engine, engine_kwargs, verbose, docker_image, singularity_image, **sorter_params)
    116 assert grouping_property in recording.get_property_keys(), f"The 'grouping_property' {grouping_property} is not " \
    117                                                            f"a recording property!"
    118 recording_dict = recording.split_by(grouping_property)    
--> 119 sorting_output = run_sorters([sorter_name], recording_dict, working_folder,
    120                              mode_if_folder_exists=mode_if_folder_exists,
    121                              engine=engine,
    122                              engine_kwargs=engine_kwargs,
    123                              verbose=verbose,
    124                              with_output=True,
    125                              docker_images={sorter_name: docker_image},
    126                              singularity_images={sorter_name: singularity_image},
    127                              sorter_params={sorter_name: sorter_params})
    129 grouping_property_values = None
    130 sorting_list = []

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\launcher.py:348, in run_sorters(sorter_list, recording_dict_or_list, working_folder, sorter_params, mode_if_folder_exists, engine, engine_kwargs, verbose, with_output, docker_images, singularity_images)
    343     print(f'Warning!! With engine="{engine}" you cannot have directly output results\n'
    344           'Use : run_sorters(..., with_output=False)\n'
    345           'And then: results = collect_sorting_outputs(output_folders)')
    346     return
--> 348 results = collect_sorting_outputs(working_folder)
    349 return results

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\launcher.py:413, in collect_sorting_outputs(working_folder)
    408 """Collect results in a working_folder.
    409 
    410 The output is a  dict with double key access results[(rec_name, sorter_name)] of SortingExtractor.
    411 """
    412 results = {}
--> 413 for rec_name, sorter_name, sorting in iter_sorting_output(working_folder):
    414     results[(rec_name, sorter_name)] = sorting
    415 return results

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\launcher.py:403, in iter_sorting_output(working_folder)
    401 for rec_name, sorter_name, output_folder in iter_working_folder(working_folder):
    402     SorterClass = sorter_dict[sorter_name]
--> 403     sorting = SorterClass.get_result_from_folder(output_folder)
    404     yield rec_name, sorter_name, sorting

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\basesorter.py:274, in BaseSorter.get_result_from_folder(cls, output_folder)
    272 log_file = output_folder / 'spikeinterface_log.json'
    273 if not log_file.is_file():
--> 274     raise SpikeSortingError('get result error: the folder do not contain spikeinterface_log.json')
    276 with log_file.open('r', encoding='utf8') as f:
    277     log = json.load(f)

SpikeSortingError: get result error: the folder do not contain spikeinterface_log.json

shunnnli, 2023-05-24:
Hi! This works!! However, the kilosort has been running for 3 hours with no sign of finishing. I'm willing to stop and rerun for all shanks if you have any suggestions!

shunnnli, 2023-05-25:
It's still running after 24 hours haha :(

[image]

alejoe91, 2023-05-25:
Mmmm...it's not supposed to be taking that much! How long is the recording?

alejoe91, 2023-05-25:
Can you also check if the GPU is beong used?

alejoe91, 2023-05-25:
you should have an NVIDIA application or somtehing, but I'm not 100% sure

shunnnli, 2023-05-25:
So I also re-create the timeout error: I stopped the kernal, shut it down, deactivate conda; then I reactivate conda and re-run the code, now running sorter on single shank would show me timeout error as above

alejoe91, 2023-05-25:
Not really sure what's happening then..it might be due to a problem with the Docker installation itself. Are you able to try the analysis on a Linux machine?

shunnnli, 2023-05-25:
I can try install it in Linux. Is there any sample output for kilosort after printing 'Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base'? I just want to know which step the code got stuck on

alejoe91, 2023-05-25:
You should have a `sorter_output` folder in `base_folder/'kilosort3_output_shank0'` with the files that Kilosort is (supposedly) creating. What is the content of that?

shunnnli, 2023-05-25:
Its a chanMap.mat file and a recording.dat file
[image]

bagibence, 2023-05-31:
Hi! I ran into the same thing and was able to fix it by updating the docker library to 6.1.2

See the relevant issue and PR here: https://github.com/docker/docker-py/issues/3124

musall, 2023-07-10:
Unfortunately, we are having the same issue running Docker under Windows with larger datasets. The sorter runs indefinitely (although uses large CPU and memory resources) and only creates chanMap.mat and recording.dat but doesn't start creating temp_wh.dat

We are running docker 6.1.3. so I don't think updating fully resolves this problem (currently rerunning it with downgraded 6.1.2 to make sure).

Does Docker create a log file that we can use to identify where the code gets stuck?

alemessurier, 2023-11-30:
I'm having the same issue, also running on windows locally, for a ~90 gb concatenated neuropixels 1.0 recording. I have docker 6.1.3. Has anyone found a solution or work around?

alemessurier, 2023-11-30:
not sure if it's related but I'm also running into this memory overflow error https://github.com/SpikeInterface/spikeinterface/issues/2201 when concatenating, so I changed the dtype of the recordings from int16 to int64 and I no longer get the warning but the print statement is still telling me the recording is much smaller than the actual size: 

> CommonReferenceRecording: 383 channels - 30.0kHz - 1 segments - 58,042,836 samples 
>                           1,934.76s (32.25 minutes) - float64 dtype - 1.63 GiB

mohebial, 2024-02-22:
I'm experiencing a similar issue where the Docker has been running for more than 24 hours, but there's no progress. It only created two files, chanMap.mat, and recording.dat, and then appeared to be idle without any error messages. I've checked my GPU and CPU, but they're also idle. The recording size is around 136GB (npx 1.0), and I've been able to sort it using a locally installed KS3. However, Docker's KS3 seems to be running indefinitely, and I would love to set up my workflow on Docker. Do you have any suggestions? Thanks!

zm711, 2024-07-19:
Hey All,

I'm going to close this issue for lack of activity, but my hypothesis for people working on this in docker is that the actually writing of the binary file is going slow because of `n_jobs>1`. On Windows multiprocessing can be super finicky. So I would say try setting `n_jobs=1` to remove that from the possibilities.

closed 2024-07-19 by zm711