SpikeInterface/spikeinterface #1670: Broken links in documentation

zm711, 2023-05-23:
https://spikeinterface.readthedocs.io/en/latest/modules_gallery/index.html

is still displayed on the documentation page, but is no longer part of the document structure (ie click on edit on github at the top right and it will raise a 404). It also has a bunch of broken examples. Not sure when or how that section got broken. 

But I just realized this is the tutorial section I thought would be helpful. So fixing this would actually provide the necessary tutorials. Rather than call it `module gallery` though, calling it `tutorials` would be more informative for new users. I also checked downloading the source .py files at the bottom and those still work to download.

zm711, 2023-05-23:
It's just a point issue with the sphinx stuff. The folder is stored in the examples/modules_gallery folder and it is trying to find the link in the doc/modules_gallery instead. I just haven't found the file to fix to point it in the correct direction.

zm711, 2023-05-24:
@alejoe91 Sounds good. I don't think I have the bandwidth this week, but I'll start messing around and seeing if I can try a couple things soon. :)

Also need to figure out if the examples are failing because of api changes or a linkage issue (I'll try fixing the `.py` and see if the examples fix or stay 'broken'). I'll dig into this more soon too. 

If I get this working are you opposed to changing the name of the section? Gallery implies pictures which someone may or may not care about (it is not clear that I would go to a gallery to gain practice with the api). Examples or tutorials would allow a user to know to go there to practice.

alejoe91, 2023-05-24:
Examples are failing because we have to point to the master version of Neo for the docs! Sorry about that!

alejoe91, 2023-05-25:
> If I get this working are you opposed to changing the name of the section? Gallery implies pictures which someone may or may not care about (it is not clear that I would go to a gallery to gain practice with the api). Examples or tutorials would allow a user to know to go there to practice.

It's called gallery because it uses `sphinx-gallery` eheheh. Not opposed to changing the name if we find a better name!

alejoe91, 2023-05-25:
#1677

zm711, 2023-05-26:
Thanks @alejoe91 :)

1) Based on some research it looks like the 'edit on Github' for rtd doesn't currently (& probably since it's been an issue for ~5 years) work for sphinx-gallery stuff. So the option is to globally turn off edit on github or just deal with it being broken on examples. So nothing to do there.

2) I'm going to close this, but while looking into this I realized the `how_to` are actually being made in a completely different way (normal doc build from `.rst` files, which relates back to #1407. I had built the fresh `.rst`, but then deleted them (at the time), but it turns out they aren't actually being fed into sphinx-gallery so the source `.py` have been proofread, but the display docs (on rtd) haven't been changed. It might be more sustainable to actually port the `how_to` into the `examples` folder so they can be autobuilt--so I'll work on that and then I can propose some new titles for the unified section. Unless you specifically want to keep `how_to` and `modules_gallery` separate with two different ways of generating those example docs?

closed 2023-05-26 by zm711

alejoe91, 2023-05-26:
Thanks for digging into this! For 2. we run the examples and build them locally because it's Neuropixels data and we need efficient processing (GPU and several cores). Feel free to make changes both to the py file! We use a tool to keep the py file in sync with a notebook and then export it to rst. :)