SpikeInterface/spikeinterface #1689: Drop recrusive option for `to_dict` if we don't find a use case by 2024

h-mayorquin, 2023-06-05:
Recently we solved some pickling loading speed with big provenance on #1674.

While profiling, we found that the most performant solution for pickling and unpickling big provenance pipelines was to NOT recursively expand Extractors in the kwargs with to_dict.

That said, we decided to also merged #1672 as @samuelgarcia wanted to keep the functionality as this was how it was before the serialization effort in https://github.com/SpikeInterface/spikeinterface/issues/1235 and in case we found a use case. I am adding this issue here to keep track of the context and discussion. The idea was that maybe there is some use case for the recursive expansion and then it would be easier to use if we have the code already at hand. If this does not come to pass by next year, I think we should get rid of this code in the `to_dict` function as this is a central function, and we should try to keep the complexity down.

DradeAW, 2023-06-20:
I found another use case for `recursive=True`:

```python
base_folder = Path(...)

npz_sorting.to_dict(relative_to=base_folder)  # Works like a charm
npz_sorting.select_units(...).to_dict(relative_to=base_folder)  # Doesn't work because path is in a parent
npz_sorting.select_units(...).to_dict(relative_to=base_folder, recursive=True)  # Works correctly
```

alejoe91, 2023-11-06:
@h-mayorquin I think we can close this

h-mayorquin, 2023-11-06:
Yes!

closed 2023-11-06 by h-mayorquin