SpikeInterface/spikeinterface #1695: stream_id not found in containerized sorting

LaurenzMuessig, 2023-06-06:
Hi guys,

I recently had to re-install spike interface from scratch. Now when I try to run a containerized sorting (kilosort2) with a concatenated recording I get the error below. There is a copy of the concatenated and processed binary on the local disk. If I run it on a single binary pointing to the raw data folder it runs fine. Is there a possibility that this bug has crept in again in the most recent version. Any help is greatly appreciated. Thanks. This should be similar issue as in #1662.  

full error message:

Starting container
Installing spikeinterface from sources in spikeinterface/kilosort2-compiled-base
Installing dev spikeinterface from remote repository
Installing extra requirements: ['neo']
Running kilosort2 sorter inside spikeinterface/kilosort2-compiled-base
Stopping container

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[3], line 82
     78     rec_Fin.save(folder=os.path.join(outDir,'raw'),verbose=True,progress_bar=True)
     79     # sort
     80     # recTemp = se.SpikeGLXRecordingExtractor(outDir,stream_id='imec0.ap')
     81     # sorting = ss.run_kilosort3(recording=rec_filt,output_folder=outDir,docker_image=True,**KS3Params) 
---> 82     sorting = ss.run_kilosort2(recording=rec_Fin,output_folder=outDir,docker_image=True,verbose=True, **KS2Params)
     83     # this needs work - rename cached traces to ap.bin, plus need to generate the .meta file 
     84     # runCatGT(fileName,outDir,params)
     85     
     86 # print('Yeepie')

File ~/si/spikeinterface/src/spikeinterface/sorters/runsorter.py:693, in run_kilosort2(*args, **kwargs)
    692 def run_kilosort2(*args, **kwargs):
--> 693     return run_sorter("kilosort2", *args, **kwargs)

File ~/si/spikeinterface/src/spikeinterface/sorters/runsorter.py:135, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    133         else:
    134             container_image = singularity_image
--> 135     return run_sorter_container(
    136         container_image=container_image,
    137         mode=mode,
    138         **common_kwargs,
    139     )
    141 return run_sorter_local(**common_kwargs)

File ~/si/spikeinterface/src/spikeinterface/sorters/runsorter.py:585, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    583 if run_error:
    584     if raise_error:
--> 585         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    586 else:
    587     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/media/SSD2/kilosort_LM/batchClust/in_container_sorter_script.py", line 9, in <module>\n    recording = load_extractor(\'/media/SSD2/kilosort_LM/batchClust/in_container_recording.json\')\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 992, in load_extractor\n    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 580, in load\n    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 406, in from_dict\n    extractor = _load_extractor_from_dict(dictionary)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 917, in _load_extractor_from_dict\n    new_kwargs[name] = _load_extractor_from_dict(value)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 917, in _load_extractor_from_dict\n    new_kwargs[name] = _load_extractor_from_dict(value)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 917, in _load_extractor_from_dict\n    new_kwargs[name] = _load_extractor_from_dict(value)\n  [Previous line repeated 1 more time]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 921, in _load_extractor_from_dict\n    new_kwargs[name] = [transform_dict_to_extractor(e) for e in value]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 921, in <listcomp>\n    new_kwargs[name] = [transform_dict_to_extractor(e) for e in value]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 914, in <lambda>\n    transform_dict_to_extractor = lambda x: _load_extractor_from_dict(x) if is_dict_extractor(x) else x\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 936, in _load_extractor_from_dict\n    extractor = extractor_class(**new_kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/spikeglx.py", line 51, in __init__\n    NeoBaseRecordingExtractor.__init__(\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 124, in __init__\n    assert stream_id in stream_ids, f"stream_id {stream_id} is not in {stream_ids}"\nAssertionError: stream_id imec0.ap is not in []\n'

alejoe91, 2023-06-06:
Hi @LaurenzMuessig 

Are you using SI from main or the latest releas (0.97.1)?

LaurenzMuessig, 2023-06-06:
From main I hope. When I do _git pull ._  it says all up to date.

Cheers

alejoe91, 2023-06-06:
Ok, just to make sure I understand correctly: you have several SpikeGLX files that you concatenate with the `si.concatenate_recordings` and then pass to the `run_sorter` function with KS2. Correct?

LaurenzMuessig, 2023-06-06:
Yep. That's exactly what I do.

LaurenzMuessig, 2023-06-06:
Thanks so much for your help with this.

alejoe91, 2023-06-08:
@LaurenzMuessig can you test if this PR #1698  fixes your issue?

In brief you have to:
1. install SpikeInterface from that PR
2. set the following env variable: `SPIKEINTERFACE_DEV_PATH="path-to-cloned-repo` (this uses your local copy of SI in the container

Let me know!

LaurenzMuessig, 2023-06-10:
Hey @alejoe91,

Sorry for being slow but I am busy with something else at the moment. Hopefully early next week I will have a chance to try this out. I'll keep you posted. Cheers

LaurenzMuessig, 2023-06-15:
HI @alejoe91 . It looks like I am still seeing the same issue using the updated code from your PR. Unless I did something wrong in setting it up? Thanks for your help. Cheers

Starting container
******
Container started with the following paths
/home/laurenz/spikeInterface/spikeinterface /sources
Installing spikeinterface from sources in spikeinterface/kilosort2-compiled-base
Installing dev spikeinterface from local machine
Installing extra requirements: ['neo']
Running kilosort2 sorter inside spikeinterface/kilosort2-compiled-base
Stopping container

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[5], line 82
     78     rec_filt.save(folder=os.path.join(outDir,'raw'),verbose=True,progress_bar=True)
     79     # sort
     80     # recTemp = se.SpikeGLXRecordingExtractor(outDir,stream_id='imec0.ap')
     81     # sorting = ss.run_kilosort3(recording=rec_filt,output_folder=outDir,docker_image=True,**KS3Params) 
---> 82     sorting = ss.run_kilosort2(recording=rec_filt,output_folder=outDir,docker_image=True,verbose=True, **KS2Params)
     83     # this needs work - rename cached traces to ap.bin, plus need to generate the .meta file 
     84     # runCatGT(fileName,outDir,params)
     85     
     86 # print('Yeepie')

File ~/si/spikeinterface/src/spikeinterface/sorters/runsorter.py:693, in run_kilosort2(*args, **kwargs)
    689 def run_ironclust(*args, **kwargs):
    690     return run_sorter("ironclust", *args, **kwargs)
--> 693 run_ironclust.__doc__ = _common_run_doc.format("ironclust")
    696 def run_kilosort(*args, **kwargs):
    697     return run_sorter("kilosort", *args, **kwargs)

File ~/si/spikeinterface/src/spikeinterface/sorters/runsorter.py:135, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    133         container_image = docker_image
    134 else:
--> 135     mode = "singularity"
    136     assert not docker_image
    137     if isinstance(singularity_image, bool):

File ~/si/spikeinterface/src/spikeinterface/sorters/runsorter.py:585, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    583 output_folder = Path(output_folder)
    584 log_file = output_folder / "spikeinterface_log.json"
--> 585 if not log_file.is_file():
    586     run_error = True
    587 else:

SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/media/SSD2/kilosort_LM/batchClust/in_container_sorter_script.py", line 9, in <module>\n    recording = load_extractor(\'/media/SSD2/kilosort_LM/batchClust/in_container_recording.json\')\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 926, in load_extractor\n    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 535, in load\n    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 368, in from_dict\n    extractor = _load_extractor_from_dict(dictionary)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 850, in _load_extractor_from_dict\n    kwargs[k] = _load_extractor_from_dict(v)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 850, in _load_extractor_from_dict\n    kwargs[k] = _load_extractor_from_dict(v)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 850, in _load_extractor_from_dict\n    kwargs[k] = _load_extractor_from_dict(v)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 856, in _load_extractor_from_dict\n    kwargs[k] = [_load_extractor_from_dict(e) for e in v]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 856, in <listcomp>\n    kwargs[k] = [_load_extractor_from_dict(e) for e in v]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 869, in _load_extractor_from_dict\n    extractor = cls(**kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/spikeglx.py", line 51, in __init__\n    NeoBaseRecordingExtractor.__init__(\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 124, in __init__\n    assert stream_id in stream_ids, f"stream_id {stream_id} is not in {stream_ids}"\nAssertionError: stream_id imec0.ap is not in []\n'

zm711, 2024-07-19:
We've had a lot of updates across the code base. If this persists after updating to the newest versions please reopen with additional information :)

closed 2024-07-19 by zm711