SpikeInterface/spikeinterface #1703: Fighting with the bot

samuelgarcia, 2023-06-07:
Some spikeinterface contributor have noticed this : we have installed a bot [pre-commit-.ci] which is here to clean the code with black.
This issue to an open discussion not to debate on massive and hidden that force all devs to use black itself, this is a lost cause.
The debate is more on pre-commit.ci itself.

Some devs like me are quite happy with a low tech tool stack for developpement (simple editor + git + shell).
I think I like frugality and simplicity. From my point of view, the pre-commit machinery is the exact oposite. So I prefer to not have it on all dev station.

Because of me, to avoid forcing local installation, we installed the bot pre-commit.ci which run for you at every commit some check + black formatting.
So if you do not have black locally, this leads a infuriating situation : every time you push something, black is run and a new commit is done automatically and then it make the next `git push` totally impossible because of conflicts.

So here I am collecting feeling from regular devs.
What do you think of this new rules of pre-commit.ci ? 
One possible anwser of course.

**Case 1** I have a sophisticated tool stack for developement:
  *  I am happy
  *  I do not care because everything is already run locally.

**Case 2** I am an old school dev (and also a bit lazy)
  * I am not happy
  * I am quite irritated

[image]

DradeAW, 2023-06-07:
I think in general the idea behind it is nice, it allows to have more readable code and to have the same convention for everyone.

I agree that developers shouldn't have to install and run it locally: this is too much of a hassle and I think it will slow down progress for SpikeInterface.

Having it in the PR is nice, except I have some issues with it:
- As you pointed out, after a `git push`, you need to remember to wait a minute and `git pull`, which is annoying.
- When you push the code, the bot will commit on top of it, which cancels the previous GitHub Actions and I receive about 5 mails / notifications that test x or y was cancelled. That every time the bot commits something, it's annoying.

Is it possible to have the bot not operate automatically for rather on demand?
I have in mind something where people make their PR like before, no bot that automatically pushes commit on your branch. Then when you guys decide that you want to merge the PR, you ask the bot to reformat the code (last push on the branch) and merge it.
If this is do-able, I think it would be the best scenario :)

samuelgarcia, 2023-06-07:
@DradeAW : I think @h-mayorquin fight a lot to implement what you suggest in a github action but the pain was enormous.

h-mayorquin, 2023-06-07:
@DradeAW @samuelgarcia 

I disagree that is a hassle to run it locally:
```
pip install pre-commit
pre-commit install
```

Then everytime that you do a commit it fixes the code for you so you don't get the problems in the CI.

I really don't think this is a big hassle than a bunc of stuff that we impose in our developers already:
1) Using pytest (more or less the same level of hassle)
2) Having datalad to run full tests (moderalty more of a hassle in my view)
3) Having to understand our ad-hoc machinery for doing paralle-processing `ChunkRecordingExecutor` for implementing something mildly trivial. (terrible hassle!)

I think that I can understand @samuelgarcia ideological instance on "I don't want to install those tools" but I disagree with qualifying it as a hassle in the context ot other burderns that we already in our users.

I think that a compromise is that we can deactivate tests (and hopefully the bot) when a PR is on draft. That way, people that don't want to install pre-commit locally can switch to ready and start getting the slower tests. If following good practices, you should not commit every five minutes once you PR is ready (which is once you change from draft to ready). I think this is good.

Also, you don't need to run pull and fix conflicts. The bot only correct styles so you can just ignore it. If you find a conflict just run either:
```
git pull -x ours 
or 
git merge -x ours  # Best practice
```
This will just erase what the bot did and you can then push without conflicts.

h-mayorquin, 2023-06-07:
@DradeAW 
The point about the test being re-run is a good one. We can condition tests on the CI to only run once the pre-commit boot has passed I think and that will avoid the re-run (and save some energy).  I don't have my notifications activated for testing so I did not see this before.

DradeAW, 2023-06-07:
> I really don't think this is a bigger hassle than a bunch of stuff that we impose in our developers already:
> 
> 1. Using pytest (more or less the same level of hassle)
> 2. Having datalad to run full tests (moderalty more of a hassle in my view)
> 3. Having to understand our ad-hoc machinery for doing parallel-processing `ChunkRecordingExecutor` for implementing something mildly trivial. (terrible hassle!, me and I think that @pauladkisson have this confusing experience with this when we started coding on the library )

1. I sometimes forget to run it, and very rarely run the full tests (takes too long)
2. I only installed it like a month ago
3. I still haven't completely understood and am often checking with samuel and alessio

I think if we implement it locally, many people will forget / be too lazy to do it.

> We can condition tests on the CI to only run once the pre-commit boot has passed I think and that will avoid the re-run (and save some energy).
If the other method is too hard to implement, I think this can be a good fix!

samuelgarcia, 2023-06-07:
@alejoe91 you are muted.

h-mayorquin, 2023-06-07:
@DradeAW I don't want to remove it from CI. Having it ok CI is so we can avoid forcing people to install it locally. The instructions above:

'''
pip install pre-commit
pre-commit install
'''

Is an opt-in to solve pain points in the CI if you don't want to modify your (plural your) previous workflow. That is, you (plural) who want to remove the pain points in the CI, can do so by running those two lines only once and forget about it.

alejoe91, 2023-06-07:
@samuelgarcia we discussed about this. I think it takes a few days to get used to it and then it's fine.

easy-electrophysiology, 2023-06-08:
Ideally pre-commit is run locally as a benefit is that during the PR reveiw all code is consistently formatted to assist reveiwing. Also, with flake8 (and, possibly down the line a type-checker such as mpyp or pyright) can help ensure bugs are not accidently pushed to the PR. I did also find pre-commit very annoying when I first started, and frequently used the `--no-verify` flag. However, as you get used to it it becomes an invaluable tool in pre-checking commits before push to PR.

If pre-commit is messing with your flow, one workaround is to use `--no-verify` as you build commits locally. Then before pushing, you can run `pre-commit run --all-files`  and commit a final commit like `fix_linting`. This isn't ideal as its possible to forget but is preferable to not using at all, as it means that for the most part formatted code is pushed to PR.

I think the idea of the the formatting been done in the PR is really nice, but in practice can lead to tricky git history issues. I couldnt find the pre-commit.ci that was referred to but assume from the conversion it is automatically committing formatted code on top of the most recently pushed commit.

I would suggest rather than the pre-commit hooks on ci fixing the commit, they just test it and fail if the code is not properly linted.  Here is an [example](https://github.com/neuroinformatics-unit/actions/blob/main/lint/action.yml) linting action and the full [workflow](https://github.com/neuroinformatics-unit/actions/blob/main/example_test_and_deploy.yml). I would advocate this approach because it means you can see when linting has failed,  in un-linted code is pushed, but doesn't break git histories.

The PR can be worked on with the linting failing, but merging blocked. It can be cleaned up at the very end with a dedicated commit after `pre-commit run --all-files` is run locally by whoever is merging. Ideally, pre-commit is run locally to allow the PR to be reveiwed post-lint and avoid this last fix-linting step on the PR, but this is a possible comprimise. We have it set up in the example that tests don't run if linting fails, but this could be skipped if you want to run tests as the PR is developed but the code is not linted.

JoeZiminski, 2023-06-08:
Ideally pre-commit is run locally as a benefit is that during the PR reveiw all code is consistently formatted to assist reveiwing. Also, with flake8 (and, possibly down the line a type-checker such as mpyp or pyright) can help ensure bugs are not accidently pushed to the PR. I did also find pre-commit very annoying when I first started, and frequently used the `--no-verify` flag. However, as you get used to it it becomes an invaluable tool in pre-checking commits before push to PR.

If pre-commit is messing with your flow, one workaround is to use `--no-verify` as you build commits locally. Then before pushing, you can run `pre-commit run --all-files`  and commit a final commit like `fix_linting`. This isn't ideal as its possible to forget but is preferable to not using at all, as it means that for the most part formatted code is pushed to PR.

I think the idea of the the formatting been done in the PR is really nice, but in practice can lead to tricky git history issues. I couldnt find the pre-commit.ci that was referred to but assume from the conversion it is automatically committing formatted code on top of the most recently pushed commit.

I would suggest rather than the pre-commit hooks on ci fixing the commit, they just test it and fail if the code is not properly linted.  Here is an [example](https://github.com/neuroinformatics-unit/actions/blob/main/lint/action.yml) linting action and the full [workflow](https://github.com/neuroinformatics-unit/actions/blob/main/example_test_and_deploy.yml). I would advocate this approach because it means you can see when linting has failed,  in un-linted code is pushed, but doesn't break git histories.

The PR can be worked on with the linting failing, but merging blocked. It can be cleaned up at the very end with a dedicated commit after `pre-commit run --all-files` is run locally by whoever is merging. Ideally, pre-commit is run locally to allow the PR to be reveiwed post-lint and avoid this last fix-linting step on the PR, but this is a possible comprimise. We have it set up in the example that tests don't run if linting fails, but this could be skipped if you want to run tests as the PR is developed but the code is not linted.

alejoe91, 2023-11-06:
UPDATE: @samuelgarcia is still alive and healty despite `pre-commit ci`

Since there's no going back, I'll close this.

closed 2023-11-06 by alejoe91

samuelgarcia, 2023-11-07:
This is NOT true, I feel sligthly deppressed since this, this is very insidious.
`--no-verify` is my last friend