SpikeInterface/spikeinterface #1705: Loading Axion Biosystems AXIS data

alexmascension, 2023-06-07:
Hi! 

Does spikeinterface support loading .raw or .spk formats of AXIS data? I don't see it in the docs, so I guess this format is not implemented. 

In any case, would it be possible to create a BaseRecording object with electrode recording data and sample frequency alone?

Thanks!

alejoe91, 2023-06-07:
Hi @alexmascension 

We use [NEO](https://github.com/NeuralEnsemble/python-neo) for reading multiple formats and the Axion system doesn't seem to be supported there. Could you open an issue on the NEO page?

You can load a recording with data and sampling rate in memory using the `NumpyRecording`: https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.NumpyRecording

alexmascension, 2023-06-07:
Perfect, I will open the issue there and try to work with `NumpyRecording` on the meantime then.
Thanks!

closed 2023-06-07 by alexmascension

ajcheng1, 2024-08-26:
I was curious if it did work to use a numpy recording to load onto Axion? If so, was it through a .raw or .spk? Could I by chance look at the code for loading the file?

alexmascension, 2024-10-13:
Hi! I created some functions that are located [here](https://github.com/NanoNeuro/MEAS_PRUEBAS_ANE/blob/main/notebooks/py_functions/MEAS_recordings.py)

ajcheng1, 2024-10-16:
Cool! Thanks so much. I was looking at the script load_axion_files.m, and I was wondering if the relevant code for loading voltages into .txt files is located in line 1 to 61? Is the time extraction not necessary?