SpikeInterface/spikeinterface #1712: run_sorter returns KeyError

avakiai, 2023-06-09:
Hi there, 

I'm trying to run spykingcircus2 on some data. I get passed the stage where I assign preprocessing steps, but then the run_sorter function will only return a KeyError and a message that tells me the last preprocessing step I asked for. I don't understand the source of the error. Any tips? 
```
 sorting = ss.run_sorter(recording_w, 'spykingcircus2', verbose=True, raise_error=True)
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[90], line 2
      1 # sorting
----> 2 sorting = ss.run_sorter(recording_w, 'spykingcircus2', verbose=True, raise_error=True)

File C:\ProgramData\Anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)

File C:\ProgramData\Anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\runsorter.py:155, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    151 if isinstance(recording, list):
    152     raise Exception(
    153         'You you want to run several sorters/recordings use run_sorters(...)')
--> 155 SorterClass = sorter_dict[sorter_name]
    157 # only classmethod call not instance (stateless at instance level but state is in folder)
    158 output_folder = SorterClass.initialize_folder(
    159     recording, output_folder, verbose, remove_existing_folder)

KeyError: WhitenRecording: 16 channels - 1 segments - 10.0kHz - 42.900s

```
My data is of the format: 
```
MCSH5RecordingExtractor: 16 channels - 1 segments - 10.0kHz - 42.900s
```

Thanks!

h-mayorquin, 2023-06-09:
Hi, @avakiai , why do you want to delete the isssue? Usually we just close them. 

Would you mind undoing the eddit and maybe also write how it was solved? 
This can be useful for the community in the future.

avakiai, 2023-06-09:
Noted!

closed 2023-06-09 by avakiai