SpikeInterface/spikeinterface #1715: issue with sliced recordings

kshtjkumar, 2023-06-11:
I have a a single recording form two different types of probes, one is from depth and one is from ECoG. To analyse the each probe separately I slice the recording in two ways, 1st way: I sliced the desired channels from the raw recording and then filtered it downsampled it and plotted it for one channel (first image below).  2nd way: I filtered the whole raw unsliced recording for all channels and did downsampling etc first then sliced it, re-referenced it and plotted (second image). Both of them show very different plots can you explain which one is correct and why ?

[image]
[image]

Further when I use mountainsort5 to sort the whitened recording i get this error :
[image]
[image]

Can you help with this issue ?

alejoe91, 2023-06-12:
Hi,

I think that the "mismatch" comes from the fact that in the first plot you ar doing: common ref --> bp filter. If the data has an offset, this would mess up the signal. Can you try first doing the filters (bp and notch) and then CMR?

Also, do you expect to see spikes from these ecog signals? do you "see" them when you plot the traces? If you don't see clear separable spikes, than probably the spike sorters won't find them or even fail.

For the second issue: by defaul mountainsort re-filters the data. Since you are downsampling them, this is failing because the low-pass cutoff is 6khz by default, but you have resampled to 2Khz. If your goal is to spike sort the data, I don't recommend downsampling the data at all! You are throwing away information that might be useful to the sorters in doing so!

kshtjkumar, 2023-06-14:
yes! thank you so much! both your proposed solutions worked. Also thank you for the suggestion, I wont be downsampling the data for spike activity. Yes, basically these are micro-electrode ECoGs and they  capture spikes but very less neuronal clusters (say 1 or 2).

closed 2023-06-14 by kshtjkumar