SpikeInterface/spikeinterface #1717: numpy.core_exceptions._ArrayMemoryError

alex2thiele, 2023-06-13:
Running spikeinterface with tridesclous using a 16 channel Neuralynx .ncs setup I receive th following error:
"numpy.core_exceptions._ArrayMemoryError: Unable to allocate 10.GiB for array with shape (173224448m16) and data type float 32".

This only happens when .ncs files >~300,000KB, not when they are ~250.000KB, using the same data, only chunked to different sizes. 

these same data work without error using ironclust.

alejoe91, 2023-06-13:
HI @alex2thiele 

How did you load your recording? @samuelgarcia can you take a look?

alex2thiele, 2023-06-13:
> ⚠ External sender. Take care when opening links or attachments. Do not provide your login details.

Hi Alessio

the loading method is "PipeLineConfig.load_recording", and we use pipeline.py (attached as zip)

best
Alex

alejoe91, 2023-06-13:
@alex2thiele I can't access the zip file. I think you have to attach it from the browser, not using the email

alejoe91, 2023-06-13:
If you have the script in a public GH repo, it would be even better to just share the link to it!

samuelgarcia, 2023-06-13:
@alex2thiele : neuralynx is really a complicated and bad data format. Many users have problems using it.
I woul advice to convert it to biary without using parralel and then making the sorting on the binary file itself.

In short

```python
rec_neuralynx = si.read_neuralynx('/path/to/my/data')
print(rec_neuralynx)
rec_bin = rec_neuralynx.save(folder='/another/folder', n_jobs=1, progress_bar=True)
print(rec_bin)
```
and then
```python
rec = si.load_extractor('/another/folder')
print(rec)
sorting = si.run_sorter('XXX', rec)

```

h-mayorquin, 2023-07-11:
@alex2thiele still having problems with this?

alex2thiele, 2023-07-12:
Hi Heberto

Unfortunately yes. The conversion to binary data did not solve the problem, the error was the same.

Best
Alex

alex2thiele, 2023-07-12:
During sorting, conversion to binary worked fine. There seems to be a size limit where the error starts to occur. If I use the same files, but chopped into a shorter bit, then no error occurs. It also doe not matter which recordings I use (e.g. different days), once the file sizes exceed the limits the error occurs. Don’t know the exact cut off, but about 300000 kB seems the limit.

Did you get the error during the conversion? or during spike sorting?

alex2thiele, 2023-07-12:
That is 16*300000kb, as it was a 16 channel recording.

alejoe91, 2023-07-12:
Ok and that's only for tridesclous...I'll let @samuelgarcia handle this!

h-mayorquin, 2023-07-12:
@alex2thiele 
Can you share some things that would be useful for us:
1) The file and script, you shared it with us before but you did as an answer from the email. You need to click in this link:
https://github.com/SpikeInterface/spikeinterface/issues/1717 and attach in github. 
2) Your OS (windows, linux, running locallly). It would be great to know how much memory (ram) you have as well!
3) Spikeinterface version (if you don't know this, then how did you install it)
4) If you can, the full trace of the error or the log in the sorter.

I think this infromation will be useful for other users if they stumble upon this issue in the future and for us to work on a solution.

alex2thiele, 2023-07-13:
With the following commands

conda create -n spikeinterface python=3.8
conda activate spikeinterface
pip install spikeinterface[full]

h-mayorquin, 2023-07-13:
Thanks a bunch, we will look into it!

zm711, 2025-01-14:
Since we no longer support python 3.8 can we close this.

I think the issue is not enough RAM. I've seen this error before on my datasets. On Windows if I try to analyze a bunch of datasets in a row all on the same kernel I see a slight memory leak (but I'm also doing a little analysis in the middle so the memory leak could be my fault and not our fault). So if my datasets are big enough I am able to generate this exact error during my sorting. Basically during sorting if there isn't enough RAM numpy raises this error to say it cannot generate an array. 

I think working on memory management is important for the library overall, but I think the solution for this size datasets is to use a computer with more RAM especially if on Windows. @h-mayorquin or @alejoe91 are you both good with closing this as stale?

closed 2025-02-05 by zm711

h-mayorquin, 2025-02-06:
Sorry for not replying Zach, good call on closing it when people do not answer.