SpikeInterface/spikeinterface #1720: Error with spikeinterface.exporters.export_to_phy

sachuriga283, 2023-06-14:
Hi, I got some errors from export_to_phy. Would anyone know how this happend? 

ValueError                                Traceback (most recent call last)
Cell In[24], line 2
      1 phy_TRD=current_file_name + "_" + "phy_ks3"
----> 2 sexp.export_to_phy(we,
      3                    output_folder=base_folder / phy_TRD,compute_pc_features=True, 
      4                    compute_amplitudes=True, 
      5                    sparsity=None, copy_binary=True, 
      6                    remove_if_exists=True, 
      7                    peak_sign='neg', 
      8                    template_mode='median',  
      9                    verbose=True)

File ~\.conda\envs\spi\lib\site-packages\spikeinterface\exporters\to_phy.py:210, in export_to_phy(waveform_extractor, output_folder, compute_pc_features, compute_amplitudes, sparsity, copy_binary, remove_if_exists, peak_sign, template_mode, dtype, verbose, **job_kwargs)
    208 if unit_groups is None:
    209     unit_groups = np.zeros(len(unit_ids), dtype='int32')
--> 210 channel_group = pd.DataFrame({'cluster_id': [i for i in range(len(unit_ids))],
    211                               'channel_group': unit_groups})
    212 channel_group.to_csv(output_folder / 'cluster_channel_group.tsv',
    213                      sep="\t", index=False)
    215 if waveform_extractor.is_extension('quality_metrics'):

File ~\.conda\envs\spi\lib\site-packages\pandas\core\frame.py:709, in DataFrame.__init__(self, data, index, columns, dtype, copy)
    703     mgr = self._init_mgr(
    704         data, axes={"index": index, "columns": columns}, dtype=dtype, copy=copy
    705     )
    707 elif isinstance(data, dict):
    708     # GH#38939 de facto copy defaults to False only in non-dict cases
--> 709     mgr = dict_to_mgr(data, index, columns, dtype=dtype, copy=copy, typ=manager)
    710 elif isinstance(data, ma.MaskedArray):
    711     from numpy.ma import mrecords

File ~\.conda\envs\spi\lib\site-packages\pandas\core\internals\construction.py:481, in dict_to_mgr(data, index, columns, dtype, typ, copy)
    477     else:
    478         # dtype check to exclude e.g. range objects, scalars
    479         arrays = [x.copy() if hasattr(x, "dtype") else x for x in arrays]
--> 481 return arrays_to_mgr(arrays, columns, index, dtype=dtype, typ=typ, consolidate=copy)

File ~\.conda\envs\spi\lib\site-packages\pandas\core\internals\construction.py:115, in arrays_to_mgr(arrays, columns, index, dtype, verify_integrity, typ, consolidate)
    112 if verify_integrity:
    113     # figure out the index, if necessary
    114     if index is None:
--> 115         index = _extract_index(arrays)
    116     else:
    117         index = ensure_index(index)

File ~\.conda\envs\spi\lib\site-packages\pandas\core\internals\construction.py:655, in _extract_index(data)
    653 lengths = list(set(raw_lengths))
    654 if len(lengths) > 1:
--> 655     raise ValueError("All arrays must be of the same length")
    657 if have_dicts:
    658     raise ValueError(
    659         "Mixing dicts with non-Series may lead to ambiguous ordering."
    660     )

ValueError: All arrays must be of the same length

alejoe91, 2023-07-03:
Hi @sachuriga283 

Can you share the full script? 

Note that you can use the code formatting both `inline` and for code blocks that makes everything easier to read:
```
code
block
example
```

ChenLuisme, 2023-07-19:
I have similar problem: 

```
from spikeinterface.postprocessing import compute_spike_amplitudes, compute_principal_components
from spikeinterface.exporters import export_to_phy

export_to_phy(we, output_folder='C:/NP_Data_Temp/kilosort3_output/phy', 
verbose=True, compute_pc_features=False)
```
Here is the error:
```

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[106], line 9
      2 from spikeinterface.exporters import export_to_phy
      4 # some computations are done before to control all options
      5 # compute_spike_amplitudes(we)
      6 # compute_principal_components(we, n_components=3, mode='by_channel_global')
      7 
      8 # the export process is fast because everything is pre-computed
----> 9 export_to_phy(we, output_folder='[C:/NP_Data_Temp/kilosort3_output/phy](file:///C:/NP_Data_Temp/kilosort3_output/phy)', 
     10 verbose=True, compute_pc_features=False)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\exporters\to_phy.py:221](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/exporters/to_phy.py:221), in export_to_phy(waveform_extractor, output_folder, compute_pc_features, compute_amplitudes, sparsity, copy_binary, remove_if_exists, peak_sign, template_mode, dtype, verbose, **job_kwargs)
    218     for column_name in qm_data.columns:
    219         # already computed by phy
    220         if column_name not in ["num_spikes", "firing_rate"]:
--> 221             metric = pd.DataFrame({'cluster_id': [i for i in range(len(unit_ids))],
    222                                    column_name: qm_data[column_name].values})
    223             metric.to_csv(output_folder / f'cluster_{column_name}.tsv',
    224                           sep="\t", index=False)
    226 if verbose:

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\pandas\core\frame.py:709](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/pandas/core/frame.py:709), in DataFrame.__init__(self, data, index, columns, dtype, copy)
    703     mgr = self._init_mgr(
    704         data, axes={"index": index, "columns": columns}, dtype=dtype, copy=copy
    705     )
    707 elif isinstance(data, dict):
    708     # GH#38939 de facto copy defaults to False only in non-dict cases
--> 709     mgr = dict_to_mgr(data, index, columns, dtype=dtype, copy=copy, typ=manager)
    710 elif isinstance(data, ma.MaskedArray):
    711     from numpy.ma import mrecords

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\pandas\core\internals\construction.py:481](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/pandas/core/internals/construction.py:481), in dict_to_mgr(data, index, columns, dtype, typ, copy)
    477     else:
    478         # dtype check to exclude e.g. range objects, scalars
    479         arrays = [x.copy() if hasattr(x, "dtype") else x for x in arrays]
--> 481 return arrays_to_mgr(arrays, columns, index, dtype=dtype, typ=typ, consolidate=copy)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\pandas\core\internals\construction.py:115](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/pandas/core/internals/construction.py:115), in arrays_to_mgr(arrays, columns, index, dtype, verify_integrity, typ, consolidate)
    112 if verify_integrity:
    113     # figure out the index, if necessary
    114     if index is None:
--> 115         index = _extract_index(arrays)
    116     else:
    117         index = ensure_index(index)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\pandas\core\internals\construction.py:655](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/pandas/core/internals/construction.py:655), in _extract_index(data)
    653 lengths = list(set(raw_lengths))
    654 if len(lengths) > 1:
--> 655     raise ValueError("All arrays must be of the same length")
    657 if have_dicts:
    658     raise ValueError(
    659         "Mixing dicts with non-Series may lead to ambiguous ordering."
    660     )

ValueError: All arrays must be of the same length
```

ChenLuisme, 2023-07-20:
> ⚠ Caution: External sender
>
> Hi,
>
> There seem to be a mismatch in the CSV files. Could you share the quality_metrics/metrics.csv file?
>
> Also the output folder from Kilosort (excluding the .dat file)

Hi Alessio,

The issue is gone after update spikeinterface to newest version 🤣 I am sorry that I overwrited the files.

Best wishes,
Chen

Chen Lu
University College London
Ear Institute
332 Gray's Inn Road
London, WC1X 8EE
Tel: +44 (0)75 2993 2905
Email: ***@***.***

alejoe91, 2023-07-20:
Closing then!

closed 2023-07-20 by alejoe91