SpikeInterface/spikeinterface #1721: How to read saved preprocessed recording for spike sorting

shunnnli, 2023-06-14:
Hi,

Can I load saved preprocessed recording (preprocessed using spikeinterface as well) for later spike sorting?

Thanks a lot!

samuelgarcia, 2023-06-15:
Hi,
of course.

```python
rec = si.read_someformat('/path/to/data')
rec_f = si.bandpass_filter(rec)
rec_saved = rec_f.save(folder='/path/saved', format='binary', n_jobs=10, progress_bar=True)
print(rec_saved)

# and then later in your day
rec_reload = si.load_extractor('/path/saved')
print(rec_reload)
# rec_reload and  rec_saved are the same!
```
Alessio and I worked hard to make a better documentation a few month ago : https://spikeinterface.readthedocs.io/en/latest/modules/index.html

I encourage you to have a overview, spikeinterface have lot of hidden features!

h-mayorquin, 2023-07-19:
@shunnnli did this solve you problem? Can we close this issue?

closed 2023-07-19 by shunnnli