SpikeInterface/spikeinterface #1730: Error in multiple sorters

alexmascension, 2023-06-21:
Hi, I'm trying to run different sorters in your module and I get different errors. I'm creating a `NumpyRecording` object to load all the data because it is from a custom probe, so I'm sure that there lays the problem.

The code to load the data is the following:
```
ROOT = '/data/Proyectos/Nanoneuro/data/rCxNs/'
SAMPLE_BASE = 'p2.+-1000mV.500us(000)'

well = (2, 3)

traces_list = []
sampling_frequency = 0
channel_ids = []

for Erow in range(1,5):
    for Ecol in range(1,5):
        filename = f'{ROOT}/{SAMPLE_BASE}/{well[0]}-{well[1]}-{Erow}-{Ecol}_voltageRaw.txt'
        if os.path.exists(filename):
            channel_ids.append(f'{Erow}-{Ecol}')
            list_voltages = np.loadtxt(filename)
            traces_list.append(list_voltages)

            if sampling_frequency == 0:
                df = pd.read_csv(f'{ROOT}/{SAMPLE_BASE}/{SAMPLE_BASE}.info', index_col=0, names=['index', 'value'], sep='\t')
                sampling_frequency = df.loc['SamplingFrequency', 'value']

trace_array = np.asarray(traces_list).transpose()

# In some cases, we might add some activation based on voltage to activate the neurons. This activation signal
# is also recorded and is not necessary, so we remove it
time_samplings_to_mask = [(1.07, 1.075), (3.07, 3.075), (5.07, 5.075), (7.07, 7.075)]

for time_sampling in time_samplings_to_mask:
    t0 = int(time_sampling[0] * sampling_frequency)
    tf = int(time_sampling[1] * sampling_frequency)
    trace_array[t0:tf, :] = 0

sample_recording = si.NumpyRecording(
    traces_list=[trace_array],
    sampling_frequency=sampling_frequency,
    channel_ids=np.asarray(channel_ids)
)

sample_recording.set_property('group', [0] * len(channel_ids))
```
Then I load the probe info:
```
type_MEAS = 16  # 16 or 64   

dist_multiplier = 350 if type_MEAS == 16 else 300
circle_radius = 50

positions = np.zeros((len(channel_ids), 2), dtype=float)
contact_vector = []
for channel_idx, channel in enumerate(channel_ids):
    x_coord, y_coord = (int(channel.split('-')[0]) - 1) * dist_multiplier, (int(channel.split('-')[1]) - 1) * dist_multiplier
    positions[channel_idx, 1], positions[channel_idx, 0] = x_coord, y_coord
    
    contact_vector.append((0, x_coord,   y_coord, 'circle', circle_radius, '', '', channel_idx, 'um', 1., 0., 0., 1.))

# later if we are using peak detection, we may need it
sample_recording.set_channel_locations(locations=positions)

probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=positions, shapes='circle', shape_params={'radius': circle_radius})
probe.device_channel_indices = np.arange(len(channel_ids))
probe.create_auto_shape('rect')

sample_recording.set_probe(probe)

# Create contact_vector
dtypes=[('probe_index', '<i8'), ('x', '<f8'), ('y', '<f8'), ('contact_shapes', '<U64'), 
        ('radius', '<f8'), ('shank_ids', '<U64'), ('contact_ids', '<U64'), ('device_channel_indices', '<i8'), 
        ('si_units', '<U64'), ('plane_axis_x_0', '<f8'), ('plane_axis_x_1', '<f8'), ('plane_axis_y_0', '<f8'), 
        ('plane_axis_y_1', '<f8')]

sample_recording.set_property('contact_vector', np.asarray(contact_vector, dtype=dtypes))
```

I set `sample_recording.is_dumpable = True` to save it.

Then I do the preprocessing as follows:
```
recording_bin = sample_recording.save(n_jobs=8, chunk_duration="1s")
recording_f = spre.bandpass_filter(recording_bin, freq_min=300, freq_max=5000)
recording_cmr = spre.common_reference(recording_f, reference='global', operator='median')
```

I am currently running the following sorters (some with docker, some without) and I get the following errors:
```
sorting_TRDC2 = ss.run_sorter('tridesclous2', recording=rec, output_folder=f'{ROOT}/tmp/TRDC2', docker_image=False)
print('Units found by tridesclous2:', sorting_TRDC2.get_unit_ids())
----
File "[/home/nanoneuro/Programs/miniconda3/envs/nanoneuro/lib/python3.11/site-packages/scipy/signal/_filter_design.py](https://file+.vscode-resource.vscode-cdn.net/home/nanoneuro/Programs/miniconda3/envs/nanoneuro/lib/python3.11/site-packages/scipy/signal/_filter_design.py)", line 2557, in iirfilter
    raise ValueError("Digital filter critical frequencies "
ValueError: Digital filter critical frequencies must be 0 < Wn < 1
```

```
sorting_pyKS = ss.run_sorter('pykilosort', recording=rec, output_folder=f'{ROOT}/tmp/pyKS', docker_image=True)
print('Units found by pykilosort:', sorting_pyKS.get_unit_ids())
----
"error_trace": "Traceback (most recent call last):\n  File \"/home/miniconda3/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/home/miniconda3/lib/python3.8/site-packages/spikeinterface/sorters/external/pykilosort.py\", line 190, in _run_from_folder\n    run(\n  File \"/src/pykilosort/pykilosort/main.py\", line 200, in run\n    out = learnAndSolve8b(ctx)\n  File \"/src/pykilosort/pykilosort/learn.py\", line 847, in learnAndSolve8b\n    wTEMP, wPCA = extractTemplatesfromSnippets(\n  File \"/src/pykilosort/pykilosort/learn.py\", line 69, in extractTemplatesfromSnippets\n    wTEMP[:, j] = cp.dot(dd[:, imax == j], amax[imax == j].T)\n  File \"cupy/_core/core.pyx\", line 1590, in cupy._core.core._ndarray_base.__setitem__\n  File \"cupy/_core/_routines_indexing.pyx\", line 52, in cupy._core._routines_indexing._ndarray_setitem\n  File \"cupy/_core/_routines_indexing.pyx\", line 936, in cupy._core._routines_indexing._scatter_op\n  File \"cupy/_core/_routines_indexing.pyx\", line 420, in cupy._core._routines_indexing._view_getitem\nIndexError: Index 0 is out of bounds for axis 1 with size 0\n",
```

For spykingcircus2 and mountainsort4, I get a proper result:
```
sorting_SPCR2 = ss.run_sorter('spykingcircus2', recording=rec, output_folder=f'{ROOT}/tmp/SPRC2', docker_image=False)
print('Units found by spykingcircus2:', sorting_SPCR2.get_unit_ids())
----
Units found by spykingcircus2: [ 0  1  2  3  4  5  6  7  8  9 10]
```

```
sorting_MS4 = ss.run_sorter('mountainsort4', recording=rec, output_folder=f'{ROOT}/tmp/MS5', docker_image=False)
print('Units found by mountainsort4:', sorting_MS4.get_unit_ids())
----
Mountainsort4 use the OLD spikeextractors mapped with NewToOldRecording
Units found by mountainsort4: [ 1  2  3  4  5  6  7  8  9 10 11 12 13 14 15]
```

In contrast, if I work with the toy example, I get the results.
```
sss_rec, sss_sort = se.toy_example()
sss_rec = si.concatenate_recordings([si.select_segment_recording(sss_rec, segment_indices=0), 
                           si.select_segment_recording(sss_rec, segment_indices=1)])

sss_bin = sss_rec.save(n_jobs=8, chunk_duration="1s")

sss_f = spre.bandpass_filter(sss_bin, freq_min=300, freq_max=5000)
sss_cmr = spre.common_reference(sss_f, reference='global', operator='median')
```

```
rec=sss_cmr

sorting_TRDC2 = ss.run_sorter('tridesclous2', recording=rec, output_folder=f'{ROOT}/tmp/TRDC2', docker_image=False)
print('Units found by tridesclous2:', sorting_TRDC2.get_unit_ids())

sorting_pyKS = ss.run_sorter('pykilosort', recording=rec, output_folder=f'{ROOT}/tmp/pyKS', docker_image=True)
print('Units found by pykilosort:', sorting_pyKS.get_unit_ids())

sorting_SPCR2 = ss.run_sorter('spykingcircus2', recording=rec, output_folder=f'{ROOT}/tmp/SPRC2', docker_image=False)
print('Units found by spykingcircus2:', sorting_SPCR2.get_unit_ids())

sorting_MS4 = ss.run_sorter('mountainsort4', recording=rec, output_folder=f'{ROOT}/tmp/MS5', docker_image=False)
print('Units found by mountainsort4:', sorting_MS4.get_unit_ids())

---
Units found by tridesclous2: [0 1 2 3 4 5 6]
Units found by pykilosort: [0 1 2 3 4 5 6 7 8]
Units found by spykingcircus2: [0 1 2 3 4 5]
Mountainsort4 use the OLD spikeextractors mapped with NewToOldRecording
Units found by mountainsort4: [1 2 3 4 5 6 7]
```

I guess that, there is some data or property missing when I construct my data, but I don't know what is missing.

alejoe91, 2023-06-21:
Hi,

What is the `sampling_frequency` of your recording? 

The first error is because you probably have a sampling frequency < 12kHz, and tridesclous2 by default attempts to filter with [300, 6000] cutoffs.
```
sorting_TRDC2 = ss.run_sorter('tridesclous2', recording=rec, output_folder=f'{ROOT}/tmp/TRDC2', docker_image=False)
print('Units found by tridesclous2:', sorting_TRDC2.get_unit_ids())
----
File "[/home/nanoneuro/Programs/miniconda3/envs/nanoneuro/lib/python3.11/site-packages/scipy/signal/_filter_design.py](https://file+.vscode-resource.vscode-cdn.net/home/nanoneuro/Programs/miniconda3/envs/nanoneuro/lib/python3.11/site-packages/scipy/signal/_filter_design.py)", line 2557, in iirfilter
    raise ValueError("Digital filter critical frequencies "
ValueError: Digital filter critical frequencies must be 0 < Wn < 1
```
To skip tridesclous2 preprocessing add the `apply_preprocessing=False` argument to the `run_*` call. (You should pass the `recording_cmr` object to the sorters).

For the `pykilosort` error, I recommend opening an issue on the `pykilosort` repo directly.

alejoe91, 2023-06-21:
Also note that I was working on updating the pykilosort version of the docker image.

Can you try to run it again? The new version (1.4.3) should now be the default on dockerhub:
https://hub.docker.com/layers/spikeinterface/pykilosort-base/1.4.3/images/sha256-aebf5ace3eadf08873609350b0a3fc5555aace541dbf962606a9e57f420c0230?context=repo

alexmascension, 2023-06-21:
HI @alejoe91,

tridesclous2 works fine now with `apply_preprocessing=False`. As for `pykilosort`, I don't know how to run the 1.4.3 version with my files. In any case, since I already have results from different sorters, I don't really mind if it still fails. I will open the issue in the future if I need that sorter.

Thanks!

alejoe91, 2023-06-21:
So now the `latest` version is `1.4.3`, so you can just rerun with `docker_image=True`.

You can also use a specific tag! For example with: `docker_image="spikeinterface/pykilosort-base:1.4.3"` 

Can I close the issue then?

alexmascension, 2023-06-21:
Yeah, sure!

closed 2023-06-21 by alexmascension