SpikeInterface/spikeinterface #1740: Using Quality matrix to evaluate recordings

kshtjkumar, 2023-06-22:
Hi I am using this to evaluate the quality of my recordings : https://allensdk.readthedocs.io/en/latest/_static/examples/nb/visual_behavior_neuropixels_quality_metrics.html#d-prime

I am not able use many of the libraries such as nn_hit_rate, d_prime etc.  Can you guide how to make graphs like the one in this tutorial using the latest version of spikeinterface ? 

I am using tetrode recordings from the Intan device.

samuelgarcia, 2023-06-23:
Hi @kshtjkumar ,
You should have a look into theses part of the documentation:
https://spikeinterface.readthedocs.io/en/latest/modules/postprocessing.html
https://spikeinterface.readthedocs.io/en/latest/modules/qualitymetrics.html
https://spikeinterface.readthedocs.io/en/latest/modules_gallery/index.html#quality-metrics-tutorial
This one explain the "nearest_neighbor" that have nn_hit_rate and nn_miss_rate
https://spikeinterface.readthedocs.io/en/latest/modules/qualitymetrics/nearest_neighbor.html

zm711, 2023-11-03:
@kshtjkumar,

Just wanted to see if you were able to read the docs and if you had more questions related to performing qualitymetrics on your data?

kshtjkumar, 2023-11-23:
@zm711  am not sure if am understanding it correctly, basically we developed a new set od electrodes, we wanted tot test their data quality. I used mountainsort5 to see the quality matrix. Am confused with waveforms and isi_violation,  when my isi_violation is less than 0.5 then my waveforms are very skewed and deformed, and when they are good the quality matrix is bad! the waveform amplitude of the template waveform is so much different from the waveform formed.
Can you help with such basic things ?

zm711, 2024-03-21:
Moved to a different thread so closing this old one.

closed 2024-03-21 by zm711