SpikeInterface/spikeinterface #1745: Can't reload waveforms back

rat-h, 2023-06-26:
I have a script which preprocessing - sorting - extracting waves - postpocessing data overnight. I'm trying to read back waveforms and got a very strange error.

First, here's the tree of one dataset
```tree
/home/spk/spikeinterface/20230508T195033Z-recording1-continuous
├── jgui_state.json
├── preprocessed
│   ├── binary.json
│   ├── probe.json
│   ├── properties
│   │   ├── contact_vector.npy
│   │   ├── group.npy
│   │   └── location.npy
│   ├── provenance.json
│   ├── si_folder.json
│   └── traces_cached_seg0.raw
├── sorting-saved
│   ├── npz.json
│   ├── properties
│   ├── provenance.json
│   ├── si_folder.json
│   └── sorting_cached.npz
├── sorting-workingdir
│   ├── sorter_output
│   │   ├── sorting
│   │   │   ├── npz.json
│   │   │   ├── properties
│   │   │   ├── provenance.json
│   │   │   ├── si_folder.json
│   │   │   └── sorting_cached.npz
│   │   ├── sorting_temp
│   │   │   ├── npz.json
│   │   │   ├── properties
│   │   │   ├── provenance.json
│   │   │   ├── si_folder.json
│   │   │   └── sorting_cached.npz
│   │   └── waveforms_temp
│   │       ├── params.json
│   │       ├── recording_info
│   │       │   ├── probegroup.json
│   │       │   └── recording_attributes.json
│   │       ├── recording.json
│   │       ├── sorting.json
│   │       ├── templates_average.npy
│   │       └── waveforms
│   │           ├── sampled_index_0.npy
│   │           ├── sampled_index_10.npy
                .....
│   │           ├── sampled_index_89.npy
│   │           ├── sampled_index_8.npy
│   │           ├── sampled_index_9.npy
│   │           ├── waveforms_0.npy
│   │           ├── waveforms_10.npy
                .....
│   │           ├── waveforms_89.npy
│   │           ├── waveforms_8.npy
│   │           └── waveforms_9.npy
│   ├── spikeinterface_log.json
│   ├── spikeinterface_params.json
│   └── spikeinterface_recording.json
└── waveforms
    ├── correlograms
    │   ├── bins.npy
    │   ├── ccgs.npy
    │   └── params.json
    ├── isi_histograms
    │   ├── bins.npy
    │   ├── isi_histograms.npy
    │   └── params.json
    ├── noise_levels
    │   ├── noise_levels.npy
    │   └── params.json
    ├── params.json
    ├── principal_components
    │   ├── params.json
    │   ├── pca_0.npy
        .....
    │   ├── pca_89.npy
    │   ├── pca_8.npy
    │   ├── pca_9.npy
    │   ├── pca_model_by_channel_local_0.pkl
    │   ├── pca_model_by_channel_local_100.pkl
        .....
    │   ├── pca_model_by_channel_local_99.pkl
    │   └── pca_model_by_channel_local_9.pkl
    ├── recording_info
    │   ├── probegroup.json
    │   └── recording_attributes.json
    ├── recording.json
    ├── similarity
    │   ├── params.json
    │   └── similarity.npy
    ├── sorting.json
    ├── spike_amplitudes
    │   ├── amplitude_segment_0.npy
    │   └── params.json
    ├── spike_locations
    │   ├── params.json
    │   └── spike_locations.npy
    ├── templates_average.npy
    ├── templates_std.npy
    ├── unit_locations
    │   ├── params.json
    │   └── unit_locations.npy
    └── waveforms
        ├── sampled_index_0.npy
        ├── sampled_index_10.npy
        .....
        ├── sampled_index_89.npy
        ├── sampled_index_8.npy
        ├── sampled_index_9.npy
        ├── waveforms_0.npy
        ├── waveforms_10.npy
        ....
        ├── waveforms_89.npy
        ├── waveforms_8.npy
        └── waveforms_9.npy
```

A very simple strip should read waveforms and do the next step - curation
```python
import sys
import spikeinterface as si
# import spikeinterface.extractors as se

# from spikeinterface.postprocessing import compute_principal_components
from spikeinterface.qualitymetrics import compute_quality_metrics

for d in sys.argv[1:]:
    try:
        rc = si.load_extractor(d+"/preprocessed")
        sr = si.load_extractor(d+"/sorting-saved")
        we = si.load_waveforms(d+'/waveforms')
    except BaseException as e:
        print(f"Cannot load waveform from {d}: {e}")
        continue
    metrics = compute_quality_metrics(we, metric_names=['snr', 'isi_violation', 'nearest_neighbor'])
    print(d)
    print(metrics)
```

but if I run it returns an error:
```bash
$ python test-curation.py 20230508T195033Z-recording1-continuous
Cannot load waveform from 20230508T195033Z-recording1-continuous: The recording could not be loaded. You can use the `with_recording=False` argument
```

If `with_recording=False` is set
```python
we = si.load_waveforms(d+'/waveforms',with_recording=False)
```
I got a very strange result:
```bash
$ python test-curation.py 20230508T195033Z-recording1-continuous
Cannot load waveform from 20230508T195033Z-recording1-continuous: [Errno 2] No such file or directory: '/home/spk/spikeiterface/20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/npz.json'
$ ls  20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/*
20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/npz.json         20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/si_folder.json
20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/provenance.json  20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/sorting_cached.npz

20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/properties:

cd 20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting/
$ cd 20230508T195033Z-recording1-continuous/sorting-workingdir/sorter_output/sorting
$ cat npz.json 
{
    "class": "spikeinterface.core.npzsortingextractor.NpzSortingExtractor",
    "module": "spikeinterface",
    "kwargs": {
        "file_path": "sorting_cached.npz"
    },
    "dumpable": true,
    "version": "0.97.1",
    "relative_paths": true,
    "annotations": {},
    "properties": {}
}
```

rat-h, 2023-06-26:
I found the source of the problem! In my processing script it was
```python
sorting            = ....
sorting_saved = sorting.save( ....
we = si.extract_waveforms( preproc_saved, sorting ....
```
If change to 
```python
sorting            = ....
sorting_saved = sorting.save( ....
we = si.extract_waveforms( preproc_saved, sorting_saved ....
```
everything works.

alejoe91, 2023-06-26:
Glad you figured it out!!! :)