SpikeInterface/spikeinterface #1747: Postprocessing to remove overlapping / double counted spikes

jazlynntan, 2023-06-27:
Hi, I was wondering if there is a function implemented to remove double-counted or overlapping spikes? Similar to the one described here: https://github.com/AllenInstitute/ecephys_spike_sorting/tree/master/ecephys_spike_sorting/modules/kilosort_postprocessing

Using Kilosort 3 on my neuropixel data, I often find overlapping spikes in traceview in phy similar to what is discussed here: https://github.com/MouseLand/Kilosort/issues/151

I suppose the postprocessing by removing spikes very close in time to each other can be done using the get_all_spike_trains() and information of the channels where the spikes were detected. However, I wanted to check if there is already an existing function for this.

Thanks!

samuelgarcia, 2023-06-28:
Hi @jazlynntan.
Are you looking for this function https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.curation.remove_duplicated_spikes

And more generaly the curation should give you some tools https://spikeinterface.readthedocs.io/en/latest/modules/curation.html

jazlynntan, 2023-06-28:
Yes looks like it! Thank you!

closed 2023-06-28 by jazlynntan