SpikeInterface/spikeinterface #1752: Changing the base path when loading waveform extractors

JoeZiminski, 2023-06-28:
Often we work between a Linux HPC and mounted drive on windows. This means the base folder path in `waveforms/sorting.json` and `waveforms/recording.json` can change, for example:

Linux: `/ceph/neuroinformatics/neuroinformatics/scratch/...`
Window: `X:\neuroinformatics\scratch\...`

Because the waveform extractor reads from the `sorting.json` or `recording.json` metadata, sometimes the paths are wrong if accessing data from a mounted drive, and data cannot be loaded without changing the metadata ([this line](https://github.com/SpikeInterface/spikeinterface/blob/08f8a55a44050e71343440dd4e0f3c2538d30306/src/spikeinterface/core/waveform_extractor.py#L159)). I think under the hood, the base extractors [expose the concept of a relative path](https://github.com/SpikeInterface/spikeinterface/blob/08f8a55a44050e71343440dd4e0f3c2538d30306/src/spikeinterface/core/base.py#L409), but this depends on the `relative_paths` field of the metadata, which I am not sure how to change (currently is `False` in our metadata).

It would be cool to expose a new argument on `load_waveforms()`, something like `override_base_path` which would allow you to specify a new base path that overrides the metadata base paths in `sorting.json`, `recording.json`. However as this change goes pretty deep into the code-base, I thought I'd raise an issue than try a PR as I guess there are many possible solutions. Another approach may be a simple convenience function like `update_metadata()` that allows you to convert all base paths in a spikeinterface output metadata. Anyways, happy to help with any solution if this is a problem potentially widespread enough to be worth addressing.

TomBugnon, 2023-07-04:
seconded

alejoe91, 2023-07-04:
@JoeZiminski do you use `use_relative_path=True` when extracting waveforms? This would set the `relative_paths` in the dictionary. However, I think that it may still fail from moving from UNIX to Windows and giving the option to override base paths could be a good idea

TomBugnon, 2023-07-04:
The `extract_waveforms` doc says
```
use_relative_path: bool

    If True, the recording and sorting paths are relative to the waveforms folder. This allows portability of the waveform folder provided that the relative paths are the same, but forces all the data files to be in the same drive. Default is False
 ```
 
So another aspect is that if the sorting and the recording are not on the same drive (which is usual if the recording originates from raw data), then the sorting either can't be saved with relative path, and once the sorting/waveform are moved the waveforms can't be loaded straightforwardly without editing sorting.json  (even if it's the recording but not the sorting that prevented using relative paths)

So yes, flexible base paths during loading for sorting and/or recording would be great!

alejoe91, 2023-07-04:
@TomBugnon I think that the docs is unclear! What I mean is that when moving the waveform folder, one needs to also move the recording and sorting files location accordingly (not that the recording and sorting locations needs to be the same)

alejoe91, 2023-07-04:
Note that you can also do the following:

```
# load recording/sorting on Windows
sorting = si.load_extractor("C://path-to-sorting-folder")
recording = si.load_extractor("D://path-to-recording-folder")

# load waveforms without recording and provide sorting object
we = si.load_waveforms("path-to-waveforms", sorting=sorting, with_recording=False)

# set recording
we.set_recording(recording)
```

Would this fit your use cases @JoeZiminski  @TomBugnon ? This is even more flexible than playing around with relative paths, which can be tricky!

TomBugnon, 2023-07-04:
Actually it seems that if the  recording is on a different drive from the waveforms, then it is saved with absolute paths even when use_relative_paths=True... so all good on my end!

alejoe91, 2023-07-04:
> Actually it seems that if the recording is on a different drive from the waveforms, then it is saved with absolute paths even when use_relative_paths=True... so all good on my end!

Yes, and that's because there is no relative path, so the `relative_to` function returns the absolute path ;)

JoeZiminski, 2023-07-04:
Thanks! 99% sure `relative_path` will fix the problem, will double check and confirm!

JoeZiminski, 2023-07-05:
Thanks @alejoe91! `use_relative_path` on `extract_waveforms` was exactly what I was looking for

closed 2023-07-05 by JoeZiminski