SpikeInterface/spikeinterface #1757: PCA computation : pickle run out of inputs

maxjuv, 2023-06-29:
Hello,
I used to get some 'shared memory issues" (#1704 ) to extract my waveforms. Since the PR #1704  has been merged, I updated my Spikeinterface. Now the waveform extraction goes well, but I have big issue with the PCAs computation. I compute the PCA with default parameters: `pca = si.compute_principal_components(we, n_jobs=job_kwargs['n_jobs'])`

Python crashes only when I compute PCA from several sortings in parallel. The associated error is  the following:
```
Traceback (most recent call last):
  File "/home/users/j/juventin/yggdrasil_python_envs/py310/lib/python3.10/site-packages/joblib/externals/loky/process_executor.py", line 436, in _process_worker
    r = call_item()
  File "/home/users/j/juventin/yggdrasil_python_envs/py310/lib/python3.10/site-packages/joblib/externals/loky/process_executor.py", line 288, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "/home/users/j/juventin/yggdrasil_python_envs/py310/lib/python3.10/site-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/home/users/j/juventin/yggdrasil_python_envs/py310/lib/python3.10/site-packages/joblib/parallel.py", line 263, in __call__
    return [func(*args, **kwargs)
  File "/home/users/j/juventin/yggdrasil_python_envs/py310/lib/python3.10/site-packages/joblib/parallel.py", line 263, in <listcomp>
    return [func(*args, **kwargs)
  File "/home/users/j/juventin/spike_sorters/spikeinterface/src/spikeinterface/postprocessing/principal_component.py", line 727, in partial_fit_one_channel
    pca_model = pickle.load(fid)
EOFError: Ran out of input
```
When I compute PCA for one single sorting, it runs smoothly.

I noticed that a "tmp" folder was created during PCA computation. Since the name of the "tmp" folder is not unique,  I suspect that when PCAs are computed from multiple sortings at the same time, the "tmp" folder is overwritten, leading to the error. . Does this sound possible ?

Cheers 
Maxime

alejoe91, 2023-06-29:
Hi Maxime! Sounds very plausible indeed. Would you have time look into the code and maybe suggesting a fix?

maxjuv, 2023-06-29:
Yep, I will try.

PR #1762 (2023-07-04): Pass tmp_folder as parameter for `compute_principal_components`
Hello,
Here is a proposal to fix #1757. I added the possibilty to specify a tmp_folder in the parameter of the PrincipalComponent object. By default is None. 
It works for me, on my local Linux.
Cheers
Maxime