SpikeInterface/spikeinterface #1758: SpikeSortingError: Spike sorting failed. You can inspect the runtime trace

MaximeSoula, 2023-06-29:
Dear all, dear Samuel, 
I have issues regarding the run of sorters of ironclust and kilosort2 : 

**The code is the following :** 

sorting_ironclust_path = "/home/msoula/Analyse EEG Jupyter/sorting_ironclust.dill"
from spikeinterface.sorters import IronClustSorter
IronClustSorter.set_ironclust_path("/home/msoula/ironclust/")

res = cmr_recording
sorter_params={}
sorter_params={'job_kwargs': {'n_jobs': nb_jobs}}

sorting_ironclust = ss.run_sorter('ironclust', 
                          res, 
                          output_folder =  "/home/msoula/Analyse EEG Jupyter/", 
                          verbose=True)

with open(sorting_ironclust_path, 'wb') as f:
    pickle.dump(sorting_ironclust, f)

**And the error says :** 

Error running ironclust
Traceback (most recent call last):
  File "/home/msoula/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/home/msoula/.local/lib/python3.8/site-packages/spikeinterface/sorters/external/ironclust.py", line 249, in _run_from_folder
    raise Exception('ironclust returned a non-zero exit code')
Exception: ironclust returned a non-zero exit code

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[18], line 9
      5 sorter_params={}
      6 sorter_params={'job_kwargs': {'n_jobs': nb_jobs}}
----> 9 sorting_ironclust = ss.run_sorter('ironclust', 
     10                           res, 
     11                           output_folder =  "/home/msoula/Analyse EEG Jupyter/", 
     12                           verbose=True)
     14 with open(sorting_ironclust_path, 'wb') as f:
     15     pickle.dump(sorting_ironclust, f)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:163, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
    162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 163 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    164 if with_output:
    165     sorting = SorterClass.get_result_from_folder(output_folder)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py:262, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    260 if has_error and raise_error:
    261     print(log['error_trace'])
--> 262     raise SpikeSortingError(
    263         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    265 return run_time

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in /home/msoula/Analyse EEG Jupyter/spikeinterface_log.json

**So I have inspected the log.json files, and that is the message that I have :** 

{
    "sorter_name": "ironclust",
    "sorter_version": "5.9.8",
    "datetime": "2023-06-29T15:35:32.122441",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/home/msoula/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/home/msoula/.local/lib/python3.8/site-packages/spikeinterface/sorters/external/ironclust.py\", line 249, in _run_from_folder\n    raise Exception('ironclust returned a non-zero exit code')\nException: ironclust returned a non-zero exit code\n",
    "run_time": null
}

Do you have any idea why am I stuck here ?
I thank you deeply in advance ! 
Best regards

samuelgarcia, 2023-06-29:
Hi Maxime
There is an internal error in ironclust. Whihch is matlab based.
This a bit hard to track but not impossible.
You you have a "run_ironclust.m" file in your folder. Open matlab and try to run it you will get a better error messsage.

Also you should not do this
```python
with open(sorting_ironclust_path, 'wb') as f:
    pickle.dump(sorting_ironclust, f)
```
but this
```python
sorting_saved = sorting_ironclust.save(folder=sorting_ironclust_path)
# and then later
sorting_saved = load_xtractor(sorting_ironclust_path)

```

MaximeSoula, 2023-07-03:
Thank you Samuel, there was indeed a problem with matlab on my servor,  I could fixe it.
Thank you also for your tip about sorting_saved.

I have an new issue tho with the function compare_multiple_sorters.
I haven't see my issue anywhere in the forum so I am opening a new topic.
Thank you again for your fast answer ! 
Best, 
Maxime

closed 2023-07-03 by alejoe91