SpikeInterface/spikeinterface #1759: No Probe attached Error when using extract_waveforms in mode folder.

florian6973, 2023-06-29:
Hi!
I think I am facing a bug because I try to extract some waveforms from a recording which does not have a probe, and I get this error

[image]

The cause seems to be that the waveforms extractor checks if there is a probe to save as json, but the `BaseRecording.getprobegroup` method returns the value error if there is no probe.

Besides, it works well with in memory extraction, but this is not what I need.

MWE:
```python
import numpy as np
import spikeinterface as si

# Create synthetic data
num_channels = 4
sampling_frequency = 10000
duration = 1.0
num_samples = int(sampling_frequency * duration)
data = np.random.randn(num_samples, num_channels)

# Create a recording extractor
recording = si.NumpyRecording(traces_list=data, sampling_frequency=sampling_frequency)

# Create synthetic spike sorting results
spike_frames = np.array([100, 300, 500, 700])
spike_labels = np.array([0, 1, 1, 0])

# Create a sorting extractor
sorting = si.NumpySorting.from_times_labels(
    [spike_frames], [spike_labels], sampling_frequency=sampling_frequency
)
# It is not dumpable and does not make sense to save it with extract_waveforms
# but this is not the point of this example

# Extract waveforms for the first sorted spike times
waveforms = si.extract_waveforms(recording, sorting, "test", overwrite=True)
```

Thanks for your help!

alejoe91, 2023-07-03:
@florian6973 the `WaveformExtractor` relies on channel locations in many instances. I recommend setting some dummy locations in this case:

```
recording.set_channel_locations([[0, 1], [0, 2], [0, 3], [0, 4]])
```

alejoe91, 2023-07-03:
right! or allow WaveformExtraction without probe, at the user's risk. @samuelgarcia ?

PR #1777 (2023-07-06): Check probe in waveform extractor and add `has_probe` method
Fixes #1759 by checking for the probe for every `mode`
closed 2023-07-06 by samuelgarcia