SpikeInterface/spikeinterface #1760: Error when computing quality metrics for Kilosort 2.5

oghenand, 2023-06-30:
Hi, I'm currently trying to compute quality metrics for kilosort 2.5 sorting and waveform results. However, I get an odd error (pasted below) when computing the metrics. The function works fine for other sorters (such as Kilosort3). Upon inspection, it seems as though the function returns an error when computing the isi violations. I tried to bypass calculating this metric and export to phy, but I get another error when trying to open the gui. Is there any fix to the original bug when trying to compute the quality metrics? I feel that if I can fix that, my export to phy should work correctly and I should be able to view the results in the gui. 

Error: 
IndexError                                Traceback (most recent call last)
Cell In[13], line 1
----> 1 qc_ks25_hour = compute_quality_metrics(ks25_hour_we_sparse)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\qualitymetrics\quality_metric_calculator.py:202, in compute_quality_metrics(waveform_extractor, load_if_exists, metric_names, qm_params, peak_sign, seed, sparsity, skip_pc_metrics, verbose, **job_kwargs)
    199     qmc = QualityMetricCalculator(waveform_extractor)
    200     qmc.set_params(metric_names=metric_names, qm_params=qm_params, peak_sign=peak_sign, seed=seed,
    201                    sparsity=sparsity, skip_pc_metrics=skip_pc_metrics)
--> 202     qmc.run(verbose=verbose, **job_kwargs)
    204 metrics = qmc.get_data()
    206 return metrics

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\core\waveform_extractor.py:1691, in BaseWaveformExtractorExtension.run(self, **kwargs)
   1690 def run(self, **kwargs):
-> 1691     self._run(**kwargs)
   1692     self._save(**kwargs)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\qualitymetrics\quality_metric_calculator.py:110, in QualityMetricCalculator._run(self, verbose, **job_kwargs)
    107 func = _misc_metric_name_to_func[metric_name]
    109 params = qm_params[metric_name] if metric_name in qm_params else {}
--> 110 res = func(self.waveform_extractor, **params)
    111 # QM with uninstall dependencies might return None
    112 if res is not None:

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\qualitymetrics\misc_metrics.py:412, in compute_sliding_rp_violations(waveform_extractor, bin_size_ms, window_size_s, exclude_ref_period_below_ms, max_ref_period_ms, contamination_values)
    409         spike_train = sorting.get_unit_spike_train(unit_id=unit_id, segment_index=segment_index)
    410         spike_train_list.append(spike_train)
--> 412     contamination[unit_id] = slidingRP_violations(spike_train_list, fs, duration, bin_size_ms, window_size_s,
    413                                                   exclude_ref_period_below_ms, max_ref_period_ms,
    414                                                   contamination_values)
    416 return contamination

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\qualitymetrics\misc_metrics.py:943, in slidingRP_violations(spike_samples, sample_rate, duration, bin_size_ms, window_size_s, exclude_ref_period_below_ms, max_ref_period_ms, contamination_values, return_conf_matrix)
    941 correlogram = None
    942 for spike_samples in spike_samples_list:
--> 943     c0 = correlogram_for_one_segment(spike_samples, np.zeros(len(spike_samples), dtype='int8'),
    944                                      bin_size=max(int(bin_size_ms / 1000 * sample_rate), 1), # convert to sample counts
    945                                      window_size=int(window_size_s * sample_rate))[0, 0]
    946     if correlogram is None:
    947         correlogram = c0

IndexError: index 0 is out of bounds for axis 0 with size 0

Thanks!

alejoe91, 2023-07-03:
Hi @oghenand 

I think this is because some units are empty! But yeah it should not throw an error.

Can you try to do this before computing waveforms?
```
print(sorting)
sorting_without_empty = sorting.remove_empty_units()
print(sorting_without_empty)
```

Cheers,
Alessio

JoeZiminski, 2023-07-04:
Funnily enough I was just about to open a PR for this issue. In my case at least, this is because the unit is empty as you suggest. I was going to suggest a check-and-skip in the loops  in `compute_isi_violations` and `compute_sliding_rp_violations` e.g.

```
        for segment_index in range(num_segs):
            spike_train = sorting.get_unit_spike_train(unit_id=unit_id, segment_index=segment_index)
            if np.any(spike_train):
                spike_train_list.append(spike_train)

        if not np.any(spike_train_list):
            continue
```
The problem with this is the `quality_metrics.csv` contains empty / nonsensical rows which could be confusing:

[image]

Maybe when `compute_quality_metrics()` is run, a quick check on empty-units can be performed and a warning raised indicating the unit numbers, suggest remove_empty_units() and mention the quality-checks for these units will not make sense. Happy to open a PR!

JoeZiminski, 2023-07-04:
BTW I will plan to use `remove_empty_units()` systematically after sorting from now on - is there any reason not to?

alejoe91, 2023-07-04:
I use it all the time too! Maybe we should add an option to run_sorter to automatically do it?

alejoe91, 2023-07-05:
@JoeZiminski it'd be good to get a PR to better handle empty units in the quality metrics computation (and raise a warning!)

Note that we're planning to release on Friday, so if you could make on before it would be great!!!

samuelgarcia, 2023-07-05:
We could force emty units to compute metrics and then add then back in the dataframe when nan no ?

alejoe91, 2023-07-05:
@samuelgarcia honestly I don't see the point of carrying empty units around. They are useless and make everything complicated! I would add an option to the `run_sorter` to automatically remove them!

JoeZiminski, 2023-07-05:
Thanks I've opened a PR to adress the issue. The option on `run_sorter` to automatically remove empty units would be handy for me. Cheers!

PR #1783 (2023-07-06): Handle empty units in quality metrics
This PR closes #1760 by handling the empty-unit case in quality metrics's `compute_sliding_rp_violations` and `compute_isi_violations`. 

It also shows a warning if empty units exist when running `compute_quality_metrics` to indicate the values for these units are un-interpretable.

To ease this, two new conveience functions `get_non_empty_units` and `get_empty_units` are introduced on basesorter.

Happy to make any required changes!
closed 2023-07-06 by samuelgarcia