SpikeInterface/spikeinterface #1761: Removing to_dict breaks path_to_unix conversion on Windows in 0.98.0dev

florian6973, 2023-07-03:
hi!

Following this discussion about bug fix releases, https://github.com/SpikeInterface/spikeinterface/discussions/1755, I was trying the coming 0.98.0dev, but I cannot run spike sorters anymore on Windows through Docker.
Indeed, the paths in `in_container_recording.json` are no longer converted to unix format.
After a small investigation, I noticed that this function was not performing as it should https://github.com/SpikeInterface/spikeinterface/blame/main/src/spikeinterface/core/core_tools.py#L775
The cause is that at some point, `parent_recording` is not a dict but for instance a `spikeinterface.core.segmentutils.ConcatenateSegmentRecording`. Surprised, I checked and I found this PR https://github.com/SpikeInterface/spikeinterface/pull/1462
Therefore, is it expected to break this feature when removing the to_dict? Can I bring it back safely to core or should we update recursive_path_modifier?

Thanks

alejoe91, 2023-07-03:
Hi @florian6973 

Good catch! Yes recently we removed a bunch of `to_dict` to simplify passing objects around. Nevertheless, the recursive dict modifier needs the old behavior. I'll make a small PR. Would you mind testing it on Windows?

Cheers,
Alessio

alejoe91, 2023-07-03:
@florian6973 I double checked the code and it looks ok. We merged the PR that fixes this behavior last week.

In this line, the `recursive=True` for the `to_dict()` function _should_ behave as previously, by unrolling all parent objects into the dictionary.
https://github.com/SpikeInterface/spikeinterface/blame/main/src/spikeinterface/sorters/runsorter.py#L388

Can you make sure you have these changes integrated?

Thanks for testing! 
Alessio

florian6973, 2023-07-03:
Thanks for your quick reply!

Mea culpa, indeed everything is correct in the code, I found the issue. It is just that in the chain I forgot I was using a modified version of CommonReference (that removes the noise by substracting the closest k channels) and I had not updated it (I had to remove the to_dict in kwargs).

florian6973, 2023-07-03:
By the way if I was doing a PR for this CR feature would it be potentially accepted or not please ?

florian6973, 2023-07-03:
Thanks for your feedback! Yes sure it is clearly a side effect unfortunately, but hopefully it removes the local artifacts first and foremost...

Yes I did, the current issue with the 'local' mode is if there are no channels in the specified "doughnut". Let's say you have a probe where channels are not evenly spaced, and for one it has no channels in the specified range, it throws the error "No reference channels available in the local annulus for selection". However I do not wish to specify a too large annulus neither for other regions of the probe.
Would it be a good idea to support empty annulus please?

alejoe91, 2023-07-03:
@florian6973 I'll close this issue and we can discuss on your future PR ;)

closed 2023-07-03 by alejoe91

florian6973, 2023-07-03:
> With the empty annulus then you would not remove any median for that channel?

It would not be ideal but it seems to be the easiest option with the existing methods of common noise removal, given that only a few channels would be concerned. Of course the simplest idea would be to stick to `global` to have a consistent method across channels, but from what I have seen the results are not as good to remove artifacts.

Please let me know what you think best to implement and thanks again for all your help! :)

florian6973, 2023-07-03:
And related to noise removal, have you already considered Independent Component Analysis? 

In this paper they suggest removing the IC with uniform weight distribution, which is supposed to represent EMG noise in a ephys recording if I am not mistaken https://www.sciencedirect.com/science/article/pii/S2667237523001054?via%3Dihub