SpikeInterface/spikeinterface #1773: NEX file format

haelashqar, 2023-07-03:
I have NEX files that I am working on getting sorted using spike interface, but it appears NEX is not supported. Is that something that may be available in the near future? or can I convert the format feasibly to a different format that works with spike interface?

alejoe91, 2023-07-03:
Hi @haelashqar 

What acquisition system are these NEX files from?

alejoe91, 2023-07-03:
If it's Neuroexplorer we don't have an interface, but since it's implemented in NEO it should be straightforward to add one here!

haelashqar, 2023-07-03:
Correct, it’s neuroexplorer obtained with an alpha omega alpha system. Thanks!

alejoe91, 2023-07-06:
Pending, but the amount of code required is minimal! See this example: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/neoextractors/intan.py

h-mayorquin, 2023-09-06:
I opened a PR for this:

https://github.com/SpikeInterface/spikeinterface/pull/1965

There are two big limitations:
1) The current neo reader does not support the new nex5 format: https://www.neuroexplorer.com/nex5-file-format/
2) The current neo reader only supports reading one channel at a time and the recording object in spikeinterface inherited that limitation. I left instructions in the docstring in how to overcome that limitation but we could do more. 

Those things could be solved but right now we don't have the resources. That is, they won't probably be solved unless someone from outsides solves it or CatalystNeuro gets a contract to do a conversion that uses this format.

@haelashqar 
Is your data in the old or the new format?
If your data is in the old format the current PR should be enough for you, with the trick that I described in the docstring.
If your data is in the new format and it fits into memory / ram. Then you can use the [reading functions ](https://www.neuroexplorer.com/downloadspage/)provided by neuroexplorer, load your data as numpy and then use our `NumpyRecording` to work from there.
If your data is in the new format but it does not fit into memory / ram then we are out of luck as you will need to extract the data, in binary (using the same functions provided by neuro explorer) and figure out how to store the concatenate data across the channels as a binary so can use our `BinaryRecordingExtractor` to import to spikeinterface.

Maybe there is a better way if you find yourself in the third situation but I am not sure.

h-mayorquin, 2023-09-13:
@haelashqar the PR is merging and the spikeinterface version on main should load some nex files. See the caveats above.

zm711, 2024-02-06:
@haelashqar, have you been able to test this?

alejoe91, 2025-06-06:
Closing because it's implemented. There is another issue for nex5

closed 2025-06-06 by alejoe91