SpikeInterface/spikeinterface #1774: compare_multiple_sorters

MaximeSoula, 2023-07-03:
Dear Samuel, 

It is more a question rather than an issue : 

When using the function Compare_multiple_sorters, it seems we're using the results from the initial sorter instead of using all the data that was processed after that. Could it be that all of this data is used when using the compare function or should I do something else to take all of those steps into account? 

Because the last manipulation in the tutorial "how to guide" is the sorting_curated_auto part : 

 keep_mask = (qm['snr'] > 10) & (qm['isi_violations_ratio'] < 0.01)
print("Mask:", keep_mask.values)
sorting_curated_auto = sorting_TDC.select_units(sorting_TDC.unit_ids[keep_mask])
print(sorting_curated_auto)

And on the function compare_multiple_sorters, we just use the variable sorting_TDC, sorting_SC2 and sorting_KS2 as parameters while they have never been changed or updated :

comp_multi = sc.compare_multiple_sorters(sorting_list=[**sorting_TDC**, **sorting_SC2**, **sorting_KS2**],
                                         name_list=['tdc', 'sc2', 'ks2'])

It is not really clear for me.

Thank you again for your answer ! 
Best regards
Maxime

alejoe91, 2023-07-03:
Hi Maxime,

Yes in this example we are giving the raw spike sorting output, but you can pass any sorting object, like the `sorting_curated_auto`! That is just an example, feel free to try out different options!

MaximeSoula, 2023-07-04:
Hi Alessio, thank you for your fast answer !
I don't know if I should put it in a new forum, but this is what happen when I use the sorting_curated_auto in the compare_multiple_sorters : 

 
type(sorting_kilosort2)
spikeinterface.extractors.phykilosortextractors.KiloSortSortingExtractor

type(sorting_curated_kilosort2)
spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting

comp_multi_intersec2 = sc.compare_multiple_sorters(sorting_list= [sorting_curated_kilosort2, sorting_curated_ironclust, sorting_curated_SpykingCircus, sorting_curated_TDC2], name_list=['kilosort2', 'ironclust', 'spykingcircus', 'tdc2'], spiketrain_mode='intersection', verbose =True, n_jobs =1) 

Multicomaprison step 1: pairwise comparison
  Comparing: kilosort2 and ironclust

---------------------------------------------------------------------------
UFuncTypeError                            Traceback (most recent call last)
/tmp/ipykernel_18502/2866959119.py in <module>
----> 1 comp_multi_intersec2 = sc.compare_multiple_sorters(sorting_list= [sorting_curated_kilosort2, sorting_curated_ironclust, sorting_curated_SpykingCircus, sorting_curated_TDC2],
      2                                          name_list=['kilosort2', 'ironclust', 'spykingcircus', 'tdc2'], spiketrain_mode='intersection', verbose =True, n_jobs =1) 

~/.local/lib/python3.8/site-packages/spikeinterface/core/core_tools.py in reader_func(*args, **kwargs)
     24     @copy_signature(source_class)
     25     def reader_func(*args, **kwargs):
---> 26         return source_class(*args, **kwargs)
     27 
     28     reader_func.__doc__ = source_class.__doc__

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/multicomparisons.py in __init__(self, sorting_list, name_list, delta_time, match_score, chance_score, n_jobs, spiketrain_mode, verbose, do_matching)
     65 
     66         if do_matching:
---> 67             self._compute_all()
     68             self._populate_spiketrains()
     69 

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/basecomparison.py in _compute_all(self)
     42 
     43     def _compute_all(self):
---> 44         self._do_comparison()
     45         self._do_graph()
     46         self._clean_graph()

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/basecomparison.py in _do_comparison(self)
    100                 if self._verbose:
    101                     print(f"  Comparing: {name_i} and {name_j}")
--> 102                 comp = self._compare_ij(i, j)
    103                 self.comparisons[(name_i, name_j)] = comp
    104 

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/multicomparisons.py in _compare_ij(self, i, j)
     69 
     70     def _compare_ij(self, i, j):
---> 71         comp = SymmetricSortingComparison(self.object_list[i], self.object_list[j],
     72                                           sorting1_name=self.name_list[i],
     73                                           sorting2_name=self.name_list[j],

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/paircomparisons.py in __init__(self, sorting1, sorting2, sorting1_name, sorting2_name, delta_time, sampling_frequency, match_score, chance_score, n_jobs, verbose)
    114                  delta_time=0.4, sampling_frequency=None, match_score=0.5, chance_score=0.1,
    115                  n_jobs=-1, verbose=False):
--> 116         BasePairSorterComparison.__init__(self, sorting1, sorting2, sorting1_name=sorting1_name,
    117                                           sorting2_name=sorting2_name,
    118                                           delta_time=delta_time,

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/paircomparisons.py in __init__(self, sorting1, sorting2, sorting1_name, sorting2_name, delta_time, match_score, chance_score, n_jobs, verbose)
     36 
     37 
---> 38         self._do_agreement()
     39         self._do_matching()
     40 

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/paircomparisons.py in _do_agreement(self)
     65 
     66         # matrix of  event match count for each pair
---> 67         self.match_event_count = make_match_count_matrix(self.sorting1, self.sorting2, self.delta_frames,
     68                                                          n_jobs=self.n_jobs)
     69 

~/.local/lib/python3.8/site-packages/spikeinterface/comparison/comparisontools.py in make_match_count_matrix(sorting1, sorting2, delta_frames, n_jobs)
    143             delayed(count_match_spikes)(sorting1.get_unit_spike_train(u1, segment_index=segment_index), 
    144                                         s2_spiketrains, delta_frames) for i1, u1 in enumerate(unit1_ids))
--> 145         match_event_counts += np.array(match_event_count_segment)
    146 
    147     match_event_counts_df = pd.DataFrame(np.array(match_event_counts),

UFuncTypeError: Cannot cast ufunc 'add' output from dtype('float64') to dtype('int64') with casting rule 'same_kind'

Do you understand by any chance why it is not working for me ?

Thank you again for your help, 
Best, 
Maxime

MaximeSoula, 2023-07-04:
Thank you very much for your answer

MaximeSoula, 2023-07-12:
We can close it

closed 2023-07-12 by MaximeSoula