SpikeInterface/spikeinterface #1778: tridesclous and spikeinterface have conflicting dependencies

RikkelBob, 2023-07-04:
The newest (git) version of spikeinterface seems to be incompatible with tridesclous because of a numpy conflict

From pipdeptree:

```
tridesclous==1.6.6.1
   numpy [required: <1.23, installed: 1.24.0]

sortingview==0.11.15
   pynwb
      pandas [required: >=1.1.5, installed: 2.0.3]
         numpy [required: >=1.23.2, installed: 1.24.0]
```

alejoe91, 2023-07-04:
@RikkelBob 

Have you tried to use the docker image? 

https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#running-sorters-in-docker-singularity-containers

RikkelBob, 2023-07-04:
@alejoe91 Yes the docker image works fine, but trying to load sorting results afterwards still requires tridesclous to be installed locally

alejoe91, 2023-07-04:
I see! @RikkelBob I recommend using the `save()` function to simplify the pipeline:
```
sorting_tdc = ss.run_sorter("tridesclous", recording, docker_image=True)
sorting_tdc = sorting_tdc.save(folder="tdc_output")

# in another session
sorting_tdc = si.load_extractor("tdc_output")
```

This way you can foget about tridesclous and its dependencies ;)

samuelgarcia, 2023-07-04:
Hi @RikkelBob.
I will try to make a release on tridesclous to change the dependencies constrains.

RikkelBob, 2023-07-06:
@alejoe91 Perfect, that works! In the [guide](https://spikeinterface.readthedocs.io/en/latest/how_to/analyse_neuropixels.html), saving is done in a different way. Might be worth it to mention this way of saving too.

```
sorting = si.run_sorter('kilosort2_5', rec, output_folder=base_folder / 'kilosort2.5_output',
                        docker_image=True, verbose=True, **params_kilosort2_5)
```

@samuelgarcia Thanks!

closed 2023-07-06 by RikkelBob