SpikeInterface/spikeinterface #1784: Kilosort 2  returned a non-zero exit code in Spike Interface

farazmoradi, 2023-07-05:
hi,

I have tried to do spike sorting with Kilosort 2 and received that error in the spikeinterface_log.jason "  raise Exception(f'{cls.sorter_name} returned a non-zero exit code')\nException: kilosort2 returned a non-zero exit code\n ". I receive this error even after MATLAB starts and I can see the plot of cluster amplitudes and numbers. 

The thing that is strange is it is working on some recordings from the same probe and animal, but for some it doesn't. Even while recording we were inspecting the recordings and everything looks fine.

alejoe91, 2023-07-06:
Hi @farazmoradi 

Then the error seems to be internal to KS2. Can you open an issue there?

Cheers,
Alessio

TsungChihTsai, 2023-11-22:
> hi,
> 
> I have tried to do spike sorting with Kilosort 2 and received that error in the spikeinterface_log.jason " raise Exception(f'{cls.sorter_name} returned a non-zero exit code')\nException: kilosort2 returned a non-zero exit code\n ". I receive this error even after MATLAB starts and I can see the plot of cluster amplitudes and numbers.
> 
> The thing that is strange is it is working on some recordings from the same probe and animal, but for some it doesn't. Even while recording we were inspecting the recordings and everything looks fine.
> 
> Thanks

I found our lab has similar problem in Kilosort3 (in docker environment). Sometime we can use, sometime it doesn't work. Do you still have this kind of problem?

alirza67, 2024-04-07:
I am also having this problem with both ks2 and ks3 when I use them in spikeinterface.
without spikeinterface, they work just fine!

TsungChihTsai, 2024-04-07:
[alirza67](https://github.com/alirza67) How to use KS2 or KS3 sorter without spikeinterface? Could you please send me the website link for this?

alirza67, 2024-04-07:
> [alirza67](https://github.com/alirza67) How to use KS2 or KS3 sorter without spikeinterface? Could you please send me the website link for this?

To use KS2 and 3, you first need Matlab installed on your PC. 
Then you can download KSx here: https://github.com/MouseLand/Kilosort/releases
There are lots of tutorial on how to use it. for instance: https://www.youtube.com/watch?v=trYXdZi-gOk

Happy sorting!

alejoe91, 2024-04-07:
@alirza67 Can you paste here the content of the spikinterface log file in the output folder?

alirza67, 2024-04-07:
> @alirza67 Can you paste here the content of the spikinterface log file in the output folder?

Thanks for your reply!
here is the content of "spikeinterface_log.json":
{
    "sorter_name": "kilosort2",
    "sorter_version": "unknown",
    "datetime": "2024-04-07T00:53:57.272703",
    "runtime_trace": [
        "",
        "E:\\py_script>D:",
        "",
        "D:\\>cd D:\\ephys\\R002\\2024-03-12\\KS2out\\sorter_output",
        "",
        "D:\\ephys\\R002\\2024-03-12\\KS2out\\sorter_output>matlab -nosplash -wait -r \"kilosort2_master('D:\\ephys\\R002\\2024-03-12\\KS2out\\sorter_output', 'E:\\mathwork\\analysis\\Kilosort-2.0')\""
    ],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"\\\\srv9021\\saeedi\\spikeinterface\\src\\spikeinterface\\sorters\\basesorter.py\", line 258, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"\\\\srv9021\\saeedi\\spikeinterface\\src\\spikeinterface\\sorters\\external\\kilosortbase.py\", line 217, in _run_from_folder\n    raise Exception(f\"{cls.sorter_name} returned a non-zero exit code\")\nException: kilosort2 returned a non-zero exit code\n",
    "run_time": null
}

zm711, 2024-06-21:
Based on my extensive experience with KS I will predict that this is due to a CUDA error. If there are certain regions of a recording that have to many 0s CUDA errors occur. This leads to failure of some recordings. Some users of KS have suggested solutions of adding `try-catch` into the code to try to rerun. It works for some, but not always. It might be worth searching through the closed issues on the KS repo to find this (since they closed all matlab issues).

closed 2024-06-21 by zm711