SpikeInterface/spikeinterface #1785: Saving deepinterpolation denoised file as binary

jazlynntan, 2023-07-06:
Hello,

I managed to run deepinterpolation and the time series plot looks good. However, when I try to save it as a .dat binary file to run on my local installation of Kilosort3, I get the error message: TypeError: __init__() got an unexpected keyword argument 'recording'. I get the same error when running Kilosort3 on the denoised recording using the singularity container directly.

I run deepinterpolation via: 
`import spikeinterface.preprocessing as sp`
`rec5 = sp.deepinterpolate(sp.zero_channel_pad(rec4,384), model_path = '2020_02_29_15_28_unet_single_ephys_1024_mean_squared_error-1050.h5')` 
where rec4 is the output of filtering, removing bad channels and common average referencing, following the code of the Neuropixel tutorial. The zero channel padding is because I removed a bad channel (the reference channel). I have the model downloaded and the path is correct.

Regarding saving, I tried:
`job_kwargs = dict(n_jobs=12, chunk_duration='1s', progress_bar=True)`
`rec = rec5.save(folder='myfolder', format='binary', **job_kwargs)`
which gave the same TypeError: __init__() got an unexpected keyword argument 'recording'

as well as:
`import spikeinterface.core as sc`
`job_kwargs = dict(chunk_duration='1s', progress_bar=True)`
`sc.write_binary_recording(rec5, file_paths = 'myfolder/deepinterpolation_denoised.dat', dtype='int16', n_jobs=12, **job_kwargs)`
which again gave the same error.

May I know if I was running deepinterpolation wrongly? Also, is it recommended to save the output as a binary .dat file via the recording.save() method or the write_binary_recording() method? I attempted to save my filtered recording (without deepinterpolation denoising) via the .save() method but it gave a .raw file that my local installation of Kilosort3 did not accept (error message that the input was not binary).

Thank you!

alejoe91, 2023-07-06:
Hi, can you paste the full error?

For running KS3, I recommend doing it via SpikeInterface since our wrapper automatically maps all inputs correctly ;) 

`sorting_KS3 = ss.run_sorter("kilosort3", recording)`

For saving to binary, we recommend using the `save()` function, which also saves a bunch of metadata and allows you to reload the recording/sorting with the `load_extractor()` function

alejoe91, 2023-07-06:
Thanks. Can you try to install SpikeInterface from the `main` branch? I think this issue might be fixed!

jazlynntan, 2023-07-13:
Hi, thanks for your quick response, I tried running the sorting but experienced a different error instead. This is the full output and error message:

Starting container
Installing spikeinterface==0.98.0 in spikeinterface/kilosort3-compiled-base
Installing extra requirements: ['neo', 'tensorflow']
Running kilosort3 sorter inside spikeinterface/kilosort3-compiled-base
Stopping container
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[36], line 3
      1 import spikeinterface.sorters as ss
      2 # si.get_default_sorter_params('kilosort3')
----> 3 sorting = ss.run_sorter('kilosort3', rec5_removed, output_folder='/data/projects/neuropixel_jt/test_kilosort3_output',
      4                         singularity_image=True, verbose=True)

File ~/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:142, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    140         else:
    141             container_image = singularity_image
--> 142     return run_sorter_container(
    143         container_image=container_image,
    144         mode=mode,
    145         **common_kwargs,
    146     )
    148 return run_sorter_local(**common_kwargs)

File ~/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:595, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, **sorter_params)
    593 if run_error:
    594     if raise_error:
--> 595         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    596 else:
    597     if with_output:

SpikeSortingError: Spike sorting in singularity failed with the following error:
2023-07-13 03:43:08.085362: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-13 03:43:08.593005: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
2023-07-13 03:43:09.080342: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:995] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero. See more at https://github.com/torvalds/linux/blob/v6.0/Documentation/ABI/testing/sysfs-bus-pci#L344-L355
2023-07-13 03:43:09.185553: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1960] Cannot dlopen some GPU libraries. Please make sure the missing libraries mentioned above are installed properly if you would like to use GPU. Follow the guide at https://www.tensorflow.org/install/gpu for how to download and setup the required libraries for your platform.
Skipping registering GPU devices...
2023-07-13 03:43:09.595311: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:375] MLIR V1 optimization pass is not enabled
2023-07-13 03:43:09.628755: W tensorflow/c/c_api.cc:304] Operation '{name:'conv2d_8/kernel/Assign' id:209 op device:{requested: '', assigned: ''} def:{{{node conv2d_8/kernel/Assign}} = AssignVariableOp[_has_manual_control_dependencies=true, dtype=DT_FLOAT, validate_shape=false](conv2d_8/kernel, conv2d_8/kernel/Initializer/stateless_random_uniform)}}' was changed by setting attribute after it was run by a session. This mutation will have no effect, and will trigger an error in the future. Either don't modify nodes after running them or create a new session.
2023-07-13 03:43:09.820931: W tensorflow/c/c_api.cc:304] Operation '{name:'conv2d_9/bias/rms/Assign' id:506 op device:{requested: '', assigned: ''} def:{{{node conv2d_9/bias/rms/Assign}} = AssignVariableOp[_has_manual_control_dependencies=true, dtype=DT_FLOAT, validate_shape=false](conv2d_9/bias/rms, conv2d_9/bias/rms/Initializer/zeros)}}' was changed by setting attribute after it was run by a session. This mutation will have no effect, and will trigger an error in the future. Either don't modify nodes after running them or create a new session.
2023-07-13 03:43:20.596828: W tensorflow/c/c_api.cc:304] Operation '{name:'conv2d_7/kernel/Assign' id:179 op device:{requested: '', assigned: ''} def:{{{node conv2d_7/kernel/Assign}} = AssignVariableOp[_has_manual_control_dependencies=true, dtype=DT_FLOAT, validate_shape=false](conv2d_7/kernel, conv2d_7/kernel/Initializer/stateless_random_uniform)}}' was changed by setting attribute after it was run by a session. This mutation will have no effect, and will trigger an error in the future. Either don't modify nodes after running them or create a new session.
2023-07-13 03:43:20.769335: W tensorflow/c/c_api.cc:304] Operation '{name:'conv2d_5/kernel/rms/Assign' id:451 op device:{requested: '', assigned: ''} def:{{{node conv2d_5/kernel/rms/Assign}} = AssignVariableOp[_has_manual_control_dependencies=true, dtype=DT_FLOAT, validate_shape=false](conv2d_5/kernel/rms, conv2d_5/kernel/rms/Initializer/zeros)}}' was changed by setting attribute after it was run by a session. This mutation will have no effect, and will trigger an error [... cut ...]
    yield fs.pop().result()
  File "/home/miniconda3/lib/python3.8/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "/home/miniconda3/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
    raise self._exception
ValueError: all the input array dimensions except for the concatenation axis must match exactly, but along dimension 1, the array at index 0 has size 383 and the array at index 1 has size 384
write_binary_recording:   0%|          | 0/1561 [23:04<?, ?it/s]

alejoe91, 2023-07-13:
@jazlynntan I recently made a PR that refactors and updates the deepinterolation step. Can you test it out? 
#1804

jazlynntan, 2023-07-13:
Hi, I tried it out but can't get past the deepinterpolation step now. Am I doing something wrong?

My code is:
```
import spikeinterface.preprocessing as sp

rec1 = sp.highpass_filter(recording, freq_min=300.)
bad_channel_ids, channel_labels = sp.detect_bad_channels(rec1)
rec2 = rec1.remove_channels(bad_channel_ids)
print('bad_channel_ids', bad_channel_ids)

rec3 = sp.phase_shift(rec2)
rec4 = sp.common_reference(rec3, operator="median", reference="global")
rec5 = sp.deepinterpolate(sp.zero_channel_pad(rec4,384), model_path = '/data/projects/neuropixel_jt/2020_02_29_15_28_unet_single_ephys_1024_mean_squared_error-1050.h5')
```

rec1 - 4 are the preprocessing steps based on the Neuropixel tutorial. My recording is on the Neuropixel 1.0 phase 3B staggered configuration probe. There is one bad channel (the reference), hence while 'recording' has 384 channels, rec2 onwards has 383. I pad rec4 to obtain 384 channels and then run deepinterpolation on that. I'm using the pretrained model from deepinterpolation.

And the error:

```
AssertionError                            Traceback (most recent call last)
Cell In[13], line 10
      8 rec3 = sp.phase_shift(rec2)
      9 rec4 = sp.common_reference(rec3, operator="median", reference="global")
---> 10 rec5 = sp.deepinterpolate(sp.zero_channel_pad(rec4,384), model_path = '/data/projects/neuropixel_jt/2020_02_29_15_28_unet_single_ephys_1024_mean_squared_error-1050.h5')

File /data/projects/neuropixel_jt/spikeinterface/src/spikeinterface/preprocessing/deepinterpolation/deepinterpolation.py:80, in DeepInterpolatedRecording.__init__(self, recording, model_path, pre_frame, post_frame, pre_post_omission, batch_size, use_gpu, disable_tf_logger, memory_gpu)
     78 network_input_shape = model.get_config()["layers"][0]["config"]["batch_input_shape"]
     79 desired_shape = network_input_shape[1:3]
---> 80 assert (
     81     desired_shape[0] * desired_shape[1] == recording.get_num_channels()
     82 ), "The desired shape of the network input must match the number of channels in the recording"
     83 assert (
     84     network_input_shape[-1] == pre_frame + post_frame
     85 ), "The desired shape of the network input must match the pre and post frames"
     87 self.model = model

AssertionError: The desired shape of the network input must match the number of channels in the recording
```

alejoe91, 2023-07-13:
HI @jazlynntan 

The problem is that the model that you are using is padding the electrodes with interleaved zeros to obtain a shape of (384, 2).

Basically you have:

...
e2   e3
0     0
e0   e1

@jeromelecoq can confirm!

Is your `rec4` a Neuropixels recording? 

**NOTE** the PR is still in progress. I will add more docs and tutorials in the following week

jazlynntan, 2023-07-13:
The original recording is a Neuropixel 1.0 recording, but rec4 is a processed version with the reference channel removed (hence one less than the usual 384). I did pad it using `sp.zero_channel_pad(rec4,384)` so I thought it would be the correct shape for the model. 

Would you suggest I instead train deepinterpolation from scratch? If so, do you mind sharing what version of deepinterpolation you use?

Noted and thank you for the work on the PR. A tutorial on denoising with deepinterpolation would be really helpful!

alejoe91, 2025-06-06:
This seems to be fixed