SpikeInterface/spikeinterface #1800: Binary recording extractor swapped parameter order

colehurwitz, 2023-07-07:
It seems that there was a change recently where `num_chan` was removed from `binaryrecordingextractor` and `dtype` became the third parameter in the init.  This broke some of our code which was passing in `num_chan` before `dtype`. Just wanted to let you know since it seems like this was supposed to be a soft deprecation.

alejoe91, 2023-07-07:
Thanks @colehurwitz and sorry about that! I recommend always using learns ;)

h-mayorquin, 2023-07-10:
Sup Cole!

Sorry for this, I was aware of this when I did it, as the documentation required some changes in that direction. It is a good practice to not change the order of parameters and we should respect that as often as possible but we could not do it here because of the following:

In order for introducing a new keyword `num_channels` that could be ignored by older version of the code this needed to be an optional parameter. Because dtype is not optional that means it had to go after dtype. Same for the old `num_chan` which needed to become optional and therefore had to be moved. 

The only solutions would have been too deprecated the old `num_chan` entirely which was undesirable or make dtype=None possible which seemed wrong as well. Maybe there is a better one that I could not think off?

I also think that is a good practice not too rely on positional arguments. I think that using keyword values when you call a function is good for self-documentation and, if you are designing a function, should highlight a bad choice of variable name

h-mayorquin, 2023-08-31:
I am going to apologize again and close this issue @colehurwitz.

Maybe using keyword only arguments would have helped us avoiding this problem for future development.

https://peps.python.org/pep-3102/

closed 2023-08-31 by h-mayorquin