SpikeInterface/spikeinterface #1801: read_spikeglx no longer working

cwindolf, 2023-07-07:
After installing latest spikeinterface (and probeinterface and neo), loading one of the Steinmetz et al induced motion NP2 datasets with `si.read_spikeglx()` seems not to be working?

```
(b) charlie@bluebeard ~ % mamba list | grep -e interface -e neo
neo                       0.13.0.dev0              pypi_0    pypi
probeinterface            0.2.18                   pypi_0    pypi
spikeinterface            0.99.0.dev0              pypi_0    pypi
```

```
IndexError                                Traceback (most recent call last)
Cell In[7], line 2
----> 2 rec = si.read_spikeglx("/moto/stats/users/ciw2107/dataset1")
      3 # restrict to channels with activity in this recording
      4 rec = rec.channel_slice(rec.channel_ids[:300])

File ~/spikeinterface/src/spikeinterface/extractors/neoextractors/spikeglx.py:50, in SpikeGLXRecordingExtractor.__init__(self, folder_path, load_sync_channel, stream_id, stream_name, all_annotations)
     48 def __init__(self, folder_path, load_sync_channel=False, stream_id=None, stream_name=None, all_annotations=False):
     49     neo_kwargs = self.map_to_neo_kwargs(folder_path, load_sync_channel=load_sync_channel)
---> 50     NeoBaseRecordingExtractor.__init__(
     51         self, stream_id=stream_id, stream_name=stream_name, all_annotations=all_annotations, **neo_kwargs
     52     )
     54     # open the corresponding stream probe for LF and AP
     55     # if load_sync_channel=False
     56     if "nidq" not in self.stream_id and not load_sync_channel:

File ~/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:210, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    205         raise ValueError(
    206             f"This reader have several streams: \nNames: {stream_names}\nIDs: {stream_ids}. "
    207             f"Specify it with the 'stream_name' or 'stream_id' arguments"
    208         )
    209     else:
--> 210         stream_id = stream_ids[0]
    211         stream_name = stream_names[0]
    212 else:

IndexError: list index out of range
```

alejoe91, 2023-07-07:
@cwindolf can you try this?
`stream_names, stream_ids = se.get_neo_streams("spikeglx", path_to_folder)`

@h-mayorquin can you take a look?

cwindolf, 2023-07-07:
Very sorry, posted the wrong traceback above due to clipboard error. Updated with the correct traceback.

I have also found the problem, which is not due to changes in spikeinterface! I had made a subfolder in this directory and copied a preprocessed version with a .meta file into that subfolder. Closing since this is probably intended behavior

cwindolf, 2023-07-07:
Thanks for taking a look!

closed 2023-07-07 by cwindolf