SpikeInterface/spikeinterface #1802: `get_sorter_params_description` returns empty directory for spykingcircus2 and tridesclous2

rat-h, 2023-07-07:
```
>>> import spikeinterface.full as si
>>> for n in 'spykingcircus tridesclous spykingcircus2 tridesclous2 waveclus'.split():
...   print(n)
...   print(si.get_sorter_params_description(n))
...   print("==========================")
... 
spykingcircus
{'detect_sign': 'Use -1 (negative), 1 (positive) or 0 (both) depending on the sign of the spikes in the recording', 'adjacency_radius': 'Radius in um to build channel neighborhood', 'detect_threshold': 'Threshold for spike detection', 'template_width_ms': 'Template width in ms. Recommended values: 3 for in vivo - 5 for in vitro', 'filter': 'Enable or disable filter', 'merge_spikes': 'Enable or disable automatic mergind', 'auto_merge': 'Automatic merging threshold', 'num_workers': 'Number of workers (if None, half of the cpu number is used)', 'whitening_max_elts': 'Max number of events per electrode for whitening', 'clustering_max_elts': 'Max number of events per electrode for clustering'}
==========================
tridesclous
{'freq_min': 'High-pass filter cutoff frequency', 'freq_max': 'Low-pass filter cutoff frequency', 'detect_threshold': 'Threshold for spike detection', 'detect_sign': 'Use -1 (negative) or 1 (positive) depending on the sign of the spikes in the recording', 'common_ref_removal': 'remove common reference with median', 'n_jobs': 'Number of jobs (when saving ti binary) - default -1 (all cores)', 'chunk_size': 'Number of samples per chunk (when saving ti binary) - default global', 'chunk_memory': "Memory usage for each job (e.g. '100M', '1G') (when saving to binary) - default global", 'total_memory': "Total memory usage (e.g. '500M', '2G') (when saving to binary) - default global", 'chunk_duration': "Chunk duration in s if float or with units if str (e.g. '1s', '500ms') (when saving to binary) - default global", 'progress_bar': 'If True, progress bar is shown (when saving to binary) - default global'}
==========================
spykingcircus2
{}
==========================
tridesclous2
{}
==========================
waveclus
{'detect_threshold': 'Threshold for spike detection', 'detect_sign': 'Use -1 (negative), 1 (positive), or 0 (both) depending on the sign of the spikes in the recording', 'feature_type': 'wav (for wavelets) or pca, type of feature extraction applied to the spikes', 'scales': 'Levels of the wavelet decomposition used as features', 'min_clus': 'Minimum increase of cluster sizes used by the peak selection on the temperature map', 'maxtemp': 'Maximum temperature calculated by the SPC method', 'template_sdnum': 'Maximum distance (in total variance of the cluster) from the mean waveform to force a spike into a cluster', 'enable_detect_filter': 'Enable or disable filter on detection', 'enable_sort_filter': 'Enable or disable filter on sorting', 'detect_filter_fmin': 'High-pass filter cutoff frequency for detection', 'detect_filter_fmax': 'Low-pass filter cutoff frequency for detection', 'detect_filter_order': 'Order of the detection filter', 'sort_filter_fmin': 'High-pass filter cutoff frequency for sorting', 'sort_filter_fmax': 'Low-pass filter cutoff frequency for sorting', 'sort_filter_order': 'Order of the sorting filter', 'mintemp': 'Minimum temperature calculated by the SPC algorithm', 'w_pre': 'Number of samples from the beginning of the spike waveform up to (including) the peak', 'w_post': 'Number of samples from the peak (excluding it) to the end of the waveform', 'alignment_window': 'Number of samples between peaks of different channels', 'stdmax': 'The events with a value over this number of noise standard deviations will be discarded', 'max_spk': 'Maximum number of spikes used by the SPC algorithm', 'ref_ms': 'Refractory time in milliseconds, all the threshold crossing inside this period are detected as the same spike', 'interpolation': 'Enable or disable interpolation to improve the alignments of the spikes', 'keep_good_only': "If True only 'good' units are returned", 'chunk_memory': 'Chunk size in Mb to write h5 file (default 500Mb)'}
==========================
>>> 

```

alejoe91, 2023-11-06:
Seems fixed in main:
[image]

zm711, 2023-11-08:
@alejoe91,

@rat-h  was requesting the descriptions 
[image]

@yger and @samuelgarcia haven't added the descriptions which describe the defaults yet. Should we reopen?

samuelgarcia, 2023-11-08:
@yger and @samuelgarcia are two bad boys!

zm711, 2024-02-06:
Now fixed by #2267 & #2276.

closed 2024-02-06 by zm711