SpikeInterface/spikeinterface #1805: `recording.has_channel_location()` doesn't work like it's supposed to

DradeAW, 2023-07-10:
Hi,

I'm having a crash while trying to run Lussac's unit tests:

The code that makes it crash:
```python
print("lol1")
print(recording.get_probegroup())
print(recording.get_property_keys())
print(recording.has_channel_location())

sorting = ss.run_sorter("spykingcircus2", recording, **params['sorter_params'])
```

I also modified `waveform_extractor.extract_waveform` to add a print in the beginning:
```python
print("lol2")
print(recording.get_probegroup())
print(recording.get_property_keys())
print(recording.has_channel_location())
assert (
    recording.has_channel_location()
), "Recording must have a probe  or channel location to extract waveforms. Use the `set_probe()` or `set_dummy_probe_from_locations()` methods."
```

And here is the output of pytest:
```bash
$ pytest tests/core/test_spike_sorting.py
============================================================== test session starts ==============================================================
platform linux -- Python 3.11.2, pytest-7.4.0, pluggy-1.0.0
cachedir: tests/.pytest_cache
rootdir: /export/home1/users/nsr/wyngaard/Documents/lussac2
configfile: pyproject.toml
plugins: cov-4.1.0
collected 1 item                                                                                                                                

tests/core/test_spike_sorting.py F                                                                                                        [100%]

=================================================================== FAILURES ====================================================================
______________________________________________________________ test_spike_sorting _______________________________________________________________

data = <lussac.core.lussac_data.LussacData object at 0x7fb8c8f001c0>

    @pytest.mark.skipif(importlib.util.find_spec("hdbscan") is None, reason="Testing spike sorting requires Python package 'hdbscan'.")
    def test_spike_sorting(data: LussacData) -> None:
    	"""
    	Tests if spike sorting can be run through Lussac without any crash.
    	Creates a shorter recording (1 shank, 5 minutes) to go faster.
    	We don't really care about quality here (e.g. no preprocessing)
    	because we only want to check that the spike sorting can be run.
    	"""
    	start_frame, end_frame = 9_000_000, 18_000_000
    	name = "test_spike_sorting"
    
    	recording = data.recording.frame_slice(start_frame, end_frame)
    	recording = recording.channel_slice(np.arange(13, 26))
    	params = {
    		'sorter_name': "spykingcircus2",
    		'preprocessing': {
    			'filter': {'band': [150., 6000.], 'filter_order': 2, 'ftype': "bessel"},
    			'common_reference': {'operator': "median"}
    		},
    		'sorter_params': {
    			'detection': {'peak_sign': "neg", "detect_threshold": 7},
    			'apply_preprocessing': False,
    			'output_folder': f"{data.tmp_folder}/sc2_ss_test"
    		}
    	}
    
    	spike_sorter = LussacSpikeSorter(recording, name)
[... 113 lines cut ...]
  
  The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.

tests/core/test_spike_sorting.py::test_spike_sorting
  /export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:668: NumbaDeprecationWarning:
  
  The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.

tests/core/test_spike_sorting.py::test_spike_sorting
  /export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/sortingcomponents/clustering/sliding_nn.py:370: NumbaDeprecationWarning:
  
  The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.

tests/core/test_spike_sorting.py::test_spike_sorting
  /export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:228: UserWarning:recording.has_channel_location()
  
  There is no Probe attached to this recording. Creating a dummy one with contact positions

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
============================================================ short test summary info ============================================================
FAILED tests/core/test_spike_sorting.py::test_spike_sorting - spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
======================================================== 1 failed, 4 warnings in 18.22s =========================================================
```

As you can see, `recording.has_channel_location()` returns False once the `recording` object has been pre-processessed, but that shouldn't happen.

DradeAW, 2023-07-10:
I think that the error comes from here:
https://github.com/SpikeInterface/spikeinterface/blob/802743208e9240e9b86888ff2da7473e7a64c302/src/spikeinterface/core/base.py#L574

The `recording` object is dumped to json then reloaded, but the `contact_vector` is lost in this transaction because the properties are not included!

alejoe91, 2023-07-10:
Nope, check #1806 ;)

DradeAW, 2023-07-10:
Ah!

Thank you :)