SpikeInterface/spikeinterface #1811: question about reading Neuropixels data

Zeinab-Mohammadi, 2023-07-10:
Hi
 I am trying to read this data [http://data.cortexlab.net/singlePhase3/] (i.e. Hopkins_20160722_g0_t0.imec.ap_CAR.bin data file in [http://data.cortexlab.net/singlePhase3/data/](url) using spikeinference as I want to sort it with Mountainsort5.
I only need 10% of the data so if I can read in .npy format it would be better. I changed the data in Matlab to .npy format using the below code (as I need to read it in python):
`fn = '/Users/.../Hopkins_20160722_g0_t0.imec.ap_CAR.bin'; 
fid = fopen(fn, 'r'); 
dat = fread(fid, [385 Inf], '*int16'); 
fclose(fid); 
chanMap = readNPY('/Users/.../channel_map.npy');
T = 11300000; %interested length, about 1/10 of data
dat2 = dat(chanMap+1,1:T); 
writeNPY(dat2, 'data_short.npy')`
I had a look at the examples, but I need help. Reading the 10% of the data in either .npy or .bin format would be great. Thanks

h-mayorquin, 2023-07-11:
Hi, @Zeinab-Mohammadi 

There seems to be a problem with your link of your data. In fact, all of your links seem to be pointing to the wrong place. I wonder what happened.

As for your error, can you share the script that makes it fail?

It works on my computer and it works in the documentation which we run as code:

```

In [1]: import spikeinterface.full as si

In [2]: hasattr(si, "get_neo_streams")
Out[2]: True
``` 

So, we will need to take a look at the context to help you.

Zeinab-Mohammadi, 2023-07-11:
Hi @h-mayorquin 
Thanks and sorry about that, it should be fine now but lmk if it still has any problem. Also, I was doing import spikeinterface as si, so now that error is ok. 
As I mentioned this is a file after CAR, so I think I should go with the below code and I am getting this error:
[image]

Zeinab-Mohammadi, 2023-07-11:
Btw, do you know about a public dataset (synthetic or experimental data, both with many channels (>100) ) that is read/processed by spikeinterface and preferably sorted by MountainSort? For example in your website link is the data in this path publicly available? `base_folder = Path('/mnt/data/sam/DataSpikeSorting/neuropixel_example/')

spikeglx_folder = base_folder / 'Rec_1_10_11_2021_g0'`

Zeinab-Mohammadi, 2023-07-12:
Hi @h-mayorquin 
Thanks, but I think this is not true and it results in error. The Hopkins_20160722_g0_t0.imec.ap_CAR.bin data is in Neuropixels_data folder, but it can't print out the stream_names for some reason.
[image]

Btw, I used get_noise_levels as based on [this code](https://spikeinterface.readthedocs.io/en/latest/how_to/analyse_neuropixels.html), this data is after **CAR** as I mentioned, and is after transferred to **.bin** file. you can see this in the mentioned site as **check noise level** after transferring the file to **binary format**.
All of these are in the link I put in my first comment and here. Thanks

alejoe91, 2023-07-12:
Ok, so if the folder doesn't preserve the original spikeglx folder, you need to load the bin directly with the `read_binary` function. You cannot pass the path to the file directly to the functions, since they require a recording object.

Zeinab-Mohammadi, 2023-07-12:
only the name of the function is [here](https://spikeinterface.readthedocs.io/en/latest/modules/extractors.html), but is there an example somewhere?

alejoe91, 2023-07-12:
I just realized it's not in the API page, we'll add it! In the meanwhile you can find all arguments here: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/core/binaryrecordingextractor.py#L13

Zeinab-Mohammadi, 2023-07-12:
As there is no example, it's better if you can write the code to use it with my file. Thanks
i.e. how should I import and use it?

alejoe91, 2023-07-12:
Sure, but I also recommend try to read the "docstrings" that I shared because it's a useful skill to be able to use them!

```
recording = si.read_binary("path-to-file.bin", num_channels=385, dtype="int16", sampling_frequency=30000, gain_to_uV=2.34)
recording = recording.remove_channels([recording.channel_ids[-1]])
```

A couple of notes: 

- the recording you are loading has an extra "SYNC" channel, so the number of channels need to be set 385. So we need to remove the extra channel (`remove_channels`)
- the file is loaded in `int16` type. The `gain_to_uV` allows you to convert to uV (1 sample = 2.34 uV)
- this will not load the geometry of the probe. You can add probe info as follows:
```
import numpy as np
locations = np.load("path-to-/channel_positions.npy")
recording.set_channel_locations(locations)
```

Zeinab-Mohammadi, 2023-07-12:
Great, thanks, but I am getting this error:
[image]

alejoe91, 2023-07-12:
Mmm I guess locations are saved by kilosort and not for all channels. Can you try `print(locations.shape)`?

then you would need to manually create a NP1 layout and set the locations (or try to find another dataset!)

There are a bunch of dataset with Neurpopixels on DANDI: https://dandiarchive.org/

You can download the nwb file and then do:
```
import spikeinterface.extractors as se

recording = se.read_nwb_recording("path-to-file.nwb")
```

Zeinab-Mohammadi, 2023-07-12:
It is (374, 2). I think they excluded the Can you please explain more about how to make that NP1 please?
Also, I can see that [here](http://data.cortexlab.net/singlePhase3/), they have this code if it is useful :
`>> fn = 'rawDataSample.bin'; 
>> fid = fopen(fn, 'r'); 
>> dat = fread(fid, [385 Inf], '*int16'); 
>> fclose(fid); 
>> chanMap = readNPY('channel_map.npy'); 
>> dat = dat(chanMap+1,:); 
>> figure; imagesc(dat(:,1:30000))`

If I can not make it, I will try another dataset.

Zeinab-Mohammadi, 2023-07-12:
one more question, previously, I made had this code in Matlab to get a shorter data in .npy format corresponding to true channels:
`fid = fopen(fn, 'r'); 
dat = fread(fid, [385 Inf], '*int16'); 
fclose(fid); 
chanMap = readNPY('/.../channel_map.npy');
T = 11300000; %interested length, about 1/10 of data
dat2 = dat(chanMap+1,1:T); 
writeNPY(dat2, 'data_short.npy')`

Zeinab-Mohammadi, 2023-07-12:
I tried to sort the data with Mountainsortin in the below file. Can you please lmk if using Mountainsort with data from spikeinterface is correct here? Is there any issue? Thanks

`"Remember to active the enviroment as below "
# module load anaconda3/2023.3
# conda activate mountainsort

file ='/.../Neuropixels_data/'

import spikeinterface.full as si
import spikeinterface.preprocessing as spre
import mountainsort5 as ms5
import spikeinterface.extractors as se
import numpy as np
import spikeinterface.widgets as sw
import matplotlib.pyplot as plt
from pathlib import Path

recording = si.read_binary(file + 'Hopkins_20160722_g0_t0.imec.ap_CAR.bin', num_channels=385, dtype="int16", sampling_frequency=30000, gain_to_uV=2.34)
recording = recording.remove_channels([recording.channel_ids[-1]])
recording

locations = np.load(file + '/channel_positions.npy')
print(locations.shape)
recording.set_channel_locations(locations)

# noise_levels_microV = si.get_noise_levels(recording, return_scaled=True)
noise_levels_int16 = si.get_noise_levels(recording, return_scaled=False)

fig, ax = plt.subplots()
_ = ax.hist(noise_levels_int16, bins=np.arange(5, 30, 2.5))
ax.set_xlabel('noise')

# here we use static plot using matplotlib backend
# fig, axs = plt.subplots(ncols=1, figsize=(20, 10))

si.plot_timeseries(recording, backend='matplotlib',  clim=(-50, 50))

# plot some channels
fig, ax = plt.subplots(figsize=(20, 10))
some_chans = recording.channel_ids[[100, 150, 200, ]]
si.plot_timeseries({'data':recording}, backend='matplotlib', mode='line', ax=ax, channel_ids=some_chans)

# Make sure the recording is preprocessed appropriately
# lazy preprocessing
recording_filtered = spre.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording_preprocessed: si.BaseRecording = spre.whiten(recording_filtered, dtype='float32')

# use scheme 1
sorting = ms5.sorting_scheme1(
    recording=recording_preprocessed,
    sorting_parameters=ms5.Scheme1SortingParameters(...)
)

# or use scheme 2
sorting = ms5.sorting_scheme2(
    recording=recording_preprocessed,
    sorting_parameters=ms5.Scheme2SortingParameters(...)
)

# or use scheme 3
sorting = ms5.sorting_scheme3(
    recording=recording_preprocessed,
    sorting_parameters=ms5.Scheme3SortingParameters(...)
)

# Now you have a sorting object that you can save to disk or use for further analysis

`

alejoe91, 2023-07-13:
Clearly the problem is that you don't have channel locations!

To replicate exactly the piece of Matlab code in SI you can do:
```
import spikeinterface as si
import numpy as np

recording = si.read_binary("path-to-file.bin", num_channels=385, dtype="int16", sampling_frequency=30000, gain_to_uV=2.34)
print(recording)

# this selects the "active" channels
channel_map = np.load("path-to/channel_map.npy")
recording = recording.channel_slice(recording.channel_ids[channel_map])
print(recording)

# load and set locations (now it should work)
locations = np.load("path-to-/channel_positions.npy")
recording.set_channel_locations(locations)
```

As for running spikeinterface (or any other sorter), you can use the `run_sorter()` function:
```
import spikeinterface.sorters as ss

sorting_ms5 = ss.run_sorter("mountainsort5", recording)
```

I recommend spending some time reading in detail the ["How to - get started"](https://spikeinterface.readthedocs.io/en/latest/how_to/get_started.html) and watching this introductory [video tutorial](https://drive.google.com/file/d/1-eiXo6aCPyaBa_uP9ldOCkUa4cZOB3vr/view?usp=share_link):

Zeinab-Mohammadi, 2023-07-13:
Great, thanks for the help and sending the links. I will look at them and ask questions if needed.

alejoe91, 2023-07-14:
Can you make sure you have the latest spikeinterface version? `pip install -U spikeinterface`

Zeinab-Mohammadi, 2023-07-14:
I did update but no difference. It is ok for **recording1** but not for **recording**, so, whatever happens, is with **recording**. Please see:
[image]

Zeinab-Mohammadi, 2023-07-14:
Also,  If I want to sort/read half of the data (or generally only T recording samples), how should I do that, please?

alejoe91, 2023-07-14:
Then it's probably a bug. @Zeinab-Mohammadi I recommend reading the documentation thoroughly ;)

Especially this part: https://spikeinterface.readthedocs.io/en/latest/modules/index.html

I'll close the issue because it's too general. Feel free to open another one for the plotting bug and one for each specific question you have. You can also search in the Issue page for some keywords to see if some similar issues were solved.

closed 2023-07-14 by alejoe91

Zeinab-Mohammadi, 2023-07-14:
ok, thank you!