SpikeInterface/spikeinterface #1826: Loading sync channel using read_openephys

neuronzoo, 2023-07-14:
Hey guys, appreciate all of your terrific work!
I am using OE to record NPX and also save the sync channel - #385. I am trying to load the sync channel using read_openephys (in spikeinterface.extractors). When I use:
`new_recording = se.read_openephys(path_to_recording_file, stream_name="Record Node 101#Neuropix-PXI-100.ProbeA-AP", load_sync_channel=True)` 
I get the following error: "set_probe(inplace=True) must have all channel indices"

samuelgarcia, 2023-07-17:
Hi @neuronzoo.
thank you for reporting this.
Indeed this is a bug when we have the sync channel it is not possible to load the probe because the channel count is N+1.
@alejoe91 can you fix this like we do in spikeglx ?

alejoe91, 2023-07-17:
Yep, I can tackle this today

PR #1832 (2023-07-17): Do not load NP probe in OE if load_sync_channel=True
Fixes #1826
closed 2023-07-17 by samuelgarcia